robotology/gazebo-yarp-plugins #170: [Gazebo 4.1] FT plugins seems to not be instantiated

francesco-romano, 2015-01-26:
In Gazebo 4.1 no `analog` ports are being creating when instantiating the iCub model.
Instead this happens for the same model and using gazebo 4.0.

Maybe a change in how FT sensors are handled upstream? @traversaro

traversaro, 2015-01-26:
The IMU is working?

traversaro, 2015-01-26:
Quick test: I was unable to reproduce this issue in Gazebo 4.1.1 installed from binaries in Ubuntu 14.04 .

francesco-romano, 2015-01-26:
Just to add some more details:
~~~
makaveli$ brew info gazebo4
/usr/local/Cellar/gazebo4/4.1.0 (937 files, 106M)
  Built from source with: --with-dartsim
makaveli$ yarp name list
registration name /fra ip 129.206.161.78 port 10000 type tcp
registration name /icubGazeboSim/applyExternalWrench/rpc:i ip 129.206.161.78 port 10021 type tcp
registration name /icubGazeboSim/head/command:i ip 129.206.161.78 port 10007 type tcp
registration name /icubGazeboSim/head/rpc:i ip 129.206.161.78 port 10006 type tcp
registration name /icubGazeboSim/head/state:o ip 129.206.161.78 port 10008 type tcp
registration name /icubGazeboSim/inertial ip 129.206.161.78 port 10002 type tcp
registration name /icubGazeboSim/left_arm/command:i ip 129.206.161.78 port 10013 type tcp
registration name /icubGazeboSim/left_arm/rpc:i ip 129.206.161.78 port 10012 type tcp
registration name /icubGazeboSim/left_arm/state:o ip 129.206.161.78 port 10014 type tcp
registration name /icubGazeboSim/left_leg/command:i ip 129.206.161.78 port 10019 type tcp
registration name /icubGazeboSim/left_leg/rpc:i ip 129.206.161.78 port 10018 type tcp
registration name /icubGazeboSim/left_leg/state:o ip 129.206.161.78 port 10020 type tcp
registration name /icubGazeboSim/right_arm/command:i ip 129.206.161.78 port 10010 type tcp
registration name /icubGazeboSim/right_arm/rpc:i ip 129.206.161.78 port 10009 type tcp
registration name /icubGazeboSim/right_arm/state:o ip 129.206.161.78 port 10011 type tcp
registration name /icubGazeboSim/right_leg/command:i ip 129.206.161.78 port 10016 type tcp
registration name /icubGazeboSim/right_leg/rpc:i ip 129.206.161.78 port 10015 type tcp
registration name /icubGazeboSim/right_leg/state:o ip 129.206.161.78 port 10017 type tcp
registration name /icubGazeboSim/torso/command:i ip 129.206.161.78 port 10004 type tcp
registration name /icubGazeboSim/torso/rpc:i ip 129.206.161.78 port 10003 type tcp
registration name /icubGazeboSim/torso/state:o ip 129.206.161.78 port 10005 type tcp
registration name fallback ip 224.2.1.1 port 10000 type mcast
*** end of message
~~~

While
~~~
makaveli$ brew info gazebo4
/usr/local/Cellar/gazebo4/4.0.0 (829 files, 105M) *
  Built from source with: --with-dartsim
makaveli$ yarp name list
registration name /fra ip 129.206.161.78 port 10000 type tcp
registration name /icubGazeboSim/applyExternalWrench/rpc:i ip 129.206.161.78 port 10033 type tcp
registration name /icubGazeboSim/head/command:i ip 129.206.161.78 port 10019 type tcp
registration name /icubGazeboSim/head/rpc:i ip 129.206.161.78 port 10018 type tcp
registration name /icubGazeboSim/head/state:o ip 129.206.161.78 port 10020 type tcp
registration name /icubGazeboSim/inertial ip 129.206.161.78 port 10002 type tcp
registration name /icubGazeboSim/left_arm/analog:o ip 129.206.161.78 port 10012 type tcp
registration name /icubGazeboSim/left_arm/analog:o/rpc:i ip 129.206.161.78 port 10011 type tcp
registration name /icubGazeboSim/left_arm/command:i ip 129.206.161.78 port 10025 type tcp
registration name /icubGazeboSim/left_arm/rpc:i ip 129.206.161.78 port 10024 type tcp
registration name /icubGazeboSim/left_arm/state:o ip 129.206.161.78 port 10026 type tcp
registration name /icubGazeboSim/left_foot/analog:o ip 129.206.161.78 port 10006 type tcp
registration name /icubGazeboSim/left_foot/analog:o/rpc:i ip 129.206.161.78 port 10005 type tcp
registration name /icubGazeboSim/left_leg/analog:o ip 129.206.161.78 port 10004 type tcp
registration name /icubGazeboSim/left_leg/analog:o/rpc:i ip 129.206.161.78 port 10003 type tcp
registration name /icubGazeboSim/left_leg/command:i ip 129.206.161.78 port 10031 type tcp
registration name /icubGazeboSim/left_leg/rpc:i ip 129.206.161.78 port 10030 type tcp
registration name /icubGazeboSim/left_leg/state:o ip 129.206.161.78 port 10032 type tcp
registration name /icubGazeboSim/right_arm/analog:o ip 129.206.161.78 port 10014 type tcp
registration name /icubGazeboSim/right_arm/analog:o/rpc:i ip 129.206.161.78 port 10013 type tcp
registration name /icubGazeboSim/right_arm/command:i ip 129.206.161.78 port 10022 type tcp
registration name /icubGazeboSim/right_arm/rpc:i ip 129.206.161.78 port 10021 type tcp
registration name /icubGazeboSim/right_arm/state:o ip 129.206.161.78 port 10023 type tcp
registration name /icubGazeboSim/right_foot/analog:o ip 129.206.161.78 port 10010 type tcp
registration name /icubGazeboSim/right_foot/analog:o/rpc:i ip 129.206.161.78 port 10009 type tcp
registration name /icubGazeboSim/right_leg/analog:o ip 129.206.161.78 port 10008 type tcp
registration name /icubGazeboSim/right_leg/analog:o/rpc:i ip 129.206.161.78 port 10007 type tcp
registration name /icubGazeboSim/right_leg/command:i ip 129.206.161.78 port 10028 type tcp
registration name /icubGazeboSim/right_leg/rpc:i ip 129.206.161.78 port 10027 type tcp
registration name /icubGazeboSim/right_leg/state:o ip 129.206.161.78 port 10029 type tcp
registration name /icubGazeboSim/torso/command:i ip 129.206.161.78 port 10016 type tcp
registration name /icubGazeboSim/torso/rpc:i ip 129.206.161.78 port 10015 type tcp
registration name /icubGazeboSim/torso/state:o ip 129.206.161.78 port 10017 type tcp
registration name fallback ip 224.2.1.1 port 10000 type mcast
*** end of message
~~~

francesco-romano, 2015-01-27:
Tested with gazebo 4.1.1
Same problem.
~~~
makaveli$ yarp name list
registration name /fra ip 147.142.151.161 port 10000 type tcp
registration name /icubGazeboSim/applyExternalWrench/rpc:i ip 147.142.151.161 port 10021 type tcp
registration name /icubGazeboSim/head/command:i ip 147.142.151.161 port 10007 type tcp
registration name /icubGazeboSim/head/rpc:i ip 147.142.151.161 port 10006 type tcp
registration name /icubGazeboSim/head/state:o ip 147.142.151.161 port 10008 type tcp
registration name /icubGazeboSim/inertial ip 147.142.151.161 port 10002 type tcp
registration name /icubGazeboSim/left_arm/command:i ip 147.142.151.161 port 10013 type tcp
registration name /icubGazeboSim/left_arm/rpc:i ip 147.142.151.161 port 10012 type tcp
registration name /icubGazeboSim/left_arm/state:o ip 147.142.151.161 port 10014 type tcp
registration name /icubGazeboSim/left_leg/command:i ip 147.142.151.161 port 10019 type tcp
registration name /icubGazeboSim/left_leg/rpc:i ip 147.142.151.161 port 10018 type tcp
registration name /icubGazeboSim/left_leg/state:o ip 147.142.151.161 port 10020 type tcp
registration name /icubGazeboSim/right_arm/command:i ip 147.142.151.161 port 10010 type tcp
registration name /icubGazeboSim/right_arm/rpc:i ip 147.142.151.161 port 10009 type tcp
registration name /icubGazeboSim/right_arm/state:o ip 147.142.151.161 port 10011 type tcp
registration name /icubGazeboSim/right_leg/command:i ip 147.142.151.161 port 10016 type tcp
registration name /icubGazeboSim/right_leg/rpc:i ip 147.142.151.161 port 10015 type tcp
registration name /icubGazeboSim/right_leg/state:o ip 147.142.151.161 port 10017 type tcp
registration name /icubGazeboSim/torso/command:i ip 147.142.151.161 port 10004 type tcp
registration name /icubGazeboSim/torso/rpc:i ip 147.142.151.161 port 10003 type tcp
registration name /icubGazeboSim/torso/state:o ip 147.142.151.161 port 10005 type tcp
registration name fallback ip 224.2.1.1 port 10000 type mcast
~~~

PS: ports are still around after closing gazebo, but they are correctly closed if the model is first removed before quitting the app.

traversaro, 2015-01-27:
We should write a quick test (similar to the `TestCamera` one) in https://github.com/robotology/robot-testing to check this kind of regressions.

francesco-romano, 2015-01-27:
Very simple for now:
https://github.com/robotology/robot-testing/pull/3

francesco-romano, 2015-01-27:
Of course the test succeed on OS X for gazebo 4.0 and fails for gazebo 4.1.x

francesco-romano, 2015-01-28:
5.0 seems to have solved this problem

closed 2015-05-04 by francesco-romano