robotology/gazebo-yarp-plugins #171: [Gazebo5] Test gazebo-yarp-plugins with Gazebo5

traversaro, 2015-01-28:
Gazebo5 have been released yesterday. In these release there are several things that could bring incompatibility (for example C++11), we should have to test `gazebo-yarp-plugins`. I cannot test Gazebo5 in Travis because it is not supported on Ubuntu 12.04, so anyone is willing to test it and report any issue? 

Subtasks:
* [ ] Test gazebo-yarp-plugins with Gazebo 5 on Ubuntu 
* [ ] Test gazebo-yarp-plugins with Gazebo 5 on OS X

EnricoMingo, 2015-01-28:
Hi, I have done some tests using Gazebo 5 and considering a "normal" usage,
it is working (ubuntu 14.04).

traversaro, 2015-01-28:
Perfect!

francesco-romano, 2015-01-28:
OS X:
- Successfully installed (yeah!)
- FT sensors ports are present
- Motor test works until joint 6 (tests are not correctly configured for gazebo)
- ?

traversaro, 2015-01-28:
Fine for me.

closed 2015-01-28 by traversaro