robotology/gazebo-yarp-plugins #173: After removing the robot iCub, I cannot put it anymore

DanielePucci, 2015-02-03:
Once I remove the robot by clicking on it and pressing "Delete", I can no longer put a new model. Can this behavior be due to the gazebo-yarp-plugins?

DanielePucci, 2015-02-03:
This issue seems to be related to the clock. As a matter of fact, by taking off the line 

export YARP_CLOCK=/clock 

in my .bashrc, this problem disappears (in this case I launch Gazebo without the additional string -slibgazebo_yarp_clock.so)

MirkoFerrati, 2015-04-04:
You can no longer put any model, because the simulator is blocked by some AnalogWrapper that is not closing properly after you tried to delete the model.
This issue is actually the same as #181 .
Gazebo 3 and Gazebo 4, with the Walkman model, produce the same behavior.
@traversaro merge this into #181?

traversaro, 2015-04-04:
Duplicate of #181, closing.

closed 2015-04-04 by traversaro