robotology/gazebo-yarp-plugins #174: [Dissemination] Create homebrew formula

francesco-romano, 2015-02-05:
As specified in #95

francesco-romano, 2015-02-24:
A first, basic and maybe not working, implementation here https://github.com/robotology/gazebo-yarp-plugins/blob/3313b11c00149de0e400bf7485cad91463ab1287/utilities/installation/brew/gazebo-yarp-plugins.rb

francesco-romano, 2015-02-24:
using `yarp` as recommended dependency seems the best option:
by default yarp is built with brew. But it is possible to remove the dependency and specify by environment variable the `YARP_DIR` path.

Strange thing: in the brew environment I get this error:
~~~
bash-3.2$ cmake .
-- YARP is version: 2.3.63.5
-- Boost version: 1.57.0
CMake Error at /usr/local/lib/cmake/gazebo/gazebo-config.cmake:46 (find_package):
  By not providing "FindOGRE.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OGRE", but
  CMake did not find one.

  Could not find a package configuration file provided by "OGRE" with any of
  the following names:

    OGREConfig.cmake
    ogre-config.cmake

  Add the installation prefix of "OGRE" to CMAKE_PREFIX_PATH or set
  "OGRE_DIR" to a directory containing one of the above files.  If "OGRE"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:16 (find_package)

-- Configuring incomplete, errors occurred!
See also "/tmp/gazebo-yarp-plugins-YaxywX/CMakeFiles/CMakeOutput.log".
~~~
but this not happens when building the project (cloned by github).
Any ideas @traversaro

francesco-romano, 2015-02-24:
@traversaro solved.
The `gazebo-config.cmake` uses `pkg-config`  and I had to add it as a dependency of the formula or in the "sandboxed" environment the command was not available.

francesco-romano, 2015-02-24:
Almost there. For some reasons the inclusion of OGRE is not passed to the compiler in the brew environment.

francesco-romano, 2015-04-23:
I just retried and cmake successfully configures the project, but I get
```bash
In file included from /tmp/gazebo-yarp-plugins20150423-51434-ygv2zs/src/gazebo_plugins/Camera.cc:8:
In file included from /tmp/gazebo-yarp-plugins20150423-51434-ygv2zs/include/gazebo_yarp_plugins/Camera.hh:12:
In file included from /usr/local/Cellar/gazebo5/5.1.0/include/gazebo-5.1/gazebo/plugins/CameraPlugin.hh:24:
In file included from /usr/local/Cellar/gazebo5/5.1.0/include/gazebo-5.1/gazebo/rendering/Camera.hh:43:
In file included from /usr/local/Cellar/gazebo5/5.1.0/include/gazebo-5.1/gazebo/rendering/ogre_gazebo.h:58:
In file included from /usr/local/include/OGRE/Paging/OgrePageManager.h:32:
/usr/local/include/OGRE/Paging/OgrePagingPrerequisites.h:32:10: fatal error: 'OgrePrerequisites.h' file not found
#include "OgrePrerequisites.h"
         ^
In file included from /tmp/gazebo-yarp-plugins20150423-51434-ygv2zs/src/yarp_drivers/CameraDriver.cpp:8:
In file included from /tmp/gazebo-yarp-plugins20150423-51434-ygv2zs/include/gazebo_yarp_plugins/CameraDriver.h:17:
In file included from /usr/local/Cellar/gazebo5/5.1.0/include/gazebo-5.1/gazebo/rendering/Camera.hh:43:
In file included from /usr/local/Cellar/gazebo5/5.1.0/include/gazebo-5.1/gazebo/rendering/ogre_gazebo.h:58:
In file included from /usr/local/include/OGRE/Paging/OgrePageManager.h:32:
/usr/local/include/OGRE/Paging/OgrePagingPrerequisites.h:32:10: fatal error: 'OgrePrerequisites.h' file not found
#include "OgrePrerequisites.h"
```

francesco-romano, 2015-04-23:
By explicitly adding `-I/usr/local/include/OGRE` to the compiler flags I manage to compile it.
@traversaro any ideas why that folder is not included by default?

traversaro, 2015-04-23:
Mhh... apparently is a regression in gazebo5, that we never noticed because travis is not building against gazebo5 (now it is: https://travis-ci.org/robotology/gazebo-yarp-plugins/jobs/59724198 ).

I don't know if it is intentional or not, tutorial on plugins just mention sdformat and boost as build dependencies... perhaps the easiest solution for the moment is to add the logic to include the OGRE headers in the CMakeLists.txt file.

traversaro, 2015-04-23:
No, the travis build is apparently failing because Gazebo5 is not available in 12.04 ... anyway can you test if the modification in https://github.com/robotology/gazebo-yarp-plugins/commit/12546c4fda257d298e17a8b10ad34a5315281004 are enough to fix the brew formula?

francesco-romano, 2015-04-23:
Open issue (after #186):

[] only compiles with `--HEAD` option (the release is too old, but I'm leaving it as a placeholder).
[] we should investigate if we can install also the icub-gazebo models in the formula
[] tests can (should) be enabled
[] double `cmake` command to make it work :disappointed: 
[ ] have a clean `audit`
```bash
brew audit gazebo-yarp-plugins 
gazebo-yarp-plugins:
 * Use the `cp` Ruby method instead of `system "cp"`

Error: 1 problems in 1 formulae
```

francesco-romano, 2015-07-14:
Update:
Commit https://github.com/robotology/gazebo-yarp-plugins/commit/98f00301144fa01ded8a12440d067e34a0397b9a adds a working formula.

We still have the following issue: double cmake. But this happens also on a normal execution..maybe it depends on some issues of gazebo.
About the icub-gazebo: I decided to drop that feature. It does not make any sense.

So, we are only left with one problem: where do we publish our formula?
Together with gazebo? (so in `orsf/simulation`)? In hombrew science (together with yarp)?

cc @traversaro @iron76

traversaro, 2015-07-14:
It is possible to get it in homebrew science even if it depends on gazebo that is contained in `osrf/simulation` ?

francesco-romano, 2015-07-14:
Unfortunately I don't know.
The problem is the either way one dependency is not in the main tap:
yarp in in x11 (but it should be moved into science)
gazebo is in osfr/simulation

traversaro, 2016-11-18:
The formula  is being distributed in https://github.com/robotology/homebrew-cask , closing.

closed 2016-11-18 by traversaro