robotology/gazebo-yarp-plugins #177: [FT Sensor plugin] Strange issue with ft sensor plugin

EnricoMingo, 2015-02-13:
I am having a strange issue with the ft sensors plugins. Basically I am using some variables to generate a model that has only the legs, only the upper body or both. If I generate the only upper body version, the force torque plugin associated to the left arm is created and it is present in gazebo but the port does not publish anything (the size of the vector of data is 0). This happen ONLY with the upper body and ONLY with that sensor. I have already checked the sdf and the plugin is there, the same for the gazebo topic. 
Any idea?
I am using the gazebo_yarp_plugins f18d3067122189c338270cb83312c2296022f4b3 version and Gazebo 5.0.0.

traversaro, 2015-02-13:
Do you have an example of non-working sdf file?

EnricoMingo, 2015-02-13:
Yes, I'll send to you.

barbalberto, 2015-02-15:
Can you post also the terminal output to see any printed errors? Thanks

EnricoMingo, 2015-02-15:
Here it is the output of the gazebo terminal:

```
GazeboYarpClock loaded.
yarp: Port /clock active at tcp://192.168.1.71:10002
Error [Param.cc:181] Unable to set value [1,0471975511965976] for key[horizontal_fov]
Error [Param.cc:181] Unable to set value [0,100000001] for key[near]
*** GazeboYarpForceTorque plugin started ***
Found yarpConfigurationFile: loading from /home/enrico/walkman/drc/iit-bigman-ros-pkg/bigman_gazebo/database/bigman_urdf/conf/bigman/bigman_gazebo_left_arm_ft.ini
Calling GazeboYarpPlugins::Handler Constructor
GazeboYarpPlugins::Handler: Inserting Sensor : default::bigman::LWrj2::left_arm
Singleton: Added a new sensor default::bigman::LWrj2::left_arm.
AnalogWrapper Debugl_arm_ft--> 0 5 0 5
yarp: Port /bigman/l_arm_ft/analog:o/forceTorque/rpc:i active at tcp://192.168.1.71:10003
yarp: created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
opening port forceTorqueGazeboYarpForceTorque Plugin: correcly opened GazeboYarpForceTorqueDriver wrapper
GazeboYarpForceTorqueDriver::open() called
GazeboYarpForceTorqueDriver::open is looking for sensor default::bigman::LWrj2::left_arm...
Looking for sensor : default::bigman::LWrj2::left_arm
Sensor default::bigman::LWrj2::left_arm was happily found!
GazeboYarpForceTorqueDriver::open() returning true
yarp: created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
GazeboYarpForceTorque Plugin: correcly opened GazeboYarpForceTorqueDriver
yarp: Port /bigman/l_arm_ft/analog:o/forceTorque active at tcp://192.168.1.71:10004
GazeboYarpForceTorque : wrapper was connected with driver 
*** GazeboYarpForceTorque plugin started ***
yarp: Anonymous port active at tcp://192.168.1.71:56373
yarp: Sending output from /clock to /192.168.1.71:56373 using tcp
yarp: Receiving input from /clock to /192.168.1.71:56373 using tcp
Found yarpConfigurationFile: loading from /home/enrico/walkman/drc/iit-bigman-ros-pkg/bigman_gazebo/database/bigman_urdf/conf/bigman/bigman_gazebo_right_arm_ft.ini
GazeboYarpPlugins::Handler: Inserting Sensor : default::bigman::RWrj2::right_arm
Singleton: Added a new sensor default::bigman::RWrj2::right_arm.
AnalogWrapper Debugr_arm_ft--> 0 5 0 5
yarp: Port /bigman/r_arm_ft/analog:o/forceTorque/rpc:i active at tcp://192.168.1.71:10005
yarp: created wrapper <analogServer>. See C++ class AnalogWrapper for documentation.
opening port forceTorqueGazeboYarpForceTorque Plugin: correcly opened GazeboYarpForceTorqueDriver wrapper
GazeboYarpForceTorqueDriver::open() called
GazeboYarpForceTorqueDriver::open is looking for sensor default::bigman::RWrj2::right_arm...
Looking for sensor : default::bigman::RWrj2::right_arm
Sensor default::bigman::RWrj2::right_arm was happily found!
GazeboYarpForceTorqueDriver::open() returning true
yarp: created device <gazebo_forcetorque>. See C++ class GazeboYarpForceTorqueDriver for documentation.
GazeboYarpForceTorque Plugin: correcly opened GazeboYarpForceTorqueDriver
yarp: Port /bigman/r_arm_ft/analog:o/forceTorque active at tcp://192.168.1.71:10006
GazeboYarpForceTorque : wrapper was connected with driver 
*** GazeboYarpControlBoard plugin started ***
GazeboYarpPlugins::Handler: Inserting Robot : bigman
Singleton: Added a new robot bigman.
GazeboYarpControlBoard: Found yarpConfigurationFile: loading from /home/enrico/walkman/drc/iit-bigman-ros-pkg/bigman_gazebo/database/bigman_urdf/conf/bigman/bigman_gazebo_torso.ini
yarp: Port /bigman/torso/rpc:i active at tcp://192.168.1.71:10007
yarp: Port /bigman/torso/command:i active at tcp://192.168.1.71:10008
yarp: Port /bigman/torso/state:o active at tcp://192.168.1.71:10009
yarp: created wrapper <controlboardwrapper2>. See C++ class ControlBoardWrapper2 for documentation.
GazeboYarpControlBoard: wrapper opened correctly
GazeboYarpControlBoard: setting robotScopedName bigman
DeviceDriver is looking for robot bigman...
Looking for robot : bigman
Robot bigman was happily found!
Robot Name: bigman
# Joints: 21
# Links: 21
.ini file found, using joint names in ini file
Joint Limits
WaistLat max_pos: 34.9504 min_pos: -34.9504
WaistSag max_pos: 79.9849 min_pos: -19.9962
WaistYaw max_pos: 162.949 min_pos: -162.949
No minimum stiffness value found in ini file, default one will be used!
No maximum stiffness value found in ini file, default one will be used!
No minimum dampings value found in ini file, default one will be used!
No maximum damping value found in ini file, default one will be used!
[... 324 lines cut ...]
# Links: 21
.ini file found, using joint names in ini file
Joint Limits
r_handj max_pos: 89.9973 min_pos: 0
No minimum stiffness value found in ini file, default one will be used!
No maximum stiffness value found in ini file, default one will be used!
No minimum dampings value found in ini file, default one will be used!
No maximum damping value found in ini file, default one will be used!
min_stiffness: [  0.000000 ]
max_stiffness: [  1000.000000 ]
min_damping: [  0.000000 ]
max_damping: [  100.000000 ]
Found PID information in plugin parameters group GAZEBO_PIDS
  P: 0.1 I: 0 D: 0.001 maxInt: 9999 maxOut: 9999
OK!
PID gain information not found in plugin parameters, using default gains ( P 500 I 0.1 D 0 )
PID gain information not found in plugin parameters, using default gains ( P 500 I 0 D 1 )
gazebo_init set pid done!
INITIAL CONFIGURATION IS:  0.000000
yarp: created device <gazebo_controlboard>. See C++ class GazeboYarpControlBoardDriver for documentation.
controlBoard opened correctly
Device initialized correctly, now sitting and waiting cause I am just the main of the yarp device, and the robot is linked to the onUpdate event of gazebo
Loaded GazeboYarpControlBoard Plugin
yarp: Sending output from /bigman/l_arm_ft/analog:o/forceTorque to /tmp/port/1 using tcp
```

EnricoMingo, 2015-02-15:
This is the output when I try to do a yarp read:

```
yarp read ... /bigman/l_arm_ft/analog:o/forceTorque
yarp: Port /tmp/port/1 active at tcp://192.168.1.71:10034
yarp: Receiving input from /bigman/l_arm_ft/analog:o/forceTorque to /tmp/port/1 using tcp
```

traversaro, 2015-02-15:
Can you try without the yarp clock plugin? Perhaps there is some race
condition.

EnricoMingo, 2015-02-15:
I confirm that without the yarp clock plugin both the force-torque are broadcasted in YARP.

traversaro, 2015-02-16:
Then a relevant issue is this one: https://github.com/robotology/yarp/issues/331 .

francesco-romano, 2015-02-16:
@EnricoMingo which version of yarp are you using? Have you tried with the latest?

EnricoMingo, 2015-02-16:
YARP ver: 3ef253cc910eb36e335ef3d7db7078d7a3316a74 (16 Oct)
I have not tried yet with master.

francesco-romano, 2015-02-16:
When you have some time, can you try on master, please?

EnricoMingo, 2015-02-16:
Sure!

EnricoMingo, 2015-02-16:
I have just tried to run the simulation with the last version of YARP and GazeboYarpPlugins but again the port is opened but nothing is published.

barbalberto, 2015-02-18:
One thing I saw in the output was:

yarp: Anonymous port active at tcp://192.168.1.71:56373
yarp: Sending output from /clock to /192.168.1.71:56373 using tcp
yarp: Receiving input from /clock to /192.168.1.71:56373 using tcp

Is it possible that the clock port is registered as 'Anonymous' whatever itt means and therefore the wrapper is not able to get the time from it?

barbalberto, 2015-10-22:
@EnricoMingo , is the problem still there?

traversaro, 2016-11-08:
Closing for now, @EnricoMingo feel free to reopen if the problem still persist.

closed 2016-11-08 by traversaro