robotology/gazebo-yarp-plugins #181: Robot not removed properly when using clock plugin

jeljaik, 2015-03-27:
While performing some checks on the synchronization of Gazebo and Matlab through the ySynchronizer, which arbitrates this process by sending rpc commands to either stop or continue the simulation, I noticed that when removing the robot Gazebo gets stuck while attempting to close `GazeboYarpIMU` with the following message:

```bash
*** GazeboYarpIMU closing ***
Closing Server Inertial...
```

This happens not only after I have used the `ySynchronizer` block but also by just placing the robot and removing it without further interaction. Totally reproducible.

MirkoFerrati, 2015-04-04:
This error is not related to the IMU, but to any analog sensor. If you disable the IMU in the URDF, you will see that the simulation blocks trying to close jointvelocity sensors or forcetorque sensors (if any). 
The IMU is only the first one to be closed. I have been investigating this issue for 5 days now, and I managed to reduce the deadlock around the AnalogWrapper::Close() function. I have no idea what's down there, among RateThreads, ThreadImpl, PortCore, RpcPorts and so on, but one of the YARP components is deadlocked for sure.
I can also add that even if the simulation is blocked, you can still yarp rpc /yourRobot/anyAnalogSensorRpcInterface and calibrate them, meaning that the communication thread is still running happily.
@traversaro  can we increase the scope and the gravity of this issue?

jeljaik, 2015-04-04:
Nuno and Prashanth should be investigating this issue as well.

naveenoid, 2015-04-04:
@prashanthr05 @nunoguedelha

MirkoFerrati, 2015-04-04:
Ok, I just found out the single line that produces deadlock: 
https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/modules/AnalogWrapper/AnalogWrapper.cpp#L469

If you comment it out, recompile yarp, recompile GazeboYarpPlugin, everything will work (except that you do not get timestamps in the envelope of the analog sensor data).
If you need this fast, you can apply this small change and stop reading.

A not so good workaround (but give me more time and it will become good) is:
- turn m_jointsensorwrapper into a polydriver*
- substitute this line (m_jointsensorwrapper.close()):
https://github.com/robotology/gazebo-yarp-plugins/blob/master/src/gazebo_plugins/JointSensors.cc#L33

with something like this:

boost::thread* closeThread= new boost::thread(boost::bind(&close_thread,m_jointsensorsWrapper));

where close_thread is a function that takes a polydriver* and calls close.

I am not posting code because I am still writing a merge that does not leave boost::thread* objects around without deleting them.

The changes above should be applied also to IMU and ForceTorque sensors.

traversaro, 2015-04-04:
@MirkoFerrati which version (checkout) of yarp are you using?

Some of the things that I think could be related:
 * Unfortunatly the clock synchronization logic in yarp is undocumented. However, the best way of learning its logic (besides looking at the original issue: https://github.com/robotology/gazebo-yarp-plugins/issues/57 is reading the source code, for example the one of the `NetworkClock` class: https://github.com/robotology/yarp/blob/master/src/libYARP_OS/src/NetworkClock.cpp . Note that code inspection is greatly simpliefied if you use a proper tool (IDE or editor plugin) for following the code execution flow.  
 * @paulfitz did some changes in the network clock synchronization code in yarp to solve a similar bug : https://github.com/robotology/yarp/issues/331 ) perhaps this bug could be related  to those changes? 
 * @barbalberto recently merged some modifications of the analogWrapper, perhaps it can take a quick look if there is anything fishy going on around `analogWrapper::close()`
 * The original design idea of @paulfitz  for clock synchronization was using a topic (persistent), but nobody had the time to properly test the use of a `/clock` topic over a `/clock` port, see https://github.com/robotology/gazebo-yarp-plugins/issues/157 (Probably not related, but it has always worried me because I never totally get all the internals of clock synchronization) 

cc @nunoguedelha @prashanthr05 

My personal impression is that that the gazebo physics thread is calling `analogWrapper::close()` during the plugin teardown. During the teardwon, something called by `analogWrapper::close()` is calling a `yarp::os::Time::delay()`, that however never exits because it waits for at least a clock tick (that triggers a `NetworkClock::read()` ). Unfortunately the clock is never updated, because it should be updated by the gazebo physics thread, that is the one calling the `delay()`! 

P.s. while I was writing this reply, I quickly checked all the `delay()` calls in yarp codebase (easy if you use an IDE!) and I guess that there are some fishy calls in 
the `DgramTwoWaysStream` class `interrupt` and `close` methods: 
https://github.com/robotology/yarp/blob/master/src/libYARP_OS/src/DgramTwoWayStream.cpp

MirkoFerrati, 2015-04-04:
@traversaro  Inside `analogWrapper::close()` there is a call to `Stamp::Update()`, which in turn enters `yarp::os::Time` and deadlocks, I don't know it the reason is `delay()` or some different function ( I guess `time::now()` directly is the responsible).
I am using an old Yarp version, along with an old GazeboYarpPlugin. I am not able of moving into master. I would appreciate if you can test this:
- comment out line https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/modules/AnalogWrapper/AnalogWrapper.cpp#L469 and try if everything works on master

Btw, I checked the implementation of ControlBoardWrapper in order to understand why the controlboards close correctly. The timestamp is also updated in the thread, but in a different way: the time is not returned from `time::now` but from the encoder sensor structure (search for this                 `timeStamp+=device.subdevices[k].encodersTimes[l];` )
, which is filled inside ControlBoardDriverEncoders in the `getEncodersTimed` with the value updated from the ControlBoardDriver `m_lastTimestamp.update(_info.simTime.Double());`

Basically, ControlBoardWrapper uses simulated timestamp, AnalogWrapper does not.

traversaro, 2015-04-04:
Sorry, unable to test this at the moment. Which `Stamp::update()` are you referring to? Then one in `analogWrapper::run()` ?

MirkoFerrati, 2015-04-04:
Yes, that one. I am testing my changes in a larger URDF model, there may be problems even with the ControlBoards, but they seem to raise only if another AnalogWrapper is closing. I will let you know.

traversaro, 2015-04-04:
Ok, thanks. Anyway if someone has some time to allocate for serious debugging, I guess that get a clear idea of the issue we should: 
* Make sure that `yarp` and `gazebo-yarp-plugins` are compiled with debug simbols (compile in `Debug` or `RelWithDebInfo`) and preferably debug symbols for gazebo are installed (via the `gazebo*-dbg` packages).  
* Try to replicate this bug with only one AnalogWrapper (to reduce the number of threads)
* Get a `gdb` backtrace for all the threads, so we can understand the deadlock .

traversaro, 2015-04-04:
@MirkoFerrati I was thinkink a bit about this, but are you sure that `analogWrapper::run` is called by the thread that is calling `analogWrapper::close()` ?  At which line of which file the `run` method is called?

traversaro, 2015-04-04:
Another fishy difference between `ControlBoardWrapper` and `AnalogWrapper` is in the close, in the AnalogWrapper the `stop` method is basically called two times:
~~~
    if (RateThread::isRunning())
    {
        RateThread::stop();
    }

    RateThread::stop();
~~~

MirkoFerrati, 2015-04-04:
No no no, `close()` calls `join()` on the thread `run()`. But if `run()` is blocked waiting for a clock do unblock some yarp::os::Time primitives, the `close()` is blocked too.

Anyway, my changes to Handler.cc are working with AnalogWrapper, but not with IMU. It looks like ServerInertial.h has a very different structure from AnalogWrapper :(

MirkoFerrati, 2015-04-04:
Update: I have the ServerInertial closing fine, but I had to put an hack in its close function in order to tell our IMUDriver to stop reading from Gazebo. 
How can I call the `close()` method of a DeviceDriver from the ServerInertial that is using it as a IGenericSensor?
Also: can I make some changes to the ServerInertial.h inside Yarp? What's the policy?

MirkoFerrati, 2015-04-04:
Branch issue181 works as expected. I don't think we can merge the hack in master, but as soon as someone knows how to do the fix in the yarp way, it should be done!
@traversaro  will you take care of this? I for sure will propagate this fix in walk-man computers.

traversaro, 2015-04-04:
* Ah, I get what you meant. If the problem is relative to `ThreadImpl::join`, it could be related to https://github.com/robotology/yarp/issues/331 . 
* Yes, the ServerInertial device is quite different from the other wrapper + devices, It is quite old and quite counter intuitive. @barbalberto is refactoring that part of yarp, so I guess he can provide more information, also related to the `IGenericSensor` / `IAnalogSensor` duality. Anyway the policy of changes in YARP is that you can propose a pull request, where the change will be discussed and eventually merge. If you want to discuss a change before opening the pull request, feel free to open an issue in https://github.com/robotology/yarp . 
* Yes, I would avoid merging all this hacks in master. I will be in Paris so I will not directly take care of this, but I guess other people in the lab will make sure to find a proper solution. If someone really need this bug fixed as soon as possible, he can always use the `issue181`.

francesco-romano, 2015-04-05:
`TheadImpl::joint` gives problem also here: https://github.com/robotology/yarp/issues/469

nunoguedelha, 2015-04-07:
I could reproduce this issue even without using the port /clock, without running Matlab/Simulink nor the synchronisation through the **`ySynchronizer`**. I just run the YARP server and Gazebo with the iCub model and the verbose option ( *gazebo --verbose /Users/nunoguedelha/dev/icub-gazebo/world/icub.world* ). When removing the robot, Gazebo gets stuck while attempting to close **`GazeboYarpIMU`**.

lornat75, 2015-04-07:
If this can be helpful, from the documentation:
void yarp::os::RateThread::stop	(		)	
Call this to stop the thread, this call blocks until the thread is terminated (and releaseThread() called).
Actually calls join. This will deadlock if called from run(), use askToStop() instead.

Not sure this is the problem you are talking about.

lornat75, 2015-04-07:
This code:

if (RateThread::isRunning())
    {
        RateThread::stop();
    }

    RateThread::stop();

seems strange though.

MirkoFerrati, 2015-04-07:
The first stop() is the right one, the second one should be commented out. The deadlock raises anyway.
What's the expected behavior of a RateThread when its destructor gets called while running?
Should it stop? Should it wait for join()? std::thread calls std::terminate and aborts the application.

lornat75, 2015-04-08:
If I am not mistaken the destructor calls stop+join.If stops is called outside the thread I would say the problem is that the thread does not exit. Can you add a fprint(stderr) in the releaseThread() function?

MirkoFerrati, 2015-04-08:
I added couts in every destructor/stop/stopMain/join/joinImpl that was involved in the process.
The stop() function basically sets a variable `closing` to true, and the thread loop is like `while(!closing)`.
The problem is that the thread does never reach the end of the current loop because it is blocked by a call to yarp::Time, which is blocked by gazebo not sending time because the simulation is blocked waiting for the destructor to destroy the thread.
Thus, the thread will never check for the value of `closing`, making the stop function useless.

MirkoFerrati, 2015-04-08:
Also, the difference between askToStop and Stop is that after setting `closing`, Stop will call Join, which is called anyway by the destructor.

prashanthr05, 2015-04-08:
Relating the issue, when I try to simulate the robot in gazebo while running the ySynchronizer, I do not have any issue while removing the robot. The IMU and ForceTorque sensors are closed properly. The versions of codyco-superbuild (latest commit on Apr07) , yarp (latest commit on Apr08) and gazebo_yarp_plugins (latest commit on Mar23) are upto-date.
The output on the terminal looks like, in the image.
[image]

Although I have not looked deeply into the code of AnalogWrapper.cpp or the JointSensor.cc, I noticed that the changes suggested by @MirkoFerrati are not done i.e commenting the StateStamp or modifying the m_jointsensorwrapper.close().

traversaro, 2015-04-08:
@prashanthr05 I guess  this is expected if you are not launching gazebo with the clock enabled (i.e. `gazebo -slibgazebo_yarp_clock.so` ) or you are launching gazebo without the `YARP_CLOCK` enviromental variable set. 

@nunoguedelha that seems a quite strange behaviour. Which version of yarp and gazebo are you using?

prashanthr05, 2015-04-08:
@traversaro : gazebo was running with clock plugin.. And about the YARP_CLOCK environment variable, it was set to /clock already but I didn't notice earlier that it was commented in the bashrc script. Although I uncommented the line and sourced the bashrc script later. And could reproduce the same results again.

nunoguedelha, 2015-04-09:
I've retested after updating all my repos to last master:
I don't have the issue if the variable `YARP_CLOCK` is unset, whatever option is specified when launching `gazebo` (so, with or without `-s libgazebo_yarp_clock.so`).
I have the issue if I set `YARP_CLOCK=/clock`, whatever option is specified when launching `gazebo` (so, with or without `-s libgazebo_yarp_clock.so`).

So the issue occurs as soon as you set `YARP_CLOCK=/clock`, even if the clock plugin is not enabled (and /clock + /clock/rpc ports created).

MirkoFerrati, 2015-04-12:
@nunoguedelha  This is expected, since the deadlock arises when trying to get an external clock which is not published. Notice that when a model is deleted from gazebo, the clock is no longer published, which is the same behavior as not running the libgazebo_yarp_clock plugin.

jeljaik, 2015-04-13:
Just to let you know, we tried as you mentioned in https://github.com/robotology/gazebo-yarp-plugins/issues/181#issuecomment-89553000 commenting out the line containing `lastStateStamp.update()` but the problem remains the same. We are using the `master` branches at the current versions of both YARP and gazebo-yarp-plugins.

MirkoFerrati, 2015-04-13:
@jeljaik That test is outdated now. It could work on AnalogWrappers, but for sure not on ServerAnalog (which handles the IMU). If you are into testing, you can try commenting out the IMU plugin from your URDF and see what happens with the lastStateStamp.update() commented out.
Anyway, since I am using an old version of YARP, there could be some more lines related to time which I missed in my version.

Another test is my branch issue181 which should solve all the problems even for latest YARP (at the cost of introducing some hacks in the code).
https://github.com/robotology/gazebo-yarp-plugins/tree/issue181

Give it a try!

jeljaik, 2015-04-13:
Thank you I'm into properly debugging this along with Nuno. Thanks for the
hints. We'll keep you posted.

nunoguedelha, 2015-04-20:
I've debugged again with Jorh the issue around the ServerInertial closure. Last debug analysis was done after a complete update of the software last Friday. All the analysis elements had somehow already been commented by you all along the thread above, but I would like to sum up the main path topics leading to the deadlock, at least the ones we've checked :
- the thread closing the `ServerInertial` runs the call tree `~GazeboYarpIMU()` => `Thread::stop()` => `ThreadImpl::close()` which ends waiting indefinitely for the thread to stop and be released. `close()` sets the thread's `stopping` flag to TRUE.
- another thread was executing the main running loop `ServerInertial::run()` (while `stopping` was false). But before it could check that `stopping=true`, it gets into an infinite wait after calling `yarp::os::Time::delay()`, as delay is waiting for the gazebo clock through port `/clock`.
- so we don't return from `ServerInertial::run()`, and don't release the thread which would unblock `~GazeboYarpIMU()`.
- YARP rpc commands are still working except for the fact they have no effect on gazebo. `GazeboYarpClock::clockUpdate()` (which gets the gazebo timing through gazebo API and posts it to `/clock`) is not called anymore by gazebo because gazebo is waiting for the ServerInertial thread to be released. We still can read gazebo time and check it's freezed.
- I have used a hack in GazeboYarpClock::clockStep() to unblock gazebo by generating a fake clock. **So nothing else should be wrong except for the clock not being posted**.

There is a global issue regarding the time handling between YARP, Gazebo and the clock plugin. If we hack the code somewhere for avoiding a delay, we get stuck somewhere else where YARP uses the Network clock (`/clock`), for instance every time the pending semaphores are checked (in `NetworkClock::read()`).

Here's how I see a possible solution :
- YARP: yarp should have in its API a service to switch YARP clock (let's say **`Yarp_API_clock_switch`**) from system to network and vice versa, dynamically. Along with this, it should handle a virtual internal clock, for avoiding glitches during the time source switching (I can detail this later on). I would start YARP with the default system clock and leave the handling of YARP_CLOCK environment variable to the clock plugin.
- Gazebo Yarp Clock plugin: we should add to its API a method that switches YARP clock (through the **`Yarp_API_clock_switch`** service). Somehow, this clock plugin service should be called after the creation (system->network clock) and before the removal of the robot model (network->system clock).

Please let me know what you think about that...

nunoguedelha, 2015-04-21:
I wrote a document with the detailed analysis (debug setup in Xcode, following the function call tree, the thread execution...). I'll post a link here very soon.

francesco-romano, 2015-04-21:
I think now we need advice from  @paulfitz also.....

barbalberto, 2015-04-21:
To recap the analysis done up to now, it seems that at a certain point the clock plugin stops to broadcast the time and this 'freezes' some thread preventing them to go on as they should, right? 

If this is right, I have a couple of questions:
- If the clock plugin keeps working, removing the robot will work?
- Why is the clock stopped when we remove the robot? Can we move the clock plugin from being a robot-plugin to be a system-plugin, loaded immediatly as gazebo starts (before the load of the robot) and removed when gazebo is closed (after the removal of the robot)?
This should make sure that the clock is always on during all the removal process.

What do you think?

lornat75, 2015-04-21:
I agree with @barbalberto that an elegant possibility could be to stop the clock at the very end of the shutdown. another way if possible is to close all wrappers explicitly, but this requires intercepting the shutdown signal and handle it explicitly.

The other solutions seem a bit dirty bc they change the time flow abruptly.

francesco-romano, 2015-04-21:
I think @nunoguedelha can answer better than be and with a lot of details.. 
Anyway to answer you both:
the clock plugin is a system plugin, thus is terminated at the very end of gazebo.
The issue is that while removing a robot, gazebo is waiting. But at the same time the robot plugin is blocked in some time-related stuffs, which are themselves waiting for the time thus creating the deadlock.

nunoguedelha, 2015-04-21:
And actually...
=> I am launching gazebo with the clock plugin, without loading any robot
=> I checked that the gazebo clock is being broadcasted by the clock plugin
=> After that, I insert the robot and remove it and the deadlock occurs.

The clock plugin method `GazeboYarpClock::clockUpdate()` is not triggered by gazebo as it should be just because gazebo is waiting. And even anyway, the gazebo timing is freezed.

So no component outside of gazebo is stoping the clock...

I uploaded here the details of the analysis:
https://github.com/nunoguedelha/IssuesAnalysis/tree/master/issue%23181_clockHangAfterRobotRemoval
(rtf format)
I'm converting it to markdown format.

nunoguedelha, 2015-04-21:
@barbalberto if I hack the clock plugin for keeping on broadcasting a fake clock (after deadlock), the removal works and everything is unblocked. We can insert a robot again. But issue is still reproducible if you remove again the robot of course.
 
@lornat75 the idea about switching the clock was actually to avoid breaking the time flow. We can switch a clock in a way that the timing value will be continuous, while the clock frequency will change.
But, as Francesco mentioned in an offline discussion, it could be tricky to handle if several robots are in the simulation.

This leads me to think that in a realistic simulation where we want to simulate several robots and be able to instantaneously remove one of them :  
- we should have an independent YARP server **instance** for each robot model inserted in the simulated world
- for removing a model, gazebo should  
=> freeze the simulation time, affecting every element in the world including every yard server instance  
=> switch the clock of the YARP instance (instance linked to the model targeted for deletion) to system clock  
=> remove the robot (respective yarp server instance could then be stopped)  
=> unfreeze the simulation time.

paulfitz, 2015-04-21:
Not sure if this is relevant, but the clock seen by a yarp-using program can be changed via the api, see http://wiki.icub.org/yarpdoc/classyarp_1_1os_1_1Time.html -- options are system time, network time, and a custom time source.

nunoguedelha, 2015-04-21:
Yes sure. I wasn't sure this wasn't just for YARP internal use. So the variable `pclock` tells if the clock comes from the `/clock` port or from the system, and `t` gives the time value in the first case.
Are these variables shared by all the threads of all models or just shared by the threads of one single model. The second option would be preferred: these clock information would then be integrated in the context instance linked to the model (I suppose there is one).

traversaro, 2015-04-22:
@nunoguedelha Thanks for the nice report!

For the information everyone gathered, given that having a different instances of yarp "instance" for model as @nunoguedelha proposed would be quite tricky , an easier solution in two steps would be:
 * **For solving the deadlock in `ServerInertial` delay:** As far as I understand, for some reason the `RateThread` is not affected by this problem. Perhaps we could migrate `ServerInertial` to `RateThread`, to avoid the `delay` in the custom periodic thread implementation currently present in `ServerInertial`.. cc @barbalberto what is your opinion on this?

* **To avoid the deadlock in `Stamp::Update` in analogWrapper (and perhaps also in `ServerInertial`? ): we should implement the [`IPrecislyTimed`](http://wiki.icub.org/yarpdoc/classyarp_1_1dev_1_1IPreciselyTimed.html) interface (i.e. the interface to provide timestamp information directly from the wrapper device) and ensure that the wrapper use that information if available, as the controlboardwrapper does. 

We this relatively simple fixes we should solve this issue.

nunoguedelha, 2015-04-22:
@traversaro Thanks!. And btw, I posted in my personal repo (for now) the previously mentioned detailed analysis, but this time in markdown format:
https://github.com/nunoguedelha/IssuesAnalysis/wiki/ISSUE-%23181-analysis

Are you sure avoiding this `delay` is enough? I would think it isn't because after `ServerInertial` returns from `run()`, it calls `threadRelease()` which reads the timing again. Actually we could potentially have other issues anywhere else as the YARP clock is freezed.

Lets see what @barbalberto says...

Tobias-Fischer, 2016-07-06:
I think this is fixed with https://github.com/robotology/gazebo-yarp-plugins/commit/b6bcb2f52afa043108c603e6f647db8f965158cc and https://github.com/robotology/yarp/pull/814
@nunoguedelha @barbalberto @traversaro 

At least I can remove and re-add the iCub without any issues in Gazebo.

traversaro, 2016-07-06:
You are launching gazebo with `YARP_CLOCK="/clock" gazebo -slibgazebo_yarp_clock.so` ? 
However those commits does not seems to be related to the issue originally described here, but it could have been similar problems that have been also solved.

Tobias-Fischer, 2016-07-07:
@traversaro sorry, should have read this properly. No, I wasn't launching Gazebo with the clock stuff, I thought it's the default behavior by now.
Interestingly, when I run gazebo with `YARP_CLOCK="/clock" gazebo -slibgazebo_yarp_clock.so`, it just gets stuck at a line looking like `[INFO]Read [97099] frames in 5[s], average period 0.05[ms], min 0.00[ms], max 3.00[ms]` without any messages of the IMU/AnalogWrapper/whatsoever drivers mentioned above.

traversaro, 2016-07-07:
Then probably there is a similar problem in the `camera` device, that was not existing (or it was not added) at the time.