robotology/gazebo-yarp-plugins #187: File reorganization

traversaro, 2015-04-30:
We recently experienced with @francesco-romano  that the file structure in this repo is quite confusing, especially for a new user that want to create/edit a plugin. 

We were then thinking to migrate to a structure where every plugin has its own directory with source code and every other resource (thrift) required. This should simplify a lot the structure of the repo. 

Given that there are no open PR, and given that the `addObjects` branch would anyway need a serious manual intervention to merge in master, I think now is a good time to proceed with this change. 

Any opinions ? 

cc @barbalberto @EnricoMingo @arocchi  @MirkoFerrati @jeljaik @prashanthr05

barbalberto, 2015-04-30:
Fine by me. If the split between gazebo_plugin and yarp_driver folders will be removed is a plus for me.

traversaro, 2015-04-30:
Exactly!

jeljaik, 2015-04-30:
totally agree

MirkoFerrati, 2015-04-30:
Fine. But don't think that addObjects requires that much effort to be
merged! The main effort is due to change of filenames and variable names!
I will merge in June if it's fine for you :)

traversaro, 2015-05-01:
Yes, for the problems of merging 'addObjects' I was referring to the workaround committed for the clock issue.

francesco-romano, 2015-05-05:
There are still some issues
- [ ] `yarp_idl`: when it autogenerates the files it does not "respect" the namespace convention: a header file with namespace `gazebo` is not saved in filesystem as `include\gazebo\file.h`. This creates issues then in the inclusion. Maybe we can open an issue in yarp
- [ ] I started to use `target_include_directories` instead of exporting global variables. The only question here is: what is the priority for the headers? build or install dir?
- [ ] I created a separate library for the thrift (currently only the clock, but the logic remains the same): this should allow clients to properly link to the autogenerated library instead of copy-paste sources (see https://github.com/robotology-playground/WBI-Toolbox for the `ySynchronizer` block). But probably I'm missing the `config.cmake` file.
- [ ] Tests should be enabled again (but anyway what is the status of the tests?)

francesco-romano, 2015-05-05:
For the first point I opened an issue in yarp: https://github.com/robotology/yarp/issues/487

traversaro, 2018-01-08:
Closing because: 
* For the first point a issue is open in YARP ( https://github.com/robotology/yarp/issues/487 )
* The second point is more of a question: in general the build include directory should have the priority to avoid conflicts if one of the upstream dependencies share the install directory with `gazebo-yarp-plugins` . If this is not the case, we will open issues to solve this problem. 
* The clock thrift logic should be moved to YARP, because it is used in several indipendent projects. I will open an issue in YARP for this. 
* RTF-based tests are running in Travis, but are not integrated in CTest. However, the old tests where more ad-hoc code to manually check the plugin, rather then proper tests.

closed 2018-01-08 by traversaro