robotology/gazebo-yarp-plugins #189: Cmake error after pulling repo

rlober, 2015-05-05:
Hey guys @francesco-romano @traversaro,

I was doing my bi-weekly update of the icub repos and after I pulled `gazebo-yarp-plugins` and tried running `cmake ..` from my build dir I get the following output/error:

```bash
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins/build$ cmake ..
-- Boost version: 1.46.1
-- Found PROTOBUF: /usr/lib/libprotobuf.so  
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'OGRE'
--   found OGRE, version 1.7.4
-- Found Ogre Cthugha (1.7.4)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Boost version: 1.46.1
-- Found the following Boost libraries:
--   serialization
--   system
-- Looking for OGRE...
-- Found Ogre Cthugha (1.7.4)
-- Looking for OGRE_Paging...
-- Looking for OGRE_Terrain...
-- Looking for OGRE_Property...
-- Looking for OGRE_RTShaderSystem...
-- checking for module 'gazebo'
--   found gazebo, version 4.1.1
-- checking for module 'sdformat'
--   found sdformat, version 2.2.0
-- checking for module 'protobuf'
--   found protobuf, version 2.4.1
-- Not processing thrift file clock_rpc.thrift, ALLOW_IDL_GENERATION=OFF
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr
Aborted (core dumped)

```

Ryan

traversaro, 2015-05-05:
O_o 
CMake version? Have you tried erasing the old build directory?

rlober, 2015-05-05:
```bash
ryan@ISIR2014057:~/Code/tutorials-icub/helloWorld/build$ cmake --version
cmake version 2.8.11.2
```
and Yes, I did erase the old build and start from scratch. When I try run `cmake ..` or ccmake ..` I get the same errors.

traversaro, 2015-05-05:
I guess it is on Ubuntu 12.04 . I will try to find a 12.04 box to replicate.

rlober, 2015-05-05:
Thanks man.

rlober, 2015-05-05:
Just an update...

I checked out a previous commit 0c4fa62caf7ffb1428860f5c7bc9a2deed371020 and this fixed the problem.

traversaro, 2015-05-05:
Mhh.. two things. 
Can you checkout the current master, after PR https://github.com/robotology/gazebo-yarp-plugins/pull/190 and try? 
If it is still not working, can you try post the output of the following commands:
~~~
which cmake
ldd `which cmake`
which pkg-config 
ldd `which pkg-config`
~~~

rlober, 2015-05-05:
Still not working... all the ouput is below:

```bash
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins$ git checkout master
Previous HEAD position was 0c4fa62... Reverted travis.xml
Switched to branch 'master'
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins$ git pull
remote: Counting objects: 1, done.
remote: Total 1 (delta 0), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (1/1), done.
From https://github.com/robotology/gazebo-yarp-plugins
   fca8ce2..6bdb463  master     -> origin/master
Updating fca8ce2..6bdb463
Fast-forward
 CMakeLists.txt                                  |   16 +++++++---------
 plugins/controlboard/src/ControlBoardDriver.cpp |    4 ++++
 2 files changed, 11 insertions(+), 9 deletions(-)
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins$ rm -rf build/
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins$ mkdir build
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins$ cd build/
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins/build$ cmake ..
-- The C compiler identification is GNU 4.6.3
-- The CXX compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Boost version: 1.46.1
-- Found the following Boost libraries:
--   serialization
--   system
-- Found PROTOBUF: /usr/lib/libprotobuf.so  
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'OGRE'
--   found OGRE, version 1.7.4
-- Found Ogre Cthugha (1.7.4)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Paging...
-- Looking for OGRE_Terrain...
-- Looking for OGRE_Property...
-- Looking for OGRE_RTShaderSystem...
-- Not processing thrift file clock_rpc.thrift, ALLOW_IDL_GENERATION=OFF
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr
Aborted (core dumped)
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins/build$ which cmake
/usr/bin/cmake
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins/build$ ldd `which cmake`
	linux-vdso.so.1 =>  (0x00007fffa877c000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007ff344cf9000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007ff344ae1000)
	libarchive.so.12 => /usr/lib/x86_64-linux-gnu/libarchive.so.12 (0x00007ff344857000)
	libcurl.so.4 => /usr/lib/x86_64-linux-gnu/libcurl.so.4 (0x00007ff3445fa000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ff3442f9000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ff3440e3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff343d24000)
[... 29 lines cut ...]
	libhcrypto.so.4 => /usr/lib/x86_64-linux-gnu/libhcrypto.so.4 (0x00007ff33f903000)
	libroken.so.18 => /usr/lib/x86_64-linux-gnu/libroken.so.18 (0x00007ff33f6ee000)
	libtasn1.so.3 => /usr/lib/x86_64-linux-gnu/libtasn1.so.3 (0x00007ff33f4dd000)
	libp11-kit.so.0 => /usr/lib/x86_64-linux-gnu/libp11-kit.so.0 (0x00007ff33f2ca000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007ff33f0c6000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007ff33eec2000)
	libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0 (0x00007ff33ec98000)
	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007ff33ea89000)
	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007ff33e83f000)
	libsqlite3.so.0 => /usr/lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007ff33e59b000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007ff33e362000)
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins/build$ which pkg-config
/usr/bin/pkg-config
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins/build$ ldd `which pkg-config`
	linux-vdso.so.1 =>  (0x00007fffc3110000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f7ea6349000)
	libpopt.so.0 => /lib/x86_64-linux-gnu/libpopt.so.0 (0x00007f7ea613c000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f7ea5d7d000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f7ea5b40000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f7ea5922000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f7ea571a000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f7ea6671000)
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins/build$ 

```

traversaro, 2015-05-05:
Mh... we where able to replicate. For the meanwhile, can you disable the `kalakris` ppa that you use for CMake 2.8.11 and get instead CMake 2.8.12 from the robotology ppa https://launchpad.net/%7Erobotology/+archive/ubuntu/ppa , as suggested in https://github.com/robotology/codyco-superbuild/#system-dependencies-2 ?

francesco-romano, 2015-05-05:
Hi @rlober 
Can you try after 9c12ed9b9616fdd516fc88d2ead63f711574fc4c ?

rlober, 2015-05-05:
@francesco-romano 
Worked like a charm! Thanks guys!

:+1:

closed 2015-05-05 by rlober