robotology/gazebo-yarp-plugins #191: Interfaces no longer open on model insert

rlober, 2015-05-11:
So I tried to open up the icub models in Gazebo (after doing a fresh update of `yarp`, `icub-main`, `icub-gazebo`, `gazebo-yarp-plugins` and `codyco-superbuild`) and none of the ports/interfaces open up as they did a week ago.

I first run `yarpserver`. Here is the output:
```bash
ryan@ISIR2014057:~$ yarpserver
__   __ _    ____  ____  
\ \ / // \  |  _ \|  _ \ 
 \ V // _ \ | |_) | |_) |
  | |/ ___ \|  _ <|  __/ 
  |_/_/   \_\_| \_\_|    

Call with --help for information on available options
Options can be set on command line or in /home/ryan/.config/yarp/yarpserver.conf
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
yarp: Port /ryan active at tcp://134.157.18.48:10000

Registering name server with itself:
 * register "/ryan" tcp "134.157.18.48" 10000
   + set "/ryan" ips "127.0.0.1" "134.157.18.48" "::1" "fe80::3664:a9ff:fec6:e09%2"
   + set "/ryan" process 25917
 * register fallback mcast "224.2.1.1" 10000
   + set fallback ips "127.0.0.1" "134.157.18.48" "::1" "fe80::3664:a9ff:fec6:e09%2"
   + set fallback process 25917
Name server can be browsed at http://134.157.18.48:10000/

Ok.  Ready!
```

Then I open gazebo in a separate terminal using simply `gazebo`.

Finally when I insert an icub model into the scene, nothing happens and the robot just falls because no interfaces are opened up.

Here is a screen shot:
[image]

Am I doing something wrong or is this a bug?

Thanks,
Ryan

traversaro, 2015-05-11:
This is probably due to the change of the filesystem structure in https://github.com/robotology/gazebo-yarp-plugins/pull/188#issuecomment-98607288 , now the plugins should be used from the **install** directory, and the `GAZEBO_PLUGIN_PATH` environmental variable should be modified accordingly.

rlober, 2015-05-11:
Ok. That fixed my problem but there is a little particularity... When I do make install it puts the libs in `.../installDir/lib/x86_64-linux-gnu/`. This is by no means a problem but it is a bit odd that a new subdirectory is created based on the system architecture. Is this on purpose? If not it would be good to remove this extra folder to avoid confusion.

traversaro, 2015-05-11:
@francesco-romano modified the CMake install procedure to use the directories defined in http://www.cmake.org/cmake/help/v3.0/module/GNUInstallDirs.html , that add the `x86_64-linux-gnu` suffix to the canonical `lib` install directory by default. Given that I agree with you that that could cause confusion, we should revert the default behaviour to the "classical" one.

rlober, 2015-05-11:
Damn you @francesco-romano !!!! Just kidding :smile: 

I guess we can close this issue. Let me know if y'all make the change.

traversaro, 2015-05-12:
cc @rlober  with this commit : https://github.com/robotology/gazebo-yarp-plugins/commit/4ea4cdfa2804f1ca666acb81a82a3185cae72068 the libs should go in `.../installDir/lib/`. Take care to change `GAZEBO_PLUGIN_PATH` on the next update!

rlober, 2015-05-15:
Hey guys,

Bad news... I just built and installed the new libs and as you guys indicated the install goes to `/installDir/lib`, but the model doesn't connect to yarp on insert (as was my initial problem in this issue). 

So I deleted the `x64...` dir and now have this install structure:
```bash
ryan@ISIR2014057:~/Code/iCub_Software/gazebo-yarp-plugins/install$ ls -l -R
.:
total 8
drwxrwxr-x 3 ryan ryan 4096 May 15 11:08 include
drwxrwxr-x 2 ryan ryan 4096 May 15 11:08 lib

./include:
total 4
drwxrwxr-x 2 ryan ryan 4096 May 15 11:08 gazebo

./include/gazebo:
total 4
-rw-r--r-- 1 ryan ryan 1940 May  5 17:58 ClockServer.h

./lib:
total 3584
-rw-r--r-- 1 ryan ryan 354082 May 15 10:57 libgazebo_yarp_camera.so
-rw-r--r-- 1 ryan ryan 293545 May 15 10:57 libgazebo_yarp_clock.so
-rw-r--r-- 1 ryan ryan 761818 May 15 10:57 libgazebo_yarp_controlboard.so
-rw-r--r-- 1 ryan ryan 492365 May 15 10:57 libgazebo_yarp_externalwrench.so
-rw-r--r-- 1 ryan ryan 311082 May 15 10:57 libgazebo_yarp_forcetorque.so
-rw-r--r-- 1 ryan ryan 305874 May 15 10:57 libgazebo_yarp_imu.so
-rw-r--r-- 1 ryan ryan 399167 May 15 10:57 libgazebo_yarp_jointsensors.so
-rw-r--r-- 1 ryan ryan 102474 May 15 10:57 libgazebo_yarp_rpc_clock.so
-rw-r--r-- 1 ryan ryan 400677 May 15 10:57 libgazebo_yarp_showmodelcom.so
-rw-r--r-- 1 ryan ryan 229833 May 15 10:57 libgazebo_yarp_singleton.so

```
with the following environment variables:
```bash
export ROBOT_CODE=/home/ryan/Code/iCub_Software
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:$ROBOT_CODE/icub-gazebo
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:$ROBOT_CODE/gazebo-yarp-plugins/install/lib
```

So, the first thing that strikes me is the lack of header files in the `include/` dir... I would have thought there would be many more? The libs all seem to be there but their headers - not so much.

francesco-romano, 2015-05-15:
Hi @rlober .
About the headers: this is normal. The libraries built are not shared libraries (well.. they should not be) but modules. This means that they are thought to be loaded at runtime, and not linked by another application during compilation. This is why there are no headers exported.

Can you try to launch gazebo with the `--verbose` option? This usually gives us some more details about the plugins it is loading. Thanks

rlober, 2015-05-15:
@francesco-romano 

```
ryan@ISIR2014057:~$ gazebo --verbose
Gazebo multi-robot simulator, version 4.1.1
Copyright (C) 2012-2014 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 134.157.18.48
Gazebo multi-robot simulator, version 4.1.1
Copyright (C) 2012-2014 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 134.157.18.48
[Wrn] [gazebo.cc:186] Waited 1seconds for namespaces.
```

On model insert...
```
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_imu.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:151] Failed to load plugin /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory

```

traversaro, 2015-05-15:
Add `$ROBOT_CODE/gazebo-yarp-plugins/install/lib` to the `LD_LIBRARY_PATH` enviromental variable and then run `sudo ldconfig`. 
Alternatively you can put a configuration file in the `/etc/ld.so.conf.d/` directory.

traversaro, 2015-05-15:
Apparently it was a bug (https://github.com/robotology/gazebo-yarp-plugins/issues/99) that should have been solved by RPATH support.
Before adding `$ROBOT_CODE/gazebo-yarp-plugins/install/lib` to the `LD_LIBRARY_PATH`, can you copy&paste the output of:
```
ldd /home/ryan/Code/iCub_Software/gazebo-yarp-plugins/install/lib/libgazebo_yarp_imu.so
```

francesco-romano, 2015-05-15:
Hi @rlober 
I`ve just updated the code cleaning up a bit the RPATH stuff.
Can you try again now?

In case it still does not work can you paste the output of:
~~~bash
cd your install dir
objdump -x libgazebo_yarp_controlboard.so | grep RPATH
ldd libgazebo_yarp_controlboard.so | grep singleton

 gazebo --verbose

~~~

rlober, 2015-05-15:
Hey guys, sorry for the slow reply, had to get some food!

So, 8c5cab7bffc86ddc86f381d4f411d56a3d55d241 has fixed the problem. I pulled, built, and installed and everything went ok.

Just in case you are interested here is some output:
```
ryan@ISIR2014057:~$ gazebo --verbose
Gazebo multi-robot simulator, version 4.1.1
Copyright (C) 2012-2014 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 134.157.18.48
Gazebo multi-robot simulator, version 4.1.1
Copyright (C) 2012-2014 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 134.157.18.48
[Wrn] [gazebo.cc:186] Waited 1seconds for namespaces.
*** GazeboYarpIMU plugin started ***
Found yarpConfigurationFile: loading from /home/ryan/Code/iCub_Software/icub-gazebo/icub/conf/gazebo_icub_inertial.ini
Calling GazeboYarpPlugins::Handler Constructor
GazeboYarpPlugins::Handler: Inserting Sensor : default::iCub_fixed::iCub::head::imu_sensor
Singleton: Added a new sensor default::iCub_fixed::iCub::head::imu_sensor.
covariance size is 9
Subdevice gazebo_imu
GazeboYarpIMUDriver is looking for sensor default::iCub_fixed::iCub::head::imu_sensor...
Looking for sensor : default::iCub_fixed::iCub::head::imu_sensor
Sensor default::iCub_fixed::iCub::head::imu_sensor was happily found!
[INFO]created device <gazebo_imu>. See C++ class GazeboYarpIMUDriver for documentation.
[INFO]No ROS group found in config file ... skipping ROS initialization. 
yarp: Port /icubGazeboSim/inertial active at tcp://134.157.18.48:10078
[INFO]Server Inertial : no ROS initialization required 
Server Inertial starting
[INFO]created wrapper <inertial>. See C++ class yarp::dev::ServerInertial for documentation.
Writing an Inertial measurement.
Loaded GazeboYarpIMU Plugin correctly
GazeboYarpIMU original parameters
(period 0.01) (sensorScopedName "default::iCub_fixed::iCub::head::imu_sensor") (name "/icubGazeboSim/inertial") (device inertial) (gazeboYarpPluginsRobotName icubGazeboSim) (subdevice gazebo_imu)
GazeboYarpIMU getOptions
device wrapped period subdevice subdevice.device subdevice.wrapped subdevice.sensorScopedName ROS name
*** GazeboYarpForceTorque plugin started ***
Found yarpConfigurationFile: loading from /home/ryan/Code/iCub_Software/icub-gazebo/icub/conf/FT/gazebo_icub_left_leg_ft.ini
GazeboYarpPlugins::Handler: Inserting Sensor : default::iCub_fixed::iCub::l_leg_ft_sensor::left_leg_ft
Singleton: Added a new sensor default::iCub_fixed::iCub::l_leg_ft_sensor::left_leg_ft.
[WARNING] AnalogServer device:

```

closed 2015-05-15 by rlober