robotology/gazebo-yarp-plugins #193: [controlboard] checkMotionDone seem to be broken

traversaro, 2015-05-14:
The iCub gazebo model is not passing the test in https://github.com/robotology/icub-tests/tree/master/src/motor-tests .

towardthesea, 2016-12-02:
Hi @traversaro 
I got this following problem:
````
icub-tests/suits (motorTest_checkMotionDone)*$ testrunner -v -s motors-icubGazeboSim.xml 
Loading motors-icubGazeboSim.xml
Staring test runner.
Test suit Motor Control Suite started...
[INFO]  (Motor Control Suite) reports: yarpmanager is setuping the fixture...
[ERROR] (Motor Control Suite) asserts error on (ret) with exception: yarpmanager (run) cannot setup the fixture because Failed to run gzserver on localhost. (State: 5, parameter: world/icub_fixed.world) 
[INFO]  (Motor Control Suite) reports: yarpmanager is tearing down the fixture...
Test suit Motor Control Suite failed!
Ending test runner.

---------- results -----------
Total number of test suites  : 1
Number of passed test suites : 0
Number of failed test suites : 1
Total number of test cases   : 0
Number of passed test cases  : 0
Number of failed test cases  : 0
````
It seems to get stuck as loading `gzserver`, but when I tried to run `gzserver` from `terminal` it could run well.

traversaro, 2016-12-02:
The problem is that it is the fixture is failing to run `gzserver world/icub_fixed.world`. 
There are two things that can cause this:
* `GAZEBO_RESOURCE_PATH` should point to the `icub-gazebo` directory so that gazebo can find automatically the requested world (this is not properly documented in `icub-gazebo`, and should be fixed.
* The directory of icub-gazebo containing the worlds have been recently renamed to `worlds`, and the fixture have been updated in `icub-tests` master, see https://github.com/robotology/icub-tests/blob/master/suits/fixtures/icubgazebosim-fixture.xml#L11 . Do you have the updated fixture.

To see an example of a working installation of icub-tests with Gazebo, please check `gazebo-yarp-plugins` Travis script in https://github.com/robotology/gazebo-yarp-plugins/blob/master/.travis.yml#L45 and https://github.com/robotology/gazebo-yarp-plugins/blob/master/.travis.yml#L92 .