robotology/gazebo-yarp-plugins #195: ApplyExternalWrench has been removed from icub.sdf

jeljaik, 2015-06-03:
Can we have the rationale behind this please? I know there was a rearrangement of the repo.

traversaro, 2015-06-04:
It is probably my fault, when I updated the model in `icub-gazebo` (commit [1]) to reflect the latest change in model generation, I did not add to the generator the option to add the `ApplyExternalWrench` to the model. We can simply add it back to the model, this was not intentional. 

crossref to the issue of iCub model generation: https://github.com/robotology/codyco-superbuild/issues/55 . 

[1] : https://github.com/robotology-playground/icub-gazebo/commit/512478610e79a4a40a48fd190a5157631098640a

jeljaik, 2015-06-04:
Ok I was thinking that it had something to do with some new native "apply external wrench" from the new Gazebo. Thought maybe we were already there :)

jeljaik, 2015-09-04:
@traversaro I have been contacted by Ryan Lober about the plugin. Can I enable it by default? or how should we treat this plugin from now on?

traversaro, 2015-09-04:
I guess it is ok to enable it, feel free to add it.

rlober, 2015-09-09:
@jeljaik @traversaro 

Hey guys! Any news on this one? @jeljaik thanks for asking about this for me!

mingxing-liu, 2015-09-14:
@jeljaik @traversaro 
Hi. my ubuntu version restricted me to gazebo 4, which does not provide the native "apply external wrench" as the newest gazebo does. Is it possible to continue using your apply external wrench plugin?

closed 2015-09-14 by traversaro