robotology/gazebo-yarp-plugins #198: Support for prismatic joints

traversaro, 2015-07-22:
Currently in `gazebo_yarp_controlboard` in several parts of the codebase we do the assumption that the controlled joints are revolute (for example in radians/degrees conversions). 
For simulating prismatic motors (as in the cer) we should consider the case of when the controlled joint is prismatic.

traversaro, 2015-09-28:
In particular, all the occurrences of the `Degrees()` [1] method , `RadiansToDegrees` function [2] and `DegreesToRadians` function [3] should be checked. 

[1] : https://github.com/robotology/gazebo-yarp-plugins/search?utf8=%E2%9C%93&q=Degree%28%29
[2] : https://github.com/robotology/gazebo-yarp-plugins/search?utf8=%E2%9C%93&q=convertRadiansToDegrees&type=Code
[3] : https://github.com/robotology/gazebo-yarp-plugins/search?utf8=%E2%9C%93&q=convertDegreesToRadians&type=Code

barbalberto, 2015-11-11:
I'm tackling this issue for CER sake.
I've fixed controlboardDriver but why you mentioned also IMU in the 2nd link? I guess it should not be related with prismatic joints.
Aldo, can you remind me what is the purpose of a jointSensor and if it is affected by revoulte/prismatic joint type?

traversaro, 2015-11-11:
In the previous comment I just searched all the occurrences of the `RadiansToDegrees` function that needs to be checked. In the context of converting the euler angles broadcast by the IMU its use is ok, and nothing should be changed. 

`jointSensor` was used to stream the joint velocities and torques when the controlboard was only streaming joint position, to mimic the behavior of the Coman robot. If those ports are not open anymore by coman/walkman, I would like to delete the plugin to reduce our maintenance burder. 
cc @liesrock @arocchi @EnricoMingo .

barbalberto, 2015-11-11:
ok, understood. :+1:  for deleting it

barbalberto, 2015-11-11:
implemented support for prismatic in commit 2f68a8d

TODO: remove jointSensor

traversaro, 2015-11-11:
Reset master to previous commit because the new version broke handling of gains. Please try to avoid pushing directly to master when critical parts of the code are modified.

traversaro, 2015-11-11:
Previous code save in `prismaticJoints` branch : https://github.com/robotology/gazebo-yarp-plugins/tree/prismaticJoints .

traversaro, 2015-11-11:
Fixed in https://github.com/robotology/gazebo-yarp-plugins/pull/211 .

closed 2015-11-11 by traversaro