robotology/gazebo-yarp-plugins #200: Reading of joint torques is failing

DanielePucci, 2015-08-03:
When reading joint torques through, e.g.

~~~
yarp read ... /icubGazeboSim/torso/stateExt:o 
~~~ 

I have the following output,

~~~
(0.401037 -0.000054 4.366778) [ok] (0.021704 0.04673 -0.001137) [ok] (0.0 0.0 0.0) [ok] (0.0 0.0 0.0) [fail] (0.0 0.0 0.0) [fail] (0.0 0.0 0.0) [fail] (0.0 0.0 0.0) [fail] (0.008931 0.038319 -8.621346) [ok] (7565168 7565168 7565168) [ok] (1718187123 1718187123 1718187123) [ok]
~~~

Then, it seems to me that the torque readings are failing.

barbalberto, 2015-08-03:
Just to spread some light on the output, the structure of the data is the following:
  1: list<double> jointPosition;              // joint encoder
  2: bool jointPosition_isValid;                // true/false, it tells if the previous array of value is valid and can be used or not
  3: list<double> jointVelocity;              
  4: bool jointVelocity_isValid;
  5: list<double> jointAcceleration;
  6: bool jointAcceleration_isValid;
  7: list<double> motorPosition;            // motor encored
  8: bool motorPosition_isValid;
  9: list<double> motorVelocity;         
  10: bool motorVelocity_isValid;
  11: list<double> motorAcceleration;
  12: bool motorAcceleration_isValid;
  13: list<double> torque;                    // joint torque
  14: bool torque_isValid;
  15: list<double> pidOutput;
  16: bool pidOutput_isValid;
  17: list<i32>      controlMode;          // position, velocity, torque ...
  18: bool controlMode_isValid;
  19: list<i32>      interactionMode;
  20: bool interactionMode_isValid;    // stiff o complaiant

The field of torques is all zeros and it says [fail] so it seems that the plugin is somehow failing to get the values. 
I'll try to check

traversaro, 2015-09-02:
Unable to reproduce, closing.

closed 2015-09-02 by traversaro