robotology/gazebo-yarp-plugins #210: gazebo_controlboard was found but could not open

kzts, 2015-11-11:
I make `*.ini` files and inclueded tags of `gazeno-yarp-plugins` in a `*.sdf` file.
But I got following error messages.
Could you tell me what I should change?

```
[ERROR]GazeboYarpControlBoardDriver::setJointNames(): cannot find joint    (  3  of  4  )  
[ERROR]jointNames is  jointNames r_shoulder_pitch r_shoulder_yaw r_shoulder_roll r_elbow_pitch 
 yarpdev: ***ERROR*** driver <gazebo_controlboard> was found but could not open
```

traversaro, 2015-11-11:
Do you have a joint named `r_shoulder_roll` in your SDF? The error is printed because the plugin is not able to find the `r_shoulder_roll` joint in your SDF model.

barbalberto, 2015-11-11:
From the message I see that jont number three cannot be found.
The following line tells you the joints the plugin is looking for into sdf file. 
In this case, the 3th joint over , namely r_shoulder_roll is not found in the sdf. This usually occours when there is a mismatch or typos in config files

I'll try to make the message more clear adding explixitly the name of the joint not found

kzts, 2015-11-11:
Thank you for answers!

I can found `r_shoulder_roll` in my sdf file as following.
Should I use this name other place?
```
    <joint name='r_shoulder_roll' type='revolute'>
      <child>r_humerous</child>
      <parent>r_shoulder_in_up</parent>
      ...
    </joint>
```

barbalberto, 2015-11-11:
check if all the previous / following tags are closed correctly, maybe it is hidden by a previous joint or link if a ending </link> or somthing is missing.

kzts, 2015-11-11:
OK! I'll check it.
I pasted all messages. It may help me to tell you what happened. 

```
*** GazeboYarpControlBoard plugin started ***
Calling GazeboYarpPlugins::Handler Constructor
GazeboYarpPlugins::Handler: Inserting Robot : ATHLETE_ROBOT_CY
Singleton: Added a new robot ATHLETE_ROBOT_CY.
GazeboYarpControlBoard: Found yarpConfigurationFile: loading from /home/k-tanaka/.gazebo/models/athletebot/conf/athletebot_right_arm.ini
[WARNING] ControlBoardWrapper2 device:
************************************************************************************
* ControlBoardWrapper2 'name' parameter for port name does not follow convention,  *
* it MUST start with a leading '/' since it is used as the full prefix port name   *
*     name:    full port prefix name with leading '/',  e.g.  /robotName/part/     *
* A temporary automatic fix will be done for you, but please fix your config file  *
************************************************************************************ 
[WARNING] *** ControlBoardWrapper2 is using DEPRECATED parameter 'threadrate', use 'period' instead *** 
[INFO]Athletebot/right_arm : no ROS initialization required 
[INFO]Athletebot/right_arm  initting YARP initialization 
yarp: Port /Athletebot/right_arm/rpc:i active at tcp://192.168.2.247:10002
yarp: Port /Athletebot/right_arm/command:i active at tcp://192.168.2.247:10003
yarp: Port /Athletebot/right_arm/state:o active at tcp://192.168.2.247:10004
yarp: Port /Athletebot/right_arm/stateExt:o active at tcp://192.168.2.247:10005
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
GazeboYarpControlBoard: wrapper opened correctly
Looking for device : ATHLETE_ROBOT_CY::right_arm
GazeboYarpControlBoard: setting robotScopedName ATHLETE_ROBOT_CY
Opening  GazeboYarpControlBoardDriver named right_arm
DeviceDriver is looking for robot ATHLETE_ROBOT_CY...
Looking for robot : ATHLETE_ROBOT_CY
Robot ATHLETE_ROBOT_CY was happily found!
Robot Name: ATHLETE_ROBOT_CY
# Joints: 16
# Links: 17
[ERROR]GazeboYarpControlBoardDriver::setJointNames(): cannot find joint    (  3  of  4  )  
 
[ERROR]jointNames is  jointNames r_shoulder_pitch r_shoulder_yaw r_shoulder_roll r_elbow_pitch 
 
yarpdev: ***ERROR*** driver <gazebo_controlboard> was found but could not open
controlBoard <right_arm> did not open!!
```

kzts, 2015-11-12:
I found a miss spell and the plugin worked happily.
Thank you!

traversaro, 2015-11-12:
Perfect! :+1:

closed 2015-11-12 by traversaro

iron76, 2015-11-12:
@kzts can you please provide me a link to the files that enable the plugins? I want to give it a try on gazebo.

barbalberto, 2015-11-12:
Also please fix the warning
```
[WARNING] ControlBoardWrapper2 device:
************************************************************************************
* ControlBoardWrapper2 'name' parameter for port name does not follow convention,  *
* it MUST start with a leading '/' since it is used as the full prefix port name   *
*     name:    full port prefix name with leading '/',  e.g.  /robotName/part/     *
* A temporary automatic fix will be done for you, but please fix your config file  *
************************************************************************************ 
```

As the message says, the parameter name should be filled with the FULLNAME of the port, starting with  ```\``` character, something like /robotName/part (or whatever you like)

kzts, 2015-11-12:
@iron76 I put the sdf on `https://github.com/kzts/athletebot`.
Please add a directory including stl files that I gave you to it.
Please try the file.
The plugin seemed to work but model appearance collapsed.

@barbalberto I don't understand the meaning of the word, "port" in "the parameter name should be filled with the FULLNAME of the port". Does "the port" mean a YARP port?

barbalberto, 2015-11-12:
Yes, it is the name of YARP port opened by the controlBoardWrapper device.

Suppose you want the wrapper to open the ports like
```
/myRobot/left_arm/command:i 
/myRobot/left_arm/rpc:i 
/myRobot/left_arm/state:o
/myRobot/left_arm/stateExt:o 
```

inside the .ini file, section wrapper should look like:
```
[WRAPPER]
device controlboardwrapper2
period 10
name /myRobot/left_arm
```

iron76, 2015-11-12:
@barbalberto. The problem seems to be also affecting our iCub-gazebo model which outputs the very same [WARNING]. I am not sure what the reason is (@traversaro might have an idea) but currently config files are structured as follows:

```
[WRAPPER]
# name of the wrapper device to be instatiated by the factory
device controlboardwrapper2
# rate of output streaming from ports in ms
threadrate 10
# output port name (a leading '/' will be added by code)
name ${gazeboYarpPluginsRobotName}/left_arm
# Total number of joints
joints 4
# list of MotorControl device to use
networks ( left_arm )
# for each network specify the joint map
left_arm  0 3 0 3
# Verbose output (on if present, off if commented out)
#verbose
```

iron76, 2015-11-12:
@kzts please have a look at https://github.com/kzts/athletebot/issues/2 and https://github.com/kzts/athletebot/issues/1. The robot seems to collapse because the PID values should be tuned.

traversaro, 2015-11-12:
Yes, I was aware of that warning. 
The fix is simply to add the `/` at beginning of the `name` parameter and change the `threadrate` to `period`. 
I did not change it when the YARP convention was changed because at that time I did not want to lose compatibility with older version of YARP, that could be used for example in people in the Walkman project. 
I think it is now safe to change it, but I have to check. I am working to have a model that (on standard non-failed run) does not print anything to terminal, see https://github.com/robotology/gazebo-yarp-plugins/pull/208 and https://github.com/robotology/gazebo-yarp-plugins/issues/87 .

kzts, 2015-11-13:
I fixed the warning and changed PID gain and avoided model collapsing!
Thank you!