robotology/gazebo-yarp-plugins #213: Remove joint sensors plugin

traversaro, 2015-11-12:
See https://github.com/robotology/gazebo-yarp-plugins/issues/198#issuecomment-155785047 . 

@arocchi  @liesrock @EnricoMingo Apparently the joint sensor plugin is still used by the coman model (see https://github.com/EnricoMingo/iit-coman-ros-pkg/blob/master/coman_urdf/urdf/gazebo/yarp_control.gazebo.xacro). I guess that before removing the sensor from GYP, we should remove it from the models.

arocchi, 2015-11-13:
Yes, let us figure this out, shouldn't be too complicated, maybe @liesrock @barbalberto can help. So, now we can have velocity/torque without the need for joint sensors?

traversaro, 2015-11-13:
Yes, they are part of the `stateExt:o` port published by the controlboardwrapper itself.

traversaro, 2016-11-08:
@liesrock Do you think this plugin can be removed by the models or they are still used?

liesrock, 2016-11-08:
Hi @traversaro, I think the sensor plugin can be removed since we have this information on the `stateExt:o` port. I already removed them in this commit 8baafd9b25bacdab255b33f9a2ba12be838c2271 , as soon as @EnricoMingo give me the ok i will merge the modification on master.

traversaro, 2016-11-08:
@liesrock Thanks a lot!

EnricoMingo, 2016-11-08:
Merged! I close the issue then.

traversaro, 2017-06-18:
Apparently, the plugin is still referenced in a few files in `ADVRHumanoids/iit-bigman-ros-pkg` https://github.com/ADVRHumanoids/iit-bigman-ros-pkg/search?utf8=%E2%9C%93&q=libgazebo_yarp_jointsensors.so&type= and in ` robotology/OpenSoT` https://github.com/robotology/OpenSoT/search?utf8=%E2%9C%93&q=libgazebo_yarp_jointsensors.so&type= . 
I guess this are leftovers fies, am I correct @liesrock @EnricoMingo ? 

I also noticed that the plugin is referenced in https://github.com/Greattc/schunk_description/blob/496e970a898da0b6c0704db89b31e5cd8aa97b9f/urdf/lwa/lwa.gazebo.xacro , a repo maintained by @Greattc , but I guess that again that repo is not maintained as well.

liesrock, 2017-06-19:
@enricomingo can answer from our side

traversaro, 2017-12-12:
Any news @liesrock @EnricoMingo ?

EnricoMingo, 2017-12-12:
Does something change in the model urdf as well as the .ini files???

traversaro, 2018-01-08:
Unless you use the `libgazebo_yarp_jointsensors.so` in one of your models you don't need to change anything.

EnricoMingo, 2018-01-08:
Actually, in bigman we use it for the joint feedback of velocity and
torque. Is everything now in controlboard?

traversaro, 2018-01-08:
Yes, all those information is now published by the `ControlBoardWrapper` in the `stateExt:o` port. All this information is read by the `RemoteControlBoard` on the client side using the usual `getEncoderVelocities` and `getTorques` methods.

EnricoMingo, 2018-01-08:
The related ini files can also be removed?

2018-01-08 11:58 GMT+01:00 Enrico Mingo <enricomingo@gmail.com>:

> Ok which means that I can remove the libgazebo_yarp_jointsensors.so from
> the bigman model. I am going to di it right now.
>
>
>> Yes, all those information is now published by the ControlBoardWrapper
>> in the stateExt:o port. All this information is read by the
>> RemoteControlBoard on the client side using the usual
>> getEncoderVelocities and getTorques methods.
>>
>>
>
>
>
> PhD Mingo Hoffman Enrico
> Istituto Italiano di Tecnologia, Genova

traversaro, 2018-01-08:
Yes!

EnricoMingo, 2018-01-08:
Ok thanks!

traversaro, 2021-09-17:
Fixed in https://github.com/robotology/gazebo-yarp-plugins/pull/574 .

closed 2021-09-17 by traversaro