robotology/gazebo-yarp-plugins #215: Error on using the plugin ApplyExternalWrench

iron76, 2015-11-13:
I am trying to use the plugin for applying external wrenches but on a robot which differs from the iCub. I am getting the following error. I assume that I need to define at least one link named root_link. Is this the case? Is this really necessary?
```
[ERROR] ApplyWrench plugin: link named root_linknot found
```

traversaro, 2015-11-13:
https://github.com/robotology/gazebo-yarp-plugins/blob/2f68a8ddd580e270a4fd8d9fc70828fc0a2ebece/plugins/externalwrench/src/ApplyExternalWrench.cc#L147 apparently the problem is in this line. Getting a link of the robot as the default link seems to be a better solution indeed.

iron76, 2015-11-14:
I leave this open until the problem is fixed (@jeljaik and @traversaro).

jeljaik, 2015-11-16:
Yes, I was actually aware. I'll be fixing soon.

jeljaik, 2015-11-16:
Fixed in a07d8c99a64520c9117b915f517b88c932bef078.

iron76, 2015-11-18:
I need to reopen the issue since the fix is not working when using the iCub fixed. The error I am getting is: 
```
ApplyWrench plugin: link named iCub::root_link not found
```