robotology/gazebo-yarp-plugins #218: PID parameters not found

carlos-cardoso, 2015-12-14:
With the following text file, we get a message that PID parameters were not found.

The file is as follows:

```
[include "gazebo_icub_robotname.ini"]

[WRAPPER]
# name of the wrapper device to be instatiated by the factory
device controlboardwrapper2
# rate of output streaming from ports in ms
threadrate 10
# output port name (a leading '/' will be added by code)
name ${gazeboYarpPluginsRobotName}/left_shoulder_arm
# Total number of joints
joints 8
# list of MotorControl device to use
networks ( left_shoulder_arm )
# for each network specify the joint map
left_shoulder_arm  0 7 0 7
# Verbose output (on if present, off if commented out)
#verbose

# Specify configuration of MotorControl devices
[left_shoulder_arm]
# name of the device to be instatiated by the factory
device gazebo_controlboard
#jointNames list
jointNames l_shoulder_scapula_joint l_shoulder_flection_joint l_shoulder_abduction_joint l_shoulder_rotation_joint l_elbow_flection_joint l_forearm_pronation_joint l_wrist_abduction_joint l_wrist_flection_joint
name left_shoulder_arm

#PIDs:
# this information is used to set the PID values in simulation for GAZEBO, we need only the first three values
[GAZEBO_PIDS]
#Left Arm
Pid0 500.0 2.0 0.1 1000 1300 9 9
Pid1 500.0 2.0 0.1 1000 1300 9 9
Pid2 500.0 2.0 0.1 1000 1300 9 9
Pid3 300.0 0.1 0.01 1000 1300 9 9
Pid4 300.0 0.1 0.01 1000 1300 9 9
Pid5 300.0 0.1 0.01 1000 1300 9 9
Pid6 300.0 0.1 0.01 1000 1300 9 9
Pid7 300.0 0.1 0.01 1000 1300 9 9

[GAZEBO_VELOCITY_PIDS]
#Left Arm
Pid0 500.0 2.0 0.1 9999 9999 9 9
Pid1 500.0 2.0 0.1 9999 9999 9 9
Pid2 500.0 2.0 0.1 9999 9999 9 9
Pid3 500.0 0.1 0.01 9999 9999 9 9
Pid4 500.0 0.1 0.01 9999 9999 9 9
Pid5 500.0 0.1 0.01 9999 9999 9 9
Pid6 500.0 0.1 0.01 9999 9999 9 9
Pid7 500.0 0.1 0.01 9999 9999 9 9
```

And this is the output is as follows:

```
GazeboYarpControlBoard: Found yarpConfigurationFile: loading from /home/vislab/indigo_ws/src/vizzy/vizzy_gazebo/media/conf/gazebo_vizzy_right_arm.ini
[WARNING] *** ControlBoardWrapper2 is using DEPRECATED parameter 'threadrate', use 'period' instead *** 
[INFO]/vizzySim/right_shoulder_arm : no ROS initialization required 
[INFO]/vizzySim/right_shoulder_arm  initting YARP initialization 
yarp: Port /vizzySim/right_shoulder_arm/rpc:i active at tcp://10.0.27.49:10014
yarp: Port /vizzySim/right_shoulder_arm/command:i active at tcp://10.0.27.49:10015
yarp: Port /vizzySim/right_shoulder_arm/state:o active at tcp://10.0.27.49:10016
yarp: Port /vizzySim/right_shoulder_arm/stateExt:o active at tcp://10.0.27.49:10017
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
No minimum stiffness value found in ini file, default one will be used!
No maximum stiffness value found in ini file, default one will be used!
No minimum dampings value found in ini file, default one will be used!
No maximum damping value found in ini file, default one will be used!
min_stiffness: [  0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000 ]
max_stiffness: [  1000.000000	 1000.000000	 1000.000000	 1000.000000	 1000.000000	 1000.000000	 1000.000000	 1000.000000 ]
min_damping: [  0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000 ]
max_damping: [  100.000000	 100.000000	 100.000000	 100.000000	 100.000000	 100.000000	 100.000000	 100.000000 ]
PID gain information not found in plugin parameters, using default gains ( P 500 I 0 D 1 )
INITIAL CONFIGURATION IS:  0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000	 0.000000
```

Are we missing something in the file?

Thanks in advance.

traversaro, 2015-12-14:
I think that is a misleading warning message because it refers to the gain of the impedance loop. I am currently on the cellphone, when I will get a PC I will check the source code.

carlos-cardoso, 2015-12-16:
It seems that it does load some of the parameters, but we don't understand why some values are changed or not loaded. As an example, we run the yarpmotorgui and checked the PID velocity and position values for joint 3 and compared to the ones in the .ini file (displayed on the left).

In the case of the position PID, it seems that the Kp is scaled. How is this computed?
In the case of the velocity PID, it looks like the Kd is not read properly.

[image]

[image]

traversaro, 2015-12-16:
I will provide a detailed reply on this as soon as possible, in the meanwhile you could be interested in this (still open : ( ) issue: https://github.com/robotology/gazebo-yarp-plugins/issues/119 .

traversaro, 2015-12-16:
Regarding the first message, I think that the code was complaining that the `GAZEBO_IMPEDANCE_POSITION_PIDS` group is missing [1], and hence was using the default value for the impedance loop (i.e. the one that you can activate by setting the `POSITION` control mode with the `COMPLIANT` interaction mode, see http://wiki.icub.org/wiki/Control_Modes ). Given that I don't think that the real Vizzy has torque feedback to implement impedance control, it don't think you use the `COMPLIANT` interaction mode even in simulation, so this is not a big problem. In https://github.com/robotology/gazebo-yarp-plugins/commit/cb65ac676df84b25586b12d2251366fee4d14728 I pushed a improved "warning" that I hope should prevent further misunderstandings. 

[1] : https://github.com/robotology/gazebo-yarp-plugins/blob/cb65ac676df84b25586b12d2251366fee4d14728/plugins/controlboard/src/ControlBoardDriver.cpp#L427
https://github.com/robotology/gazebo-yarp-plugins/commit/cb65ac676df84b25586b12d2251366fee4d14728

traversaro, 2015-12-16:
Regarding the second message. 

For the position control, the gains (for SDF `revolute` joints) in the configuration file have unit of measurements:

# .ini Configuration files revolute joint gains 
| Kp | Kd | Ki | 
|:---:|:----:|:---:|
| `Nm`/`[rad]` | `Nm`*`s`/`[rad]` | `Nm`/(`[rad]`*`s`)  |

# Revolute joint gains in YARP `IPidControl` interface methods
| Kp | Kd | Ki | 
|:---:|:----:|:---:|
| `Nm`/`deg` | `Nm`*`s`/`deg` | `Nm`/(`deg`*`s`)  |

This inconsistency is the subject of the issue https://github.com/robotology/gazebo-yarp-plugins/issues/119 . Introducing a way of setting gains in degrees from the `.ini` configuration file could solve this inconsistency. 

The `Kd` for the VELOCITY PID (actually PI) is not set because as the Gazebo low level velocity controller has the torque exerted on the joint, and the torque exerted on the joint has an instantaneous relation with the joint acceleration, having a derivative term would make the VELOCITY PID a non causal system.

randaz81, 2016-02-11:
Beware! You are using and obsolete format `[GAZEBO_PIDS]` that will be dropped soon.

This is the correct format, the same used on the real robot: 

```
[POSITION_CONTROL]
controlUnits  metric_units
controlLaw    joint_pid_gazebo_v1
kp           100.0 100.0 20.0  20.0  20.0
kd           0.1   0.1  0.1   0.1   0.1
ki           0.1   0.1  0.1   0.1   0.1
maxInt       9999  9999 9999  9999  9999
maxOutput    9999  9999 9999  9999  9999
shift        0     0    0     0     0
ko           0     0    0     0     0
stictionUp   0     0    0     0     0
stictionDwn  0     0    0     0     0

[VELOCITY_CONTROL]
controlUnits  metric_units
controlLaw    joint_pid_gazebo_v1
kp           20.0  20.0 20.0  20.0  20.0
kd           0.0   0.0  0.0   0.0   0.0
ki           0.01  0.01 0.01  0.01  0.01
maxInt       9999  9999 9999  9999  9999
maxOutput    9999  9999 9999  9999  9999
shift        0     0    0     0     0
ko           0     0    0     0     0
stictionUp   0     0    0     0     0
stictionDwn  0     0    0     0     0

[IMPEDANCE_CONTROL]
controlUnits  metric_units
controlLaw    joint_pid_gazebo_v1
stiffness    0.0   0.0   0.0   0.0   0.0
damping      0.0   0.0   0.0   0.0   0.0
```

* Currently `joint_pid_gazebo_v1` is the only valid choice for parameter `controlUnits`
* If you choose `metric_units`, gains will be expressed using `Nm/deg` (preferred choice when working icub) 
* If you choose `si_units`, gain will be expressed using `rad/deg`.

closed 2016-02-11 by randaz81

traversaro, 2016-03-02:
@randaz81 as noted by @francesco-romano , `si` and `metric system` are tipically used as synonims (at least I was not able to find a refence that the degrees are the angle unit for the metric system) can we use some other names for the `controlUnits` parameters?

randaz81, 2016-03-02:
yes, any suggestion?

traversaro, 2016-03-02:
By the way I just realized that probably I was there when `metric_units` was first used. :D I'll think about it.