robotology/gazebo-yarp-plugins #219: Clock plugin should be publicized on the main README.md

jeljaik, 2015-12-15:
Even though the Yarp clock synchronizer is an optional feature and plugin, I believe it is of the utmost importance for anyone running simulations at a low real time factor and it should be reported somewhere. As far as I know, its usage is only reported [here](https://github.com/robotology/gazebo-yarp-plugins/issues/57#issuecomment-34787381) and [here](https://github.com/robotology/gazebo-yarp-plugins/issues/57#issuecomment-34787381), a bit hard to find for new users. I believe this should be highlighted in the main Readme.

francesco-romano, 2015-12-15:
Now that the plugins have separate folder in the repository I think we should add a `README.md` [here](https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/clock) and reference it from the main readme.

jeljaik, 2015-12-15:
Well yeah, that makes total sense. Still I believe it would be nice to have at least a brief description of what each plugin does in the main Readme so that I know from the very first time I use iCub on Gazebo, that I have the option to synchronize the clock.