robotology/gazebo-yarp-plugins #226: Cease to support Gazebo 2.2 , 3 and 4 ?

traversaro, 2016-02-07:
Since last January, Gazebo 2.2 , 3 and 4 are not supported anymore by the OSRF (see http://gazebosim.org/#collapseVersion1_9 ). 

Given that making sure that the `gazebo-yarp-plugins` are compiling against all the old version of Gazebo is quite time consuming (it involves add a lot of `#ifdef` for whatever function changed signatures across versions) it may be convenient to drop support for older version of Gazebo.
Furthermore, older version of Gazebo has several well known issues https://github.com/robotology/gazebo-yarp-plugins#old-gazebo-versions so dropping support for them is a way of making sure that users use the newer versions of Gazebo. 

The current users I am aware of are: 
* iCub users are currently all using the latest version of Gazebo. 
* Walkman/Coman users are using currently using by default the Gazebo5 version with ROS Indigo (see http://gazebosim.org/tutorials?tut=ros_wrapper_versions) 
* Vizzy documentation (in https://github.com/vislab-tecnico-lisboa/vizzy#environment) claim that they try to be compatible with the default Gazebo versions in Hydro and Indigo (i.e. 1.9 and 2.2). In that case, dropping support for Gazebo 2.2 would break compilation for them. @plinioMoreno can you confirm that you have this requirement?

francesco-romano, 2016-02-07:
:+1: 

Or, if for Vizzy old gazebo versions are a must, we can branch out the code, and support only (critical) bug fixes to the old versions of the plugins (still time consuming, but less time consuming)

traversaro, 2016-02-07:
The problem of that option is that once in a while there are also modifications to `gazebo-yarp-plugins` to make sure that it still compiles fine against the latest yarp, and we would need to backport also those changes to the `vizzy` branch. If this is the case, until a proper release process is deployed for yarp I think that the least time consuming option is still to support old Gazebo version in the same codebase.

plinioMoreno, 2016-02-11:
Sorry for the late reply. About Vizzy, the gazebo 2 is just because of the simulation of the mobile platform and it is not related to the motor control of gazebo-yarp-plugins. In fact in the tests we did for the version-vergence controller we were using gazebo 5, because earlier versions do not have the Joint interfaces (external pointers) accessible outside of gazebo.

So from Vizzy side you can go ahead and drop support on the older gazebo versions, which is not needed for gazebo-yarp-plugins.

traversaro, 2016-02-11:
Thanks @plinioMoreno , we will procede to drop support for Gazebo version <= 4 .

traversaro, 2016-11-08:
This can be close when we enfore the use of Gazebo >= 5.0 at CMake time ( https://github.com/robotology/gazebo-yarp-plugins/issues/61 ).

closed 2016-12-21 by traversaro