robotology/gazebo-yarp-plugins #231: Problems in switching control mode

naveenoid, 2016-02-26:
Strange behaviour is being seen when switching control modes..I have tested from Position (using minimum jerk) to Position direct. From brief intervals some joints seem to go to idle or have some other such behaviour. The same joint doesnt go to idle each time either.The tests I ran were using the codyco-modules walkplayer.

traversaro, 2016-02-26:
I checked the control mode switching code and everything seems ok but perhaps I am missing something, fyi @randaz81 .

randaz81, 2016-03-01:
Maybe a race condition? Can you give me a pointer to the mentioned walkplayer module?

francesco-romano, 2016-03-02:
I replicated the issue when switching from torque to position mode.
The robot falls, like if the fault was pressed.

randaz81, 2016-03-02:
Does `Onupdate()` run on a different thread respect to the yarp interface methods? Probably yes. In this case weird things will happen if setControlMode() re-initializes the trajectory to the current joint position while OnUpdate() is executing. In this case the solution is to put a mutex to protect the execution of all yarp methods (not only setControlMode) vs OnUpdate.

traversaro, 2016-03-02:
Yep, OnUpdate is called by the physics thread.

randaz81, 2016-03-08:
@naveenoid any update after introducing the mutex?

naveenoid, 2016-03-09:
Testing today.

traversaro, 2016-03-17:
@naveenoid any updates?

naveenoid, 2016-03-20:
Im still having odd behaviour. For some reason, in between switching some joints are going to limits or very briefly idle. Im trying to investigate systematically. FYI Im investigating using this : https://github.com/robotology/codyco-modules/tree/feature/impedanceWalk/src/modules/walkPlayer/src

I will report back asap..need this fixed.

traversaro, 2016-03-24:
@jeljaik you told me it was fixed by https://github.com/robotology/gazebo-yarp-plugins/commit/c90da7ce9911e84f3b8214d938fa4b835887c948 , right?

jeljaik, 2016-03-25:
That commit was actually exclusively related to a misbehavior of the `InteractionMode`, which, just by chance, exhibited a similar behavior to the one described in this issue. Nevertheless, this might be something else and @naveenoid is supposed to test and see if  commit https://github.com/robotology/gazebo-yarp-plugins/commit/c90da7ce9911e84f3b8214d938fa4b835887c948 might have fixed something related to this issue.

naveenoid, 2016-03-31:
For now the control mode switching seems stable. I will close this issue until I test and see it again. I havent tested all kinds of mode switching, only between position and position direct with and without compliant Interaction mode. Maybe someone testing torque control switching can test and re-open this issue if needed.

closed 2016-03-31 by naveenoid