robotology/gazebo-yarp-plugins #237: CMakeLists says that gazebo is lower than 3.9

plinioMoreno, 2016-04-27:
but I installed the ros-indigo packages for gazebo 5 and gzserver reports:

```
gzserver --version
Gazebo multi-robot simulator, version 5.3.0
Copyright (C) 2012-2015 Open Source Robotics Foundation.

```

In fact I had gazebo 2 installed previously, but I removed it with `sudo apt-get remove --purge ros-indigo-gazebo*`

Should I delete files by hand? Which files should I check?

traversaro, 2016-04-27:
You could check if you still have a gazebo-config.cmake or gazeboConfig.cmake around.

plinioMoreno, 2016-04-29:
I checked and the correct procedure is to remove gazebo with `sudo apt-get remove --purge gazeboX*` and like that works properly and the warning does not show up anymore.

Thanks for the quick reply

closed 2016-04-29 by plinioMoreno