robotology/gazebo-yarp-plugins #240: Velocity joint control does not work properly

plinioMoreno, 2016-05-05:
@randaz81 and @traversaro I have been able to control joints in position and mixed mode, but in velocity mode the following behavior happens:

If the motion of the joint is affected by gravity (the motor axis is parallel to the world plane), the joint is not able to counteract gravity and moves towards the limit.
If the motion of the joint is not affected by gravity (the motor axis is parallel to the gravity vector), the joint oscillates between the minimum and maximum limits.

The parameters I'm using are as follows:

```
[VELOCITY_CONTROL]
controlUnits  metric_units
controlLaw    joint_pid_gazebo_v1
kp           20.0  20.0 20.0  20.0  20.0
kd           0.0   0.0  0.0   0.0   0.0
ki           0.01  0.01 0.01  0.01  0.01
maxInt       9999  9999 9999  9999  9999
maxOutput    9999  9999 9999  9999  9999
shift        0     0    0     0     0
ko           0     0    0     0     0
stictionUp   0     0    0     0     0
stictionDwn  0     0    0     0     0
```

And from the yarpmotorgui I get the following output:
```
[INFO]joint: 2 in VELOCITY mode
[INFO]Changing reference acceleration of joint  2  to 100000 
[INFO]joint: 2 in POSITION mode
[INFO]joint: 0 in VELOCITY mode
[INFO]Changing reference acceleration of joint  0  to 100000
```

Let me know if you have any suggestion, and I wonder why the acceleration is set so high

plinioMoreno, 2016-05-06:
Hi again,

Additional information about the tests I've done:

- I tested several PID values and the behaviour is always the same. Also played with the shift to see if I was scaling badly the parameters, but always get the same behavior.

- I am aiming at controlling Vizzy's [head motors](https://github.com/vislab-tecnico-lisboa/vizzy/tree/master/vizzy_description/urdf/head). All the files head.yarp.* are the ones loaded. I am using Gazebo 7.0.

- I tried also using the `simpleClient` executable, where I can control all the values (especially the acceleration that is so high when I use yarpmotorgui), but I get always the same behavior.

Any suggestion would be very appreciated.

traversaro, 2016-05-11:
Sorry for the late answer @plinioMoreno! 
As far I remember the velocity control on the Gazebo simulator uses directly the Gazebo JointController, that closes a velocity PID directly on the measured velocity, while the mixed mode and the velocity control on the real robot actually use a position loop underneath the velocity control loop. 

As far as I know, all the people that tried to use directly the velocity control in Gazebo report terrible performance (probably due to the architecture of the low level velocity control loop itself). 

I can look more in details of this after friday (again sorry for the late answer, but we have a strict deadline on friday).

plinioMoreno, 2016-05-11:
Thanks for the answer @traversaro. In fact I played for a while with several PID configurations and read several threads on similar issues when aiming at controlling joints with gazebo-ros control plugins and I got the idea that the velocity control does not work properly on gazebo.

I also added friction and damping coefficients. It looked like the velocity controller has a drift (not staying at the initial position), and when I commanded the head motors using the Vizzy's control gaze (that works on the real robot), the controller sent a very large torque that made Vizzy flip over.

What is your plan on simulating the velocity control for the icub joints? Is there an easy way to implement a stable velocity controller? @randaz81, what is your opinion on this issue?

randaz81, 2016-05-12:
I spoke with @traversaro some time ago, and we both have the same opinion: we should write our own controller (PID) and do not rely on the gazebo implementation... @traversaro Can we discuss about this next week?

traversaro, 2016-05-16:
Just for reference: 
@arocchi @EnricoMingo @liesrock have you ever used the velocity control on Gazebo?

liesrock, 2016-05-16:
Hi @traversaro: neither me nor @arocchi ever used it....I don't know about @EnricoMingo.

EnricoMingo, 2016-05-17:
Never used.

pattacini, 2020-09-20:
Now that the iCub Gazebo simulator has reached almost the feature parity w/ iCub_SIM (we've got everything implemented but the skin), I'd like to raise again attention on this issue.

Velocity control is NOT working at all. An example of this can be derived from https://github.com/vvv-school/tutorial_joint-interface, which I've recently upgraded to the use of Gazebo in place of iCub_SIM.

`position` and `positiondirect` modes do work as expected, whereas `velocity` mode is completely out of this world.

traversaro, 2020-09-21:
Hi @pattacini, I guess that the more logical solution is to implement also in gazebo-yarp-plugins (as an additional option) the velocity control as it is implemented in `iCub_SIM` and in the real iCub firmware, i.e. by integrating the velocity setpoint to compute a position reference that is then sent to the low level position control. In this way, we should have something that works fine without the need to tune any gain for the "velocity" PID control. In theory it should be easy to do so, as the logic should be already present as part of the mixed control mode (see https://github.com/robotology/gazebo-yarp-plugins/blob/v3.5.0/plugins/controlboard/src/ControlBoardDriver.cpp#L609), but also that implementation needs to be checked as those 1000.0 and 1.0 hardcoded are a bit fishy.

traversaro, 2020-09-21:
I check the change and it should be doable.

closed 2020-09-22 by traversaro