robotology/gazebo-yarp-plugins #243: New YARP workflow

traversaro, 2016-05-26:
Historically, the `gazebo-yarp-plugins` master branch has been compatible with the master branch of `YARP` (in the past we try to be compatible also with the latest YARP release, but due to the breaking change in YARP at some point we just gave up). 

With the new workflow of YARP, this policy should be updated. 
For example this commit: https://github.com/robotology/gazebo-yarp-plugins/commit/4ac18d614abbee3e7cc49735f90fb979088eeabd broke compilation of `gazebo-yarp-plugins`, because it depended on a breaking change present only in the `devel` branch of https://github.com/robotology/yarp/commit/2159b37e1d984ba6f0794f61b41cc19014a4adba . 

For the time being (to avoid breaking the compilation of `gazebo-yarp-plugins` for most users) I created a `devel` branch also in `gazebo-yarp-plugins`, containing the commit that depends on the `devel` branch in yarp, while reverting the same commit in the `master` branch. 
However a proper policy, discussing for example if the master of `gyp` be compatible with YARP `master` and `devel`, should be prepared. 
If the coupling between `gazebo-yarp-plugins` and YARP interfaces is indeed so high, a possible convenient option is to just move `gazebo-yarp-plugins` in the `yarp` repository.

traversaro, 2016-05-26:
@randaz81 @drdanz @lornat75 @aerydna

barbalberto, 2016-05-27:
yes, you are right. 
master of `gazebo-yarp-plugins` should be compatible with master of `yarp`.
That was my fault, I've not thought this commit would break compilation.

traversaro, 2016-05-27:
Yes, that was a combination of factor, including the fact that according to the new workflow, no method should be removed from public API : https://github.com/robotology/yarp/blob/master/.github/CONTRIBUTING.md#deprecated-features . 
I don't know if it may be worth to document in the workflow document eventual exception for this role (for example for interfaces that are still a work in progress).

drdanz, 2016-05-27:
Sorry, my fault.
I didn't notice that @aerydna's commit changed the src/libYARP_dev/include/yarp/dev/IDepthSensor.h file as well, I thought it was changing only the driver. We can restore the old method, marking that as deprecated, or maybe change it back, since yarp::os::Property derives from yarp::os::Searchable... I don't know if the change to Property is really that necessary... @barbalberto what do you think?

barbalberto, 2016-05-27:
The change is necessary, and that interface is gonna change a lot in the next future since it is a wip.
In yarp we pushed the change in the devel, but since there were no devel yet in gazebo plugin we set it into master. My bad, we should have created a branch in gazebo anyway.

drdanz, 2016-05-27:
As discussed, I just opened a pull request to remove the "unstable" interfaces and devices from the `master` branch of YARP, see robotology/yarp#779.
Unfortunately there is no way to remove them from the tagged version and from the released binaries, therefore I added this information to the release notes.

traversaro, 2018-01-08:
We have been using the new master + devel model in the last year, and it seems to be working nice. If there are new issues we will open them accordingly.

closed 2018-01-08 by traversaro