robotology/gazebo-yarp-plugins #244: Implementing MotorEncoders

hu-yue, 2016-06-07:
Hi,
Pretty long ago (I think almost one year ago) I started implementing some stuff to work with SEAs in Gazebo and implemented MotorEncoders for this reason.
Since now we are (finally) back working on this again, I'd like to ask if there's any plan from this point of view or if it is rather just us from Heidelberg who are interested in this and I should synchronize the (very old) sea branch with master.

Thanks!
@traversaro

hu-yue, 2016-06-07:
Of course I'm referring to #192, but I openend a new issue because the question was more "are there any plans going on implementing MotorEncoders now."

MirkoFerrati, 2016-06-07:
Hi,
me and a couple of colleagues in Centro E.Piaggio would be interested in trying SEA joints with the robot model of Walk-Man. This is not a priority for us, but if you need some testers, we could try and allocate some time for it.
If you are willing to provide support to our tests, can you provide a starting guide for migrating normal joints to SEA in the URDF and in the yarp config file?

hu-yue, 2016-06-08:
@MirkoFerrati Thanks for the offer! That would be appreciated.
I am working on it already, but can't say when it will be usable.
I will do some tests myself first, then I'll write down the migration procedure and let you know.