robotology/gazebo-yarp-plugins #245: Error while compiling gazebo_yarp_depthCamera component

pattacini, 2016-06-12:
Just got the error reported below while compiling `gazebo-yarp-plugins/master` against `yarp/devel`.

```sh
[ 96%] Building CXX object plugins/depthCamera/CMakeFiles/gazebo_yarp_depthCamera.dir/src/DepthCamera.cc.o
In file included from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/PolyDriver.h:11:0,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/depthCamera/include/gazebo/DepthCamera.hh:14,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/depthCamera/src/DepthCamera.cc:8:
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/Drivers.h: In instantiation of ‘yarp::dev::DeviceDriver* yarp::dev::DriverCreatorOf<T>::create() [with T = yarp::dev::GazeboYarpDepthCameraDriver]’:
/usr/local/src/robot/gazebo-yarp-plugins/plugins/depthCamera/src/DepthCamera.cc:101:1:   required from here
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/Drivers.h:133:20: error: invalid new-expression of abstract class type ‘yarp::dev::GazeboYarpDepthCameraDriver’
         return new T;
                    ^
In file included from /usr/local/src/robot/gazebo-yarp-plugins/plugins/depthCamera/src/DepthCamera.cc:9:0:
/usr/local/src/robot/gazebo-yarp-plugins/plugins/depthCamera/include/yarp/dev/DepthCameraDriver.h:45:18: note:   because the following virtual functions are pure within ‘yarp::dev::GazeboYarpDepthCameraDriver’:
 class yarp::dev::GazeboYarpDepthCameraDriver:   public yarp::dev::DeviceDriver,
                  ^
In file included from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/IRGBDSensor.h:29:0,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/depthCamera/include/yarp/dev/DepthCameraDriver.h:11,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/depthCamera/src/DepthCamera.cc:9:
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/IDepthSensor.h:86:18: note: 	virtual bool yarp::dev::IDepthSensor::getDeviceInfo(yarp::os::Property&)
     virtual bool getDeviceInfo(yarp::os::Property &device_info) = 0;
                  ^
plugins/depthCamera/CMakeFiles/gazebo_yarp_depthCamera.dir/build.make:62: recipe for target 'plugins/depthCamera/CMakeFiles/gazebo_yarp_depthCamera.dir/src/DepthCamera.cc.o' failed
make[2]: *** [plugins/depthCamera/CMakeFiles/gazebo_yarp_depthCamera.dir/src/DepthCamera.cc.o] Error 1
CMakeFiles/Makefile2:1155: recipe for target 'plugins/depthCamera/CMakeFiles/gazebo_yarp_depthCamera.dir/all' failed
make[1]: *** [plugins/depthCamera/CMakeFiles/gazebo_yarp_depthCamera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

traversaro, 2016-06-12:
I think that following https://github.com/robotology/yarp/pull/779 we need to remove `gazebo_yarp_depthcamera`, to avoid problems like this. For the time being you can just switch to the `devel` branch of this repo (but this should be documented in the README. For more info see #243 .

pattacini, 2016-06-12:
Switching to `gazebo-yarp-plugins/devel` solved the problem.
Thanks @traversaro

closed 2016-06-12 by pattacini