robotology/gazebo-yarp-plugins #248: Gazebo crashed after removing iCub

S-Dafarra, 2016-06-21:
~~~***** Internal Program Error - assertion (diffTime.sec < maxSensorUpdate) failed in void gazebo::sensors::SensorManager::SensorContainer::RunLoop():
/var/lib/jenkins/workspace/gazebo7-debbuilder/build/gazebo-7.2.0/gazebo/sensors/SensorManager.cc(551): Took over 1000*max_step_size to update a sensor.~~~

This error appeared on the terminal after the crash.
I noticed it happens often when I remove the robot.

traversaro, 2016-06-21:
Relevant line in Gazebo source code : https://bitbucket.org/osrf/gazebo/src/5adf274c0172e4707ac86523c3aedd3d332f3c7c/gazebo/sensors/SensorManager.cc?at=default&fileviewer=file-view-default#SensorManager.cc-551 .

Tobias-Fischer, 2016-07-06:
Can you try whether this still happens after applying https://github.com/robotology/gazebo-yarp-plugins/pull/253 and https://github.com/robotology/gazebo-yarp-plugins/pull/252 and https://github.com/robotology/yarp/pull/814 (with the latter not being merged yet!)?

traversaro, 2016-07-06:
We don't know how to reproduce this issue, so let's wait one month if it does not happen anymore and then we can close the issue.

traversaro, 2016-11-08:
@S-Dafarra Please update (and reopen) this issue if you experience this problem again.

closed 2016-11-08 by traversaro