robotology/gazebo-yarp-plugins #254: iCub Failing on Start-Up Gazebo

TeoKa, 2016-07-12:
~~~
[INFO]/icubGazeboSim/torso : no ROS initialization required 
[INFO]/icubGazeboSim/torso  initting YARP initialization 
yarp: Port /icubGazeboSim/torso/rpc:i active at tcp://10.0.0.122:10009
yarp: Port /icubGazeboSim/torso/command:i active at tcp://10.0.0.122:10010
yarp: Port /icubGazeboSim/torso/state:o active at tcp://10.0.0.122:10011
yarp: Port /icubGazeboSim/torso/stateExt:o active at tcp://10.0.0.122:10012
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
[ERROR]GazeboYarpControlBoard : wrapper interface not found, load failed.
[INFO]/icubGazeboSim/head : no ROS initialization required 
[INFO]/icubGazeboSim/head  initting YARP initialization 
yarp: Port /icubGazeboSim/head/rpc:i active at tcp://10.0.0.122:10013
yarp: Port /icubGazeboSim/head/command:i active at tcp://10.0.0.122:10014
yarp: Port /icubGazeboSim/head/state:o active at tcp://10.0.0.122:10015
yarp: Port /icubGazeboSim/head/stateExt:o active at tcp://10.0.0.122:10016
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
[ERROR]GazeboYarpControlBoard : wrapper interface not found, load failed.
[INFO]/icubGazeboSim/left_arm : no ROS initialization required 
[INFO]/icubGazeboSim/left_arm  initting YARP initialization 
yarp: Port /icubGazeboSim/left_arm/rpc:i active at tcp://10.0.0.122:10017
yarp: Port /icubGazeboSim/left_arm/command:i active at tcp://10.0.0.122:10018
yarp: Port /icubGazeboSim/left_arm/state:o active at tcp://10.0.0.122:10019
yarp: Port /icubGazeboSim/left_arm/stateExt:o active at tcp://10.0.0.122:10020
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
[ERROR]GazeboYarpControlBoard : wrapper interface not found, load failed.
[INFO]/icubGazeboSim/right_arm : no ROS initialization required 
[INFO]/icubGazeboSim/right_arm  initting YARP initialization 
yarp: Port /icubGazeboSim/right_arm/rpc:i active at tcp://10.0.0.122:10021
yarp: Port /icubGazeboSim/right_arm/command:i active at tcp://10.0.0.122:10022
yarp: Port /icubGazeboSim/right_arm/state:o active at tcp://10.0.0.122:10023
yarp: Port /icubGazeboSim/right_arm/stateExt:o active at tcp://10.0.0.122:10024
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
[ERROR]GazeboYarpControlBoard : wrapper interface not found, load failed.
[INFO]/icubGazeboSim/left_leg : no ROS initialization required 
[INFO]/icubGazeboSim/left_leg  initting YARP initialization 
yarp: Port /icubGazeboSim/left_leg/rpc:i active at tcp://10.0.0.122:10025
yarp: Port /icubGazeboSim/left_leg/command:i active at tcp://10.0.0.122:10026
yarp: Port /icubGazeboSim/left_leg/state:o active at tcp://10.0.0.122:10027
yarp: Port /icubGazeboSim/left_leg/stateExt:o active at tcp://10.0.0.122:10028
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
[ERROR]GazeboYarpControlBoard : wrapper interface not found, load failed.
[INFO]/icubGazeboSim/right_leg : no ROS initialization required 
[INFO]/icubGazeboSim/right_leg  initting YARP initialization 
yarp: Port /icubGazeboSim/right_leg/rpc:i active at tcp://10.0.0.122:10029
yarp: Port /icubGazeboSim/right_leg/command:i active at tcp://10.0.0.122:10030
yarp: Port /icubGazeboSim/right_leg/state:o active at tcp://10.0.0.122:10031
yarp: Port /icubGazeboSim/right_leg/stateExt:o active at tcp://10.0.0.122:10032
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
[ERROR]GazeboYarpControlBoard : wrapper interface not found, load failed.
yarp: Port /icubGazeboSim/applyExternalWrench/rpc:i active at tcp://10.0.0.122:10033

~~~

traversaro, 2016-07-13:
The error causing the problem is:
~~~
[ERROR]GazeboYarpControlBoard : wrapper interface not found, load failed.
~~~
Happening at https://github.com/robotology/gazebo-yarp-plugins/blob/d2c00df022fece585f5ad2ab41a6843b93b59963/plugins/controlboard/src/ControlBoard.cc#L113 (i.e. a failure in opening the `controlboardwrapper` device). 
I had the same problem on my Mac some time ago, and I discovered that the problem was due to a `dynamic_cast` failing when it was not supposed to fail, so I imagined was a problem of multiple versions of YARP being present in the system, and I was assuming was a problem due to my messy system, but apparently is something more serious. Checking again.

traversaro, 2016-07-13:
I replicated the issue: 
~~~
[ERROR]GazeboYarpControlBoard : wrapper interface not found, load failed.
~~~
Trying to replicate it in self contained main.

traversaro, 2016-07-13:
I failed to solve the problem this evening. 
I was successful anyhow in ruling out Gazebo from the problem, and replicated the problem as a Yarp unit test in https://github.com/traversaro/yarp/commit/21398625fc8c32367cc33679dda3850b8c6cab5e .

traversaro, 2016-07-14:
I tested the `dynamic_cast` on all the interfaces inherited by the `ControlBoardWrapper` class , see https://gist.github.com/traversaro/fa12bf54d842ca7cf4c97310f644acb0 . 
Results : 
~~~
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
--> Success in casting to N4yarp3dev12DeviceDriverE
--> Success in casting to N4yarp2os10RateThreadE
XXX Failure in casting to N4yarp3dev11IPidControlE
--> Success in casting to N4yarp3dev17IPositionControl2E
--> Success in casting to N4yarp3dev15IPositionDirectE
--> Success in casting to N4yarp3dev17IVelocityControl2E
--> Success in casting to N4yarp3dev14IEncodersTimedE
--> Success in casting to N4yarp3dev6IMotorE
--> Success in casting to N4yarp3dev14IMotorEncodersE
XXX Failure in casting to N4yarp3dev17IAmplifierControlE
--> Success in casting to N4yarp3dev14IControlLimitsE
--> Success in casting to N4yarp3dev17IRemoteCalibratorE
--> Success in casting to N4yarp3dev19IControlCalibrationE
--> Success in casting to N4yarp3dev20IControlCalibration2E
--> Success in casting to N4yarp3dev16IOpenLoopControlE
--> Success in casting to N4yarp3dev14ITorqueControlE
--> Success in casting to N4yarp3dev17IImpedanceControlE
--> Success in casting to N4yarp3dev13IControlMode2E
XXX Failure in casting to N4yarp3dev16IMultipleWrapperE
--> Success in casting to N4yarp3dev9IAxisInfoE
--> Success in casting to N4yarp3dev15IPreciselyTimedE
--> Success in casting to N4yarp3dev16IInteractionModeE
--> Success in casting to N4yarp3dev16IRemoteVariablesE
~~~

traversaro, 2016-07-14:
The issue just disappear on my computer. O_o

traversaro, 2016-07-14:
Recap: the issue only appears if: 
  * YARP is built using the superbuild
  * OS X is the operating system

It disappears if the YARP test compilation are enabled ( see https://github.com/robotology/codyco-superbuild/commit/9e452e5f3e9417068309926a57b6404f44794c5c )... 
This workaround permits @TeoKa to work, but clearly is not a proper solution to the problem.

traversaro, 2016-07-14:
The problem was that YARP in the codyco-superbuild was compiled with the `YARP_CLEAN_API` option, that is not tested at all [1] and so it is risky to use it. 
`YARP_CLEAN_API` remove the symbols that are not explicitly exported in the API, and so it removed the symbols for the `yarp::dev::IMultipleWrapper` interface, because this interface was not correctly exported (fixed in https://github.com/robotology/yarp/pull/824/commits/2edd594a5259d3e9c65ec2f2a6777225d9c3c624). 

Just to be sure to avoid this kind of problems in the future, I removed the `YARP_CLEAN_API` option from the superbuild : https://github.com/robotology/codyco-superbuild/commit/daa6958b42a0dd8c27995ca98afbd4108f591cbe and this (together with correctly exporting the interfaces) should solve this issue. 

[1] : https://github.com/robotology/yarp/pull/824#issuecomment-232632795

closed 2016-07-14 by traversaro