robotology/gazebo-yarp-plugins #256: Memory leak

Tobias-Fischer, 2016-07-20:
Hi,
Unfortunately there seems to be a memory leak somewhere within `gazebo-yarp-plugins`. I noticed that the memory used by `gzserver` grows indefinitely, and at some stage `gzserver` crashes.
I deleted all references to yarp plugins in the `icub-gazebo` model, and now all works nicely. The experiment I am doing basically just moves the ikart+iCub around, and saves some stuff in a rosbag. Nothing fancy, really, but the experiment is running for quite some time.

Not sure how to investigate this unfortunately.

Best, Tobi

traversaro, 2016-07-20:
If you are using the `dev` branch of yarp, it could be related to https://github.com/robotology/yarp/issues/833 .

Tobias-Fischer, 2016-07-20:
I didn't see this yet, thanks, that might well be the issue.

barbalberto, 2016-07-21:
The yarp master branch is not affected, try with it

traversaro, 2016-07-25:
Fixed by https://github.com/robotology/yarp/commit/f467a922be4d535cdae77777d992cb6f1284aaec (and possibly https://github.com/robotology/gazebo-yarp-plugins/pull/257).

closed 2016-07-25 by traversaro