robotology/gazebo-yarp-plugins #258: World Port Control with gazebo-yarp

lucamrs, 2016-11-08:
Hi there,
i'm using iCub_SIM to simulate grasping behaviour via joints position control and i wanted to try gazebo-yarp to test the functions and performance of the new simulator.
The problem is that, if my understanding is correct, there's no backward compatibility between programs that interacted with Yarp/iCub_SIM before and the new gazebo-yarp interface.
Connecting to the world port and sending commands to it doesn't seem to have any effect at all.
I can see that some yarp related functions ended up in "WorldInterface" but i couldn't find a basic tutorial on how to setup and use them.
It would be useful to have some examples lines, al least for the basis object creation and removal commands (eg: world mk sbox.../world del ...).
Thank you for your help

randaz81, 2016-11-08:
yes, unfortunately the interface commands have different names from icub_sim. 
I'll check for the tutorial...

charleswilmot, 2017-01-10:
Hello,

I'm also very interested in some examples. I use gazebo through the python bindings, and I haven't found any classes or methods called `WorldInterface`.

@randaz81 Is there bindings for that plugin? If not, is there a workaround I could use?

Thank you very much

(tag #159)

pattacini, 2017-03-15:
Duplicate of https://github.com/robotology/QA/issues/201

closed 2017-03-15 by pattacini