robotology/gazebo-yarp-plugins #262: gazebo-yarp-plugins not compiling against latest YARP (both master and devel)

traversaro, 2016-11-28:
~~~
In file included from /Users/traversaro/src/codyco-superbuild/libraries/GazeboYARPPlugins/plugins/depthCamera/src/DepthCamera.cc:9:
/Users/traversaro/src/codyco-superbuild/libraries/GazeboYARPPlugins/plugins/depthCamera/include/yarp/dev/DepthCameraDriver.h:99:34: error: unknown type name 'DepthSensor_status'; did you mean 'RGBDSensor_status'?
    virtual bool getDeviceStatus(DepthSensor_status *status);
                                 ^~~~~~~~~~~~~~~~~~
                                 RGBDSensor_status
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:73:10: note: 'RGBDSensor_status' declared here
    enum RGBDSensor_status
         ^
In file included from /Users/traversaro/src/codyco-superbuild/libraries/GazeboYARPPlugins/plugins/depthCamera/src/DepthCamera.cc:8:
In file included from /Users/traversaro/src/codyco-superbuild/libraries/GazeboYARPPlugins/plugins/depthCamera/include/gazebo/DepthCamera.hh:14:
In file included from /Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/PolyDriver.h:11:
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/Drivers.h:133:20: error: allocating an object of abstract class type 'yarp::dev::GazeboYarpDepthCameraDriver'
        return new T;
                   ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/Drivers.h:111:5: note: in instantiation of member function 'yarp::dev::DriverCreatorOf<yarp::dev::GazeboYarpDepthCameraDriver>::create' requested here
    DriverCreatorOf(const char *name, const char *wrap, const char *code) :
    ^
/Users/traversaro/src/codyco-superbuild/libraries/GazeboYARPPlugins/plugins/depthCamera/src/DepthCamera.cc:51:45: note: in instantiation of member function 'yarp::dev::DriverCreatorOf<yarp::dev::GazeboYarpDepthCameraDriver>::DriverCreatorOf' requested here
    ::yarp::dev::Drivers::factory().add(new ::yarp::dev::DriverCreatorOf< ::yarp::dev::GazeboYarpDepthCameraDriver>
                                            ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:89:18: note: unimplemented pure virtual method 'getRgbHeight' in 'GazeboYarpDepthCameraDriver'
    virtual int  getRgbHeight() = 0;
                 ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:90:18: note: unimplemented pure virtual method 'getRgbWidth' in 'GazeboYarpDepthCameraDriver'
    virtual int  getRgbWidth() = 0;
                 ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:91:18: note: unimplemented pure virtual method 'setRgbResolution' in 'GazeboYarpDepthCameraDriver'
    virtual bool setRgbResolution(int width, int height) = 0;
                 ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:92:18: note: unimplemented pure virtual method 'getRgbFOV' in 'GazeboYarpDepthCameraDriver'
    virtual bool getRgbFOV(double &horizontalFov, double &verticalFov) = 0;
                 ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:93:18: note: unimplemented pure virtual method 'setRgbFOV' in 'GazeboYarpDepthCameraDriver'
    virtual bool setRgbFOV(double horizontalFov, double verticalFov) = 0;
                 ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:94:18: note: unimplemented pure virtual method 'getRgbIntrinsicParam' in 'GazeboYarpDepthCameraDriver'
    virtual bool getRgbIntrinsicParam(yarp::os::Property &intrinsic) = 0;
                 ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IVisualParams.h:125:18: note: unimplemented pure virtual method 'getRgbMirroring' in 'GazeboYarpDepthCameraDriver'
    virtual bool getRgbMirroring(bool &mirror) = 0;
                 ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IVisualParams.h:133:18: note: unimplemented pure virtual method 'setRgbMirroring' in 'GazeboYarpDepthCameraDriver'
    virtual bool setRgbMirroring(bool mirror) = 0;
                 ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:99:20: note: unimplemented pure virtual method 'getDepthHeight' in 'GazeboYarpDepthCameraDriver'
    virtual int    getDepthHeight() = 0;
                   ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:100:20: note: unimplemented pure virtual method 'getDepthWidth' in 'GazeboYarpDepthCameraDriver'
    virtual int    getDepthWidth() = 0;
                   ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:101:20: note: unimplemented pure virtual method 'setDepthResolution' in 'GazeboYarpDepthCameraDriver'
    virtual bool   setDepthResolution(int width, int height) = 0;
                   ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:102:20: note: unimplemented pure virtual method 'getDepthFOV' in 'GazeboYarpDepthCameraDriver'
    virtual bool   getDepthFOV(double &horizontalFov, double &verticalFov) = 0;
                   ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:103:20: note: unimplemented pure virtual method 'setDepthFOV' in 'GazeboYarpDepthCameraDriver'
    virtual bool   setDepthFOV(double horizontalFov, double verticalFov) = 0;
                   ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:108:20: note: unimplemented pure virtual method 'getDepthIntrinsicParam' in 'GazeboYarpDepthCameraDriver'
    virtual bool   getDepthIntrinsicParam(yarp::os::Property &intrinsic) = 0;
                   ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:104:20: note: unimplemented pure virtual method 'getDepthAccuracy' in 'GazeboYarpDepthCameraDriver'
    virtual double getDepthAccuracy() = 0;
                   ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:105:20: note: unimplemented pure virtual method 'setDepthAccuracy' in 'GazeboYarpDepthCameraDriver'
    virtual bool   setDepthAccuracy(double accuracy) = 0;
                   ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:106:20: note: unimplemented pure virtual method 'getDepthClipPlanes' in 'GazeboYarpDepthCameraDriver'
[... 54 lines cut ...]
                          ^
/Users/traversaro/src/gazebo-superbuild/build/install/include/gazebo-8/gazebo/rendering/Camera.hh:509:20: note: 'DisconnectNewImageFrame' has been explicitly marked deprecated here
      public: void DisconnectNewImageFrame(event::ConnectionPtr &_c)
                   ^
/Users/traversaro/src/codyco-superbuild/libraries/GazeboYARPPlugins/plugins/depthCamera/src/DepthCameraDriver.cpp:398:51: error: unknown type name 'DepthSensor_status'
bool GazeboYarpDepthCameraDriver::getDeviceStatus(DepthSensor_status *status)
                                                  ^
/Users/traversaro/src/codyco-superbuild/libraries/GazeboYARPPlugins/plugins/depthCamera/src/DepthCameraDriver.cpp:402:32: error: no member named 'DepthSensor_status' in 'yarp::dev::IRGBDSensor'; did you mean 'RGBDSensor_status'?
        *status = IRGBDSensor::DepthSensor_status::DEPTHSENSOR_OK_IN_USE;
                  ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
                               RGBDSensor_status
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:73:10: note: 'RGBDSensor_status' declared here
    enum RGBDSensor_status
         ^
/Users/traversaro/src/codyco-superbuild/libraries/GazeboYARPPlugins/plugins/depthCamera/src/DepthCameraDriver.cpp:402:52: error: no member named 'DEPTHSENSOR_OK_IN_USE' in 'yarp::dev::IRGBDSensor::RGBDSensor_status'; did you mean 'RGBD_SENSOR_OK_IN_USE'?
        *status = IRGBDSensor::DepthSensor_status::DEPTHSENSOR_OK_IN_USE;
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
                                                   RGBD_SENSOR_OK_IN_USE
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRGBDSensor.h:77:9: note: 'RGBD_SENSOR_OK_IN_USE' declared here
        RGBD_SENSOR_OK_IN_USE        = 2,
        ^
3 warnings and 4 errors generated.
~~~

Related to https://github.com/robotology/yarp/pull/974 , I guess.

traversaro, 2016-11-28:
In https://github.com/robotology/yarp/pull/974 @barbalberto wrote:
> yes, the gazebo plugin in devel has to be updated to the new interface

@barbalberto @aerydna Is this planned in the near future? Given the recent merge of devel in master, this means that gazebo-yarp-plugins compilation is broken on both master and devel.

traversaro, 2016-11-28:
Actually in current master `depthCamera` is disabled, so at least the master is still compiling.

barbalberto, 2016-11-28:
yes, we are gonna disable the plugin for now, untill it is updated.
Sorry about that, but time was too short to update it

traversaro, 2016-11-28:
Nope, also master is broken on some laser stuff (I guess updated in `devel`): 
~~~
[ 96%] Building CXX object plugins/lasersensor/CMakeFiles/gazebo_yarp_lasersensor.dir/src/LaserSensor.cc.o
In file included from /Users/traversaro/src/codyco-superbuild/libraries/GazeboYARPPlugins/plugins/lasersensor/src/LaserSensor.cc:8:
In file included from /Users/traversaro/src/codyco-superbuild/libraries/GazeboYARPPlugins/plugins/lasersensor/include/gazebo/LaserSensor.hh:13:
In file included from /Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/PolyDriver.h:11:
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/Drivers.h:133:20: error: allocating an object of abstract class type 'yarp::dev::GazeboYarpLaserSensorDriver'
        return new T;
                   ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/Drivers.h:111:5: note: in instantiation of member function 'yarp::dev::DriverCreatorOf<yarp::dev::GazeboYarpLaserSensorDriver>::create' requested here
    DriverCreatorOf(const char *name, const char *wrap, const char *code) :
    ^
/Users/traversaro/src/codyco-superbuild/libraries/GazeboYARPPlugins/plugins/lasersensor/src/LaserSensor.cc:56:45: note: in instantiation of member function 'yarp::dev::DriverCreatorOf<yarp::dev::GazeboYarpLaserSensorDriver>::DriverCreatorOf' requested here
    ::yarp::dev::Drivers::factory().add(new ::yarp::dev::DriverCreatorOf< ::yarp::dev::GazeboYarpLaserSensorDriver>
                                            ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRangefinder2D.h:52:18: note: unimplemented pure virtual method 'getLaserMeasurement' in 'GazeboYarpLaserSensorDriver'
    virtual bool getLaserMeasurement(std::vector<LaserMeasurementData> &data) = 0;
                 ^
/Users/traversaro/src/codyco-superbuild/build/install/include/yarp/dev/IRangefinder2D.h:59:18: note: unimplemented pure virtual method 'getRawData' in 'GazeboYarpLaserSensorDriver'
    virtual bool getRawData(yarp::sig::Vector &data) = 0;
                 ^
1 error generated.
~~~

Then we can disable the `depthCamera` and merge https://github.com/robotology/gazebo-yarp-plugins/pull/261 ? 

@barbalberto no worries about not updating the plugin, but at least an heads up would have been nice...

barbalberto, 2016-11-28:
I just disabled  `depthCamera`. I don't know about laser

traversaro, 2016-11-28:
I think laser is just an inconsistency between 2.3.66 and 2.3.68, should be fixed by https://github.com/robotology/gazebo-yarp-plugins/issues/262 .

barbalberto, 2016-11-28:
So if devel now compiles without the depth camera, I think we can merge

traversaro, 2016-11-29:
Fixed by https://github.com/robotology/gazebo-yarp-plugins/pull/261 .

closed 2016-11-29 by traversaro