robotology/gazebo-yarp-plugins #266: Help and examples needed  |   python -> YARP -> Gazebo -> iCUB

charleswilmot, 2016-12-19:
Hello,

I need a bit of help to get iCub babbling in Gazebo. I would like to make him move via a python script. (If you think I'm not asking the question at the right place, feel free to delete my issue)

So far, I think I've installed everything properly:

- [X] Gazebo 7.4.0
- [X] Yarp 2.3.68 with the python bindings
- [X] Gazebo-YARP-plugin

`GAZEBO_PLUGIN_PATH` is set to `gazebo-yarp-plugins/install/lib/x86_64-linux-gnu`

and after running the gazebo executable I get the answer
```
$ gazebo ./icub-gazebo/worlds/icub_fixed.world 
yarp: Port /icubSim/applyExternalWrench/rpc:i active at tcp://141.2.249.190:10002/
```
Which look pretty ok to me, even thought only one port exists.

Here is a list of the ports I have:
```
$ ./yarp name list
registration name /icubSim/applyExternalWrench/rpc:i ip 141.2.249.190 port 10002 type tcp
registration name /root ip 141.2.249.190 port 10000 type tcp
registration name fallback ip 224.2.1.1 port 10000 type mcast
*** end of message
```

How can I determine if Gazebo and Yarp are well configured?

I'm now wondering what I have to do in order to
1. get iCub moving from the command line (with YARP)
2. get iCub moving from a python script

#### Any advice, link, tutorial or example would be very appreciated!

Thank you
[image]

traversaro, 2016-12-19:
I assume you set the `GAZEBO_MODEL_PATH` environment variable? 
If this env variables is properly set,I guess something is going wrong when loading the iCub in Gazebo. 
Can you run `gazebo` with the `--verbose` flag? This should enabled the verbose output to be displayed.

Thanks for the nice picture!

charleswilmot, 2016-12-19:
Hello again @traversaro ,

Yes, my `GAZEBO_MODEL_PATH` is set to `/home/wilmot/.software/icub-gazebo`

The output with the flag `--verbose` tells that the files `/home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//lib*.so` doesn't exist, whereas they do:

```
$ gazebo --verbose ./icub-gazebo/worlds/icub_fixed.world 
Gazebo multi-robot simulator, version 7.4.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 7.4.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 141.2.249.190
[Msg] Publicized address: 141.2.249.190
[Err] [Node.cc:105] No namespace found
[Wrn] [msgs.cc:1793] Conversion of sensor type[imu] not suppported.
[Err] [InsertModelWidget.cc:299] Missing model.config for model "/home/wilmot/.software/icub-gazebo/.git"
[Err] [InsertModelWidget.cc:299] Missing model.config for model "/home/wilmot/.software/icub-gazebo/worlds"
[Wrn] [msgs.cc:1793] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1793] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1793] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1793] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1793] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1793] Conversion of sensor type[force_torque] not suppported.
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_imu.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_camera.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_camera.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_forcetorque.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/wilmot/.software/gazebo-yarp-plugins/install/lib/x86_64-linux-gnu//libgazebo_yarp_controlboard.so: libgazebo_yarp_singleton.so: cannot open shared object file: No such file or directory
[... cut ...]
[Wrn] [msgs.cc:1793] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1793] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1793] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1793] Conversion of sensor type[force_torque] not suppported.
[Wrn] [msgs.cc:1793] Conversion of sensor type[force_torque] not suppported.
[Err] [Scene.cc:2928] Light [sun] not found. Use topic ~/factory/light to spawn a new light.
```

I'll investigate that right now!

charleswilmot, 2016-12-19:
Ok so here is what I found,

```
ldd ../../../../../gazebo-yarp-plugins/install/lib/x86_64-linux-gnu/libgazebo_yarp_imu.so
	libgazebo_yarp_singleton.so => not found
```
( https://github.com/robotology/gazebo-yarp-plugins/issues/191#issuecomment-102345964 )

I'll add the plugin's lib path to the `LD_LIBRARY_PATH` and see what's happening.

traversaro, 2016-12-19:
It is strange, it seems to be an instance of https://github.com/robotology/gazebo-yarp-plugins/issues/99 emerging back from the past. : / 

Are you compiling with some strange option or from some old checkout? Even your installation prefix `lib/x86_64-linux-gnu` it is strange... `GAZEBO_YARP_PLUGINS_ENABLE_RPATH` is set to TRUE? `CMAKE_INSTALL_LIBDIR` is set to `lib`?

pattacini, 2016-12-20:
@charleswilmot look at the latest post on our [**G+ community**](https://plus.google.com/u/0/communities/116897884020660435466) 😄 

Also, you should've received an invitation by email to join **robotology**.

charleswilmot, 2017-01-03:
@pattacini Thank you for the invitation Ugo! Unfortunately I missed the invitation, it has been cancelled during my Christmas break.

@traversaro Yes you're right, `GAZEBO_YARP_PLUGINS_ENABLE_RPATH` was set to `ON`.
Regarding `CMAKE_INSTALL_LIBDIR`, it was set to `lib/x86_64-linux-gnu` by default (I guess). 
I'm not sure whether I should re-compile.

Now that it's working (with `LD_LIBRARY_PATH` set correctly), I will try to make [those](https://github.com/robotology/icub-tutorials/tree/master/python) examples run on my install.

pattacini, 2017-01-03:
@charleswilmot invitation resent 😉

charleswilmot, 2017-01-04:
## Motor control
It looks like the example about motor controls is not totally correct (`tmp` wasn't initialized and the program didn't wait for the commands to be executed). So I propose to change it with the following example:

```
import yarp
import time
yarp.Network.init()

def default_input(question,default):
    answer = input(question+" (default: "+default+")\n")
    return default if not answer else answer

# ask user to choose a device, default value = right arm
motors = default_input("Choose motor set (possible: head, right_hand, right_leg, ...)","right_arm")

# prepare a property object
props = yarp.Property()
props.put("device","remote_controlboard")
props.put("local","/client/" + motors)
props.put("remote","/icubSim/" + motors)

# create remote driver
armDriver = yarp.PolyDriver(props)

#query motor control interfaces
iPos = armDriver.viewIPositionControl()
iVel = armDriver.viewIVelocityControl()
iEnc = armDriver.viewIEncoders()

#retrieve number of joints
jnts=iPos.getAxes()

print('Controlling', jnts, 'joints')

# read encoders
encs=yarp.Vector(jnts)
iEnc.getEncoders(encs.data())

## store as home position
home= yarp.Vector(jnts, encs.data())

#initialize a new tmp vector identical to encs
tmp = yarp.Vector(jnts, encs.data())

# for each joint angle,
for i in range(jnts):
    # move 20 deg and wait 5 sec
    tmp[i] += 20
    iPos.positionMove(tmp.data())
    time.sleep(5)
    # move -40 deg and wait 5 sec
    tmp[i] -= 40
    iPos.positionMove(tmp.data())
    time.sleep(5)
    # move 20 deg and wait 5 sec
    tmp[i] += 20
    iPos.positionMove(tmp.data())
    time.sleep(5)

## go back to initial position
iPos.positionMove(home.data())

print("done")
```

## Imaging

On my installation, the example about imaging is not working. I would greatly appreciate if you could help me one more time in order to get things working.

First of all, `yarp name list` doesn't list `/icubSim/cam`, but I have an `/icubSim/cam/left` and an `/icubSim/cam/right` (left and right eyes), so I replaced `/icubSim/cam` with `/icubSim/cam/left` to receive images from the camera sensors.

The following code doesn't modify the array `img_array` which is supposed to contain the image from the left camera:

```
import numpy
import yarp
import scipy.ndimage
import matplotlib.pyplot as plt

# Initialise YARP
yarp.Network.init()

def yarp_to_python():
    # Create a port and connect it to the iCub simulator virtual camera
    input_port = yarp.Port()
    input_port.open("/python-image-port")
    yarp.Network.connect("/icubSim/cam/left", "/python-image-port")
    
    # Create numpy array to receive the image and the YARP image wrapped around it
    img_array = numpy.zeros((240, 320, 3), dtype=numpy.uint8)
    yarp_image = yarp.ImageRgb()
    yarp_image.resize(320, 240)
    yarp_image.setExternal(img_array, img_array.shape[1], img_array.shape[0])
    
    # Alternatively, if using Python 2.x, try:
    #yarp_image.setExternal(img_array.__array_interface__['data'][0], img_array.shape[1], img_array.shape[0])
    
    # Read the data from the port into the image
    input_port.read(yarp_image)
    
    # display the image that has been read
    plt.imshow(img_array)
    plt.show()
    # Cleanup
    input_port.close()

if __name__ == '__main__':
    yarp_to_python()
```

plinioMoreno, 2017-01-04:
From my experience, in the python code, try to read several times from the port. Since it's a `yarp.Port`, sometimes the image_array is an empty image.

charleswilmot, 2017-01-04:
Thank you for your help @plinioMoreno , unfortunately it didn't help

For your information, here is the list of the ports:

```
/yarp/install/bin$ ./yarp name list
registration name /icubSim/applyExternalWrench/rpc:i ip 141.2.249.190 port 10081 type tcp
registration name /icubSim/cam/left ip 141.2.249.190 port 10003 type tcp
registration name /icubSim/cam/right ip 141.2.249.190 port 10004 type tcp
registration name /icubSim/head/command:i ip 141.2.249.190 port 10078 type tcp
registration name /icubSim/head/rpc:i ip 141.2.249.190 port 10077 type tcp
registration name /icubSim/head/state:o ip 141.2.249.190 port 10079 type tcp
registration name /icubSim/head/stateExt:o ip 141.2.249.190 port 10080 type tcp
registration name /icubSim/inertial ip 141.2.249.190 port 10002 type tcp
registration name /icubSim/left_arm/analog:o ip 141.2.249.190 port 10006 type tcp
registration name /icubSim/left_arm/analog:o/rpc:i ip 141.2.249.190 port 10005 type tcp
registration name /icubSim/left_arm/command:i ip 141.2.249.190 port 10091 type tcp
registration name /icubSim/left_arm/rpc:i ip 141.2.249.190 port 10090 type tcp
registration name /icubSim/left_arm/state:o ip 141.2.249.190 port 10092 type tcp
registration name /icubSim/left_arm/stateExt:o ip 141.2.249.190 port 10093 type tcp
registration name /icubSim/left_arm_no_hand/command:i ip 141.2.249.190 port 10083 type tcp
registration name /icubSim/left_arm_no_hand/rpc:i ip 141.2.249.190 port 10082 type tcp
registration name /icubSim/left_arm_no_hand/state:o ip 141.2.249.190 port 10084 type tcp
registration name /icubSim/left_arm_no_hand/stateExt:o ip 141.2.249.190 port 10085 type tcp
registration name /icubSim/left_foot/analog:o ip 141.2.249.190 port 10012 type tcp
registration name /icubSim/left_foot/analog:o/rpc:i ip 141.2.249.190 port 10011 type tcp
registration name /icubSim/left_hand/analog:o ip 141.2.249.190 port 10099 type tcp
registration name /icubSim/left_hand/analog:o/rpc:i ip 141.2.249.190 port 10098 type tcp
registration name /icubSim/left_hand/command:i ip 141.2.249.190 port 10046 type tcp
registration name /icubSim/left_hand/rpc:i ip 141.2.249.190 port 10045 type tcp
registration name /icubSim/left_hand/state:o ip 141.2.249.190 port 10047 type tcp
registration name /icubSim/left_hand/stateExt:o ip 141.2.249.190 port 10048 type tcp
registration name /icubSim/left_hand_finger/command:i ip 141.2.249.190 port 10042 type tcp
registration name /icubSim/left_hand_finger/rpc:i ip 141.2.249.190 port 10041 type tcp
registration name /icubSim/left_hand_finger/state:o ip 141.2.249.190 port 10043 type tcp
registration name /icubSim/left_hand_finger/stateExt:o ip 141.2.249.190 port 10044 type tcp
registration name /icubSim/left_hand_index/command:i ip 141.2.249.190 port 10030 type tcp
registration name /icubSim/left_hand_index/rpc:i ip 141.2.249.190 port 10029 type tcp
registration name /icubSim/left_hand_index/state:o ip 141.2.249.190 port 10031 type tcp
registration name /icubSim/left_hand_index/stateExt:o ip 141.2.249.190 port 10032 type tcp
registration name /icubSim/left_hand_middle/command:i ip 141.2.249.190 port 10034 type tcp
registration name /icubSim/left_hand_middle/rpc:i ip 141.2.249.190 port 10033 type tcp
registration name /icubSim/left_hand_middle/state:o ip 141.2.249.190 port 10035 type tcp
registration name /icubSim/left_hand_middle/stateExt:o ip 141.2.249.190 port 10036 type tcp
registration name /icubSim/left_hand_pinky/command:i ip 141.2.249.190 port 10038 type tcp
registration name /icubSim/left_hand_pinky/rpc:i ip 141.2.249.190 port 10037 type tcp
registration name /icubSim/left_hand_pinky/state:o ip 141.2.249.190 port 10039 type tcp
registration name /icubSim/left_hand_pinky/stateExt:o ip 141.2.249.190 port 10040 type tcp
registration name /icubSim/left_hand_thumb/command:i ip 141.2.249.190 port 10026 type tcp
registration name /icubSim/left_hand_thumb/rpc:i ip 141.2.249.190 port 10025 type tcp
registration name /icubSim/left_hand_thumb/state:o ip 141.2.249.190 port 10027 type tcp
registration name /icubSim/left_hand_thumb/stateExt:o ip 141.2.249.190 port 10028 type tcp
registration name /icubSim/left_leg/analog:o ip 141.2.249.190 port 10010 type tcp
registration name /icubSim/left_leg/analog:o/rpc:i ip 141.2.249.190 port 10009 type tcp
registration name /icubSim/left_leg/command:i ip 141.2.249.190 port 10018 type tcp
registration name /icubSim/left_leg/rpc:i ip 141.2.249.190 port 10017 type tcp
registration name /icubSim/left_leg/state:o ip 141.2.249.190 port 10019 type tcp
registration name /icubSim/left_leg/stateExt:o ip 141.2.249.190 port 10020 type tcp
registration name /icubSim/right_arm/analog:o ip 141.2.249.190 port 10008 type tcp
registration name /icubSim/right_arm/analog:o/rpc:i ip 141.2.249.190 port 10007 type tcp
registration name /icubSim/right_arm/command:i ip 141.2.249.190 port 10095 type tcp
registration name /icubSim/right_arm/rpc:i ip 141.2.249.190 port 10094 type tcp
registration name /icubSim/right_arm/state:o ip 141.2.249.190 port 10096 type tcp
registration name /icubSim/right_arm/stateExt:o ip 141.2.249.190 port 10097 type tcp
registration name /icubSim/right_arm_no_hand/command:i ip 141.2.249.190 port 10087 type tcp
registration name /icubSim/right_arm_no_hand/rpc:i ip 141.2.249.190 port 10086 type tcp
registration name /icubSim/right_arm_no_hand/state:o ip 141.2.249.190 port 10088 type tcp
registration name /icubSim/right_arm_no_hand/stateExt:o ip 141.2.249.190 port 10089 type tcp
[... cut ...]
registration name /icubSim/right_leg/stateExt:o ip 141.2.249.190 port 10024 type tcp
registration name /icubSim/torso/command:i ip 141.2.249.190 port 10074 type tcp
registration name /icubSim/torso/rpc:i ip 141.2.249.190 port 10073 type tcp
registration name /icubSim/torso/state:o ip 141.2.249.190 port 10075 type tcp
registration name /icubSim/torso/stateExt:o ip 141.2.249.190 port 10076 type tcp
registration name /root ip 141.2.249.190 port 10000 type tcp
registration name fallback ip 224.2.1.1 port 10000 type mcast
*** end of message
```

charleswilmot, 2017-01-04:
Ok I found why it wasn't working. The resolution of the images produced by the camera is `480x640`. Simply replacing 240/320 by 480/640 solved the problem.

Here is the code that worked for me:
```
import numpy
import yarp
import scipy.ndimage
import matplotlib.pyplot as plt
import time

# Initialise YARP
yarp.Network.init()

def yarp_to_python():
    # Create a port and connect it to the iCub simulator virtual camera
    input_port = yarp.Port()
    input_port.open("/python-image-port")
    yarp.Network.connect("/icubSim/cam/left", "/python-image-port")
    
    # Create numpy array to receive the image and the YARP image wrapped around it
    img_array = numpy.ones((480, 640, 3), dtype=numpy.uint8)
    yarp_image = yarp.ImageRgb()
    yarp_image.resize(640, 480)
    yarp_image.setExternal(img_array, img_array.shape[1], img_array.shape[0])
    
    # Alternatively, if using Python 2.x, try:
    #yarp_image.setExternal(img_array.__array_interface__['data'][0], img_array.shape[1], img_array.shape[0])
    
    # Read the data from the port into the image
    input_port.read(yarp_image)
    
    # display the image that has been read
    plt.imshow(img_array)
    plt.show()
    # Cleanup
    input_port.close()

if __name__ == '__main__':
    yarp_to_python()
```

Thank you all for your patience!
Charles

closed 2017-01-16 by charleswilmot