robotology/gazebo-yarp-plugins #275: Test compilation against Gazebo 8

traversaro, 2017-02-01:
Gazebo 8 has been recently released ( http://gazebosim.org/blog/gazebo8 ) but it does not support Ubuntu 14.04 Trusty. As Travis only supports Trusty at the moment, we have no way of checking if `gazebo-yarp-plugins` compile against Gazebo 8. 

By randomly searching only it seems that it should be possible to use Xenial on Travis using Docker, but I don't fully understand how this is supposed to work and which are the limitations. Perhaps @diegoferigo knows something more on this?

diegoferigo, 2017-02-01:
I know what's Travis's purpose, and briefly checked in the past its documentation. However, I never had the chance to have some hands on it. As already discussed f2f, in the future I'd like to dig into Docker integration. Definitely this setup could be a great starting point.

Some pointers for future reference:
* OSRF official [ROS images][1]
* My personal [ROS desktop-full image][2] (still WIP)

[1]: https://github.com/osrf/docker_images
[2]: https://github.com/diegoferigo/dockerfiles/tree/master/ROS

traversaro, 2017-02-21:
cc @arocchi

traversaro, 2017-02-22:
Another possible strategy to test against recent version of Gazebo is to add a macOS-based Travis build: now that OSRF distributes bottles of Gazebo on homebrew it should be feasible to get it from homebrew in Travis.

traversaro, 2017-02-22:
An exaple of repo successfully using docker on Travis to test Ubuntu 16.04 is https://github.com/RobotLocomotion/director/blob/master/.travis.yml .

traversaro, 2017-03-06:
Fixed by https://github.com/robotology/gazebo-yarp-plugins/issues/276 .

closed 2017-03-06 by traversaro