robotology/gazebo-yarp-plugins #277: Gazebo 8 - ignition

iron76, 2017-02-03:
When compiling I am getting the following error:

``` gazebo-yarp-plugins/plugins/depthCamera/src/DepthCameraDriver.cpp:19:10: fatal  error:  'ignition/math2/ignition/math/Angle.hh' file not found```

Apparently in my installation I have `ignition/math3/ignition/math/Angle.hh`.

traversaro, 2017-02-03:
Can you paste the output of : `brew list` and the content of the `CMakeCache.txt` present in the build of `gazebo-yarp-plugins`?

traversaro, 2017-02-03:
PR https://github.com/robotology/gazebo-yarp-plugins/pull/278 should fix the issue.

traversaro, 2017-02-03:
This was indirectly caused by https://github.com/robotology/gazebo-yarp-plugins/issues/275 .

iron76, 2017-02-04:
Sure. Here they are.

[CMakeCache.txt](https://github.com/robotology/gazebo-yarp-plugins/files/752133/CMakeCache.txt)
[brew_list.txt](https://github.com/robotology/gazebo-yarp-plugins/files/752134/brew_list.txt)

traversaro, 2017-02-04:
@iron76 The issue should have been solved by https://github.com/robotology/gazebo-yarp-plugins/pull/278 .

traversaro, 2017-02-21:
This was solved by https://github.com/robotology/gazebo-yarp-plugins/pull/278 , please re-open if the problem persists.

closed 2017-02-21 by traversaro