robotology/gazebo-yarp-plugins #283: Error while compiling the component gazebo_arp_depthCamera

nunoguedelha, 2017-02-28:
I'm having an compiling error similar to https://github.com/robotology/gazebo-yarp-plugins/issues/245 when compiling `gazebo-yarp-plugins`, and more precisely the component `gazebo_arp_depthCamera`. Both **Yarp** and **Gazeb-yarp-plugins** are on the `devel` branch:
```
In file included from /Users/nunoguedelha/dev/gazebo-yarp-plugins/plugins/depthCamera/src/DepthCamera.cc:8:
In file included from /Users/nunoguedelha/dev/gazebo-yarp-plugins/plugins/depthCamera/include/gazebo/DepthCamera.hh:14:
In file included from /Users/nunoguedelha/dev/yarp/src/libYARP_dev/include/yarp/dev/PolyDriver.h:10:
/Users/nunoguedelha/dev/yarp/src/libYARP_dev/include/yarp/dev/Drivers.h:133:20: error: allocating an object of abstract class type 'yarp::dev::GazeboYarpDepthCameraDriver'
        return new T;
                   ^
/Users/nunoguedelha/dev/yarp/src/libYARP_dev/include/yarp/dev/Drivers.h:111:5: note: in instantiation of member function 'yarp::dev::DriverCreatorOf<yarp::dev::GazeboYarpDepthCameraDriver>::create' requested here
    DriverCreatorOf(const char *name, const char *wrap, const char *code) :
    ^
/Users/nunoguedelha/dev/gazebo-yarp-plugins/plugins/depthCamera/src/DepthCamera.cc:51:45: note: in instantiation of member function 'yarp::dev::DriverCreatorOf<yarp::dev::GazeboYarpDepthCameraDriver>::DriverCreatorOf' requested here
    ::yarp::dev::Drivers::factory().add(new ::yarp::dev::DriverCreatorOf< ::yarp::dev::GazeboYarpDepthCameraDriver>
                                            ^
In file included from /Users/nunoguedelha/dev/gazebo-yarp-plugins/plugins/depthCamera/src/DepthCamera.cc:9:
In file included from /Users/nunoguedelha/dev/gazebo-yarp-plugins/plugins/depthCamera/include/yarp/dev/DepthCameraDriver.h:11:
/Users/nunoguedelha/dev/yarp/src/libYARP_dev/include/yarp/dev/IRGBDSensor.h:79:18: note: unimplemented pure virtual method 'getRgbSupportedConfigurations' in 'GazeboYarpDepthCameraDriver'
    virtual bool getRgbSupportedConfigurations(yarp::sig::VectorOf<CameraConfig> &configurations)=0;
                 ^
/Users/nunoguedelha/dev/yarp/src/libYARP_dev/include/yarp/dev/IRGBDSensor.h:80:18: note: unimplemented pure virtual method 'getRgbResolution' in 'GazeboYarpDepthCameraDriver'
    virtual bool getRgbResolution(int &width, int &height)=0;
```

This time the fault goes to the virtual methods `getRgbResolution` and `getRgbSupportedConfigurations` in `GazeboYarpDepthCameraDriver` class, inherited from `yarp::dev::IRGBDSensor` (where they are pure virtual) _but not implemented yet_.

This has been introduced by the PR https://github.com/robotology/yarp/pull/1077.

nunoguedelha, 2017-02-28:
@traversaro @barbalberto @nicogene

barbalberto, 2017-02-28:
Yes, sorry, there was a mistake in the header file.
It has been fixed in commit 1cc5b46a72c1923255108381849ba52c03f60d3d
Please update yarp, it should work now.

Nicogene, 2017-02-28:
Hi @nunoguedelha,

my bad, I thought that defining these methods in `IRgbVisualParams` was sufficient.

traversaro, 2017-02-28:
To prevent this kind of failures, I added a Cron Job to execute the tests on `gazebo-yarp-plugins` daily (see https://travis-ci.org/robotology/gazebo-yarp-plugins/settings, Cron Job section). 
Furthermore, we need to test `devel` branch against yarp `devel` branch. I can manually edit the `.travis.yml`, but I don't know what is the proper way to avoid that this is propagated to `master` the next time we make a release.

traversaro, 2017-02-28:
Related : https://github.com/robotology/gazebo-yarp-plugins/pull/284 .

nunoguedelha, 2017-02-28:
thanks for the info guys.

closed 2017-02-28 by traversaro