robotology/gazebo-yarp-plugins #297: Devel travis build always failing

traversaro, 2017-05-01:
I guess this is due to the fact that the devel branch is using yarp master. 
@diegoferigo can you check this? Thanks

diegoferigo, 2017-05-02:
No, Travis uses `yarp:devel` on branches different than `master`. I guess the issue here is that the docker image with `yarp:devel` is not very updated, since its regeneration is performed manually. Let's wait dockerhub rebuilding it, I will follow up on this.

diegoferigo, 2017-05-04:
Due to a long chain of dependencies, this issue will be fixed as soon as https://github.com/robotology/yarp/pull/1194 is merged in `yarp:devel`.

diegoferigo, 2017-05-09:
The PR has been merged in `yarp:devel`, and the last commits in [robotology/docker_images](https://github.com/robotology-playground/docker_images) fix this issue.

closed 2017-05-09 by diegoferigo

traversaro, 2017-05-09:
👍