robotology/gazebo-yarp-plugins #299: Travis is failing in both master and devel

traversaro, 2017-05-29:
~~~
=========================================================
Loading /icub-tests/suits/basics-icubGazeboSim.xml
[94mStaring test runner.[0m
[94mTest suit Basic Tests Suite started...[0m
[92m[INFO]  [0m(Basic Tests Suite) reports: yarpmanager is setuping the fixture...
*** Error in `testrunner': free(): invalid pointer: 0x00007fffd4711850 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f01f61407e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x7fe0a)[0x7f01f6148e0a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f01f614c98c]
/usr/local/lib/rtf/yarpmanager.so(+0x7f368)[0x7f01f574e368]
/usr/local/lib/rtf/yarpmanager.so(+0x1ccfc)[0x7f01f56ebcfc]
/usr/local/lib/rtf/yarpmanager.so(+0x15ea7)[0x7f01f56e4ea7]
/usr/local/bin/../lib/libRTF.so(_ZN3RTF14FixtureManager5setupEv+0xc8)[0x7f01f6e625b8]
/usr/local/bin/../lib/libRTF.so(_ZN3RTF8TestSuit3runERNS_10TestResultE+0x7f1)[0x7f01f6e678c1]
/usr/local/bin/../lib/libRTF.so(_ZN3RTF10TestRunner3runERNS_10TestResultE+0x45)[0x7f01f6e661d5]
testrunner(main+0xa3c)[0x40898c]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f01f60e9830]
testrunner(_start+0x29)[0x409c49]
======= Memory map: ========
00400000-00426000 r-xp 00000000 00:31 2345                               /usr/local/bin/testrunner
00625000-00626000 r--p 00025000 00:31 2345                               /usr/local/bin/testrunner
00626000-00627000 rw-p 00026000 00:31 2345                               /usr/local/bin/testrunner
011a0000-01387000 rw-p 00000000 00:00 0                                  [heap]
7f01ec000000-7f01ec021000 rw-p 00000000 00:00 0 
7f01ec021000-7f01f0000000 ---p 00000000 00:00 0 
7f01f3c11000-7f01f3c1e000 r-xp 00000000 00:31 972                        /usr/local/lib/libYARP_rtf.so.2.3.69.9
7f01f3c1e000-7f01f3e1d000 ---p 0000d000 00:31 972                        /usr/local/lib/libYARP_rtf.so.2.3.69.9
7f01f3e1d000-7f01f3e1e000 r--p 0000c000 00:31 972                        /usr/local/lib/libYARP_rtf.so.2.3.69.9
7f01f3e1e000-7f01f3e1f000 rw-p 0000d000 00:31 972                        /usr/local/lib/libYARP_rtf.so.2.3.69.9
7f01f3e1f000-7f01f3e31000 r-xp 00000000 00:31 966                        /usr/local/lib/FtSensorTest.so
7f01f3e31000-7f01f4030000 ---p 00012000 00:31 966                        /usr/local/lib/FtSensorTest.so
7f01f4030000-7f01f4031000 r--p 00011000 00:31 966                        /usr/local/lib/FtSensorTest.so
7f01f4031000-7f01f4032000 rw-p 00012000 00:31 966                        /usr/local/lib/FtSensorTest.so
7f01f4032000-7f01f4045000 r-xp 00000000 00:31 2235                       /lib/x86_64-linux-gnu/libbsd.so.0.8.2
7f01f4045000-7f01f4244000 ---p 00013000 00:31 2235                       /lib/x86_64-linux-gnu/libbsd.so.0.8.2
7f01f4244000-7f01f4245000 r--p 00012000 00:31 2235                       /lib/x86_64-linux-gnu/libbsd.so.0.8.2
7f01f4245000-7f01f4246000 rw-p 00013000 00:31 2235                       /lib/x86_64-linux-gnu/libbsd.so.0.8.2
7f01f4246000-7f01f4247000 rw-p 00000000 00:00 0 
7f01f4247000-7f01f426c000 r-xp 00000000 00:31 41                         /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f01f426c000-7f01f446b000 ---p 00025000 00:31 41                         /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f01f446b000-7f01f446f000 r--p 00024000 00:31 41                         /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f01f446f000-7f01f4470000 rw-p 00028000 00:31 41                         /lib/x86_64-linux-gnu/libtinfo.so.5.9
7f01f4470000-7f01f44a1000 r-xp 00000000 00:31 2062                       /usr/lib/x86_64-linux-gnu/libedit.so.2.0.53
7f01f44a1000-7f01f46a1000 ---p 00031000 00:31 2062                       /usr/lib/x86_64-linux-gnu/libedit.so.2.0.53
7f01f46a1000-7f01f46a3000 r--p 00031000 00:31 2062                       /usr/lib/x86_64-linux-gnu/libedit.so.2.0.53
7f01f46a3000-7f01f46a4000 rw-p 00033000 00:31 2062                       /usr/lib/x86_64-linux-gnu/libedit.so.2.0.53
7f01f46a4000-7f01f46a8000 rw-p 00000000 00:00 0 
7f01f46a8000-7f01f46af000 r-xp 00000000 00:31 2234                       /lib/x86_64-linux-gnu/librt-2.23.so
7f01f46af000-7f01f48ae000 ---p 00007000 00:31 2234                       /lib/x86_64-linux-gnu/librt-2.23.so
7f01f48ae000-7f01f48af000 r--p 00006000 00:31 2234                       /lib/x86_64-linux-gnu/librt-2.23.so
7f01f48af000-7f01f48b0000 rw-p 00007000 00:31 2234                       /lib/x86_64-linux-gnu/librt-2.23.so
7f01f48b0000-7f01f4a36000 r-xp 00000000 00:31 2112                       /usr/lib/libACE-6.3.3.so
7f01f4a36000-7f01f4c36000 ---p 00186000 00:31 2112                       /usr/lib/libACE-6.3.3.so
7f01f4c36000-7f01f4c47000 r--p 00186000 00:31 2112                       /usr/lib/libACE-6.3.3.so
7f01f4c47000-7f01f4c48000 rw-p 00197000 00:31 2112                       /usr/lib/libACE-6.3.3.so
7f01f4c48000-7f01f4c4a000 rw-p 00000000 00:00 0 
7f01f4c4a000-7f01f4c8d000 r-xp 00000000 00:31 977                        /usr/local/lib/libYARP_math.so.2.3.69.9
7f01f4c8d000-7f01f4e8c000 ---p 00043000 00:31 977                        /usr/local/lib/libYARP_math.so.2.3.69.9
7f01f4e8c000-7f01f4e8d000 r--p 00042000 00:31 977                        /usr/local/lib/libYARP_math.so.2.3.69.9
7f01f4e8d000-7f01f4e8e000 rw-p 00043000 00:31 977                        /usr/local/lib/libYARP_math.so.2.3.69.9
7f01f4e8e000-7f01f4ecc000 r-xp 00000000 00:31 982                        /usr/local/lib/libYARP_sig.so.2.3.69.9
7f01f4ecc000-7f01f50cb000 ---p 0003e000 00:31 982                        /usr/local/lib/libYARP_sig.so.2.3.69.9
7f01f50cb000-7f01f50cc000 r--p 0003d000 00:31 982                        /usr/local/lib/libYARP_sig.so.2.3.69.9
7f01f50cc000-7f01f50cd000 rw-p 0003e000 00:31 982                        /usr/local/lib/libYARP_sig.so.2.3.69.9
7f01f50cd000-7f01f52bf000 r-xp 00000000 00:31 949                        /usr/local/lib/libYARP_OS.so.2.3.69.9
7f01f52bf000-7f01f54be000 ---p 001f2000 00:31 949                        /usr/local/lib/libYARP_OS.so.2.3.69.9
7f01f54be000-7f01f54c8000 r--p 001f1000 00:31 949                        /usr/local/lib/libYARP_OS.so.2.3.69.9
7f01f54c8000-7f01f54cb000 rw-p 001fb000 00:31 949                        /usr/local/lib/libYARP_OS.so.2.3.69.9
[... 47 lines cut ...]
7f01f7089000-7f01f709d000 r-xp 00000000 00:31 1264                       /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2
7f01f709d000-7f01f729d000 ---p 00014000 00:31 1264                       /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2
7f01f729d000-7f01f729e000 r--p 00014000 00:31 1264                       /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2
7f01f729e000-7f01f729f000 rw-p 00015000 00:31 1264                       /usr/lib/x86_64-linux-gnu/libtinyxml.so.2.6.2
7f01f729f000-7f01f72aa000 r-xp 00000000 00:31 988                        /usr/local/lib/libRTF_lua.so
7f01f72aa000-7f01f74a9000 ---p 0000b000 00:31 988                        /usr/local/lib/libRTF_lua.so
7f01f74a9000-7f01f74aa000 r--p 0000a000 00:31 988                        /usr/local/lib/libRTF_lua.so
7f01f74aa000-7f01f74ab000 rw-p 0000b000 00:31 988                        /usr/local/lib/libRTF_lua.so
7f01f74ab000-7f01f74d1000 r-xp 00000000 00:31 32                         /lib/x86_64-linux-gnu/ld-2.23.so
7f01f764b000-7f01f766e000 rw-p 00000000 00:00 0 
7f01f7691000-7f01f76bd000 rw-p 00000000 00:00 0 
7f01f76cd000-7f01f76d0000 rw-p 00000000 00:00 0 
7f01f76d0000-7f01f76d1000 r--p 00025000 00:31 32                         /lib/x86_64-linux-gnu/ld-2.23.so
7f01f76d1000-7f01f76d2000 rw-p 00026000 00:31 32                         /lib/x86_64-linux-gnu/ld-2.23.so
7f01f76d2000-7f01f76d3000 rw-p 00000000 00:00 0 
7fffd46f2000-7fffd4713000 rw-p 00000000 00:00 0                          [stack]
7fffd471c000-7fffd471e000 r--p 00000000 00:00 0                          [vvar]
7fffd471e000-7fffd4720000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)

travis_time:end:283d9b3b:start=1496060573393716564,finish=1496060575854288987,duration=2460572423
~~~

@diegoferigo any idea? Is this related to the inversion of the RTF <--> YARP dependency between master and devel?

diegoferigo, 2017-05-29:
This morning I triggered the generation of updated docker images, and currently master and devel Dockerfiles are different in order to handle the RTF <--> YARP dependency order. Until yesterday it was working, probably something is wrong with the most recent commits somewhere else.

diegoferigo, 2017-05-29:
@Nicogene I think you may know, it could be related to the changes you did recently on `yarpmanager`.

Nicogene, 2017-05-29:
The changes I made on `yarpmanager` are mainly related to the gui experience, but tomorrow I will investigate, unfortunately I don't know very much about the interaction between `RTF` and `yarpmanager`

Nicogene, 2017-05-29:
PR robotology/yarp#1239 and robotology/yarp#1240 should fix the issue, sorry guys again :sweat_smile:

diegoferigo, 2017-06-05:
After the regeneration of the docker images with @Nicogene fixes, all is up and running. Thanks!

closed 2017-06-05 by diegoferigo