robotology/gazebo-yarp-plugins #300: [gazebo + iCub] Simulating iCub faster than realtime

traversaro, 2017-06-05:
@emilio-cartoni commented on [Mon Jun 05 2017](https://github.com/robotology/QA/issues/227)

I would like to simulate the iCub on Gazebo faster than realtime, but it seems the plugins cannot work faster than 1kHz (realtime).

I have followed the tips here (#161) to achieve a fast simulation on Gazebo: namely I have removed all iCub parts except the torso and the left arm.
Gazebo now operates in realtime (Real Time Factor 1.00) instead of running at 0.30 when simulating the whole iCub.
However, I need to run very long simulations and a real time factor (RTF) higher than 1 would be highly desiderable.
Given that it's not possible to increase the max_step_size in Gazebo without messing with the PIDs (see #161), I've tried to increase the real_time_update_rate in Gazebo from 1.000 to 10.000.
The result is that Gazebo now runs at 3.5 Real Time Factor (limited by the CPU, in theory with 10k update rate and 0.001 step size it could go up to 10 RTF).
However, when running faster than real time, the iCub stops responding.
When the real_time_update rate is higer than 1000, if I connect to the iCub with yarpmotorgui or using some other controller (i.e. a custom c++ program) everything looks fine but the iCub does not move.
If I restore the real_time_update_rate to 1000, I can again move the iCub from the motorgui (without restarting the gui or anything else).
Even if I set a real_time_update_rate of 1001 (thus staying at basically 1.00 RTF) the iCub cannot be moved.
Interestingly, if I set  a real_time_update_rate of 500, thus slowing down the simulation to 0.5 RTF, I can move the iCub.
So it seems that the controllers are designed to work at a maximum of 1000 Hz - even 1001 Hz won't work.

Is there any workaround or setting to be able to achieve faster than real-time iCub simulations?

---

@traversaro commented on [Mon Jun 05 2017](https://github.com/robotology/QA/issues/227#issuecomment-306182248)

Hi @emilio-cartoni , 
  what you are describing seems indeed a bug, either in `gazebo`, in `gazebo-yarp-plugins` or in something YARP related. I will then move this issue to `gazebo-yarp-plugins` repository, that is probably the piece of software most related to this problem.

traversaro, 2017-06-05:
Hi @emilio-cartoni ,  
  can you try to run the Gazebo simulation with the YARP controller synched with the Gazebo simulation through the Clock plugin, as documented in  https://github.com/robotology/gazebo-yarp-plugins/tree/master/plugins/clock , and see if the problem still appears? Thanks.

emilio-cartoni, 2017-06-05:
Hi @traversaro,
I've done the following:
1) opened 3 terminals, in each of them I did YARP_CLOCK=/clock
2) on first terminal, I opened yarpserver
3) on second terminal, I opened gazebo with gazebo -slibgazebo_yarp_clock.so
The output of Gazebo shows correct initialization of clock (I guess):

> [DEBUG]GazeboYarpClock loaded. Clock port will be  /clock 
> yarp: Port /clock active at tcp://150.146.65.88:10002/
> yarp: Port /clock/rpc active at tcp://150.146.65.88:10003/

4) Then I loaded the iCub "arm only" on Gazebo (I've cut the torso as well).

5) On third terminal, I launched yarpmotorgui and moved the arm successfully

6) In Gazebo, I set real_time_update_rate to 2000....I try to move again the arm from the yarpmotorgui but it doesn't move

6) In Gazebo, I set real_time_update_rate back to 1000....I try to move again the arm from the yarpmotorgui and this time it works.

Did I use the Clock correctly or was something else needed?

traversaro, 2017-06-05:
> Did I use the Clock correctly or was something else needed?

You did use the Clock correctly. It seems there is a bug somewhere. 
I don't think anyone in IIT can debug this immediately. If you are willing to debug the issue yourself, we can provide some guidance in what check first.

barbalberto, 2017-06-05:
I think you shall run the gazebo without the YARP_CLOCK variable.
The rate in which wrappers will publish the data won't be speeded up (they will publish every 30ms), but the data and timestamps shall be correct. You can speed up publishing rate by chamgin the wrapper config file.

emilio-cartoni, 2017-06-05:
@traversaro Ok, I can try to pinpoint the issue better if you give me some pointers.

@barbalberto I'm not sure what do you mean by wrappers... you mean the controlboard plugins?
In the arm config files, e.g. gazebo_icub_left_arm_no_hand_for_no_hand_model I see that the "period" (# rate of output streaming from ports in ms) variable is set to 10 - is this what you are referring to?

traversaro, 2017-06-05:
@emilio-cartoni The first thing I would check is if the `GazeboYarpControlBoardDriver::onUpdate` (see https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/controlboard/src/ControlBoardDriver.cpp#L414) method is actually being called when you increase `real_time_update_rate`. If that method is being called, I would check if the `GazeboYarpControlBoardDriver::sendPositionToGazebo` is being called. 
If both this method are working fine, I would suspect that the problem is on the Gazebo level.

barbalberto, 2017-06-06:
I mean the frequency at which the port will publish the data. The controlBoard plugin will run at the correct speed up frequency

emilio-cartoni, 2017-06-06:
@traversaro Thanks, I've inserted some debug message on both functions and I can see that both `GazeboYarpControlBoardDriver::onUpdate` and `GazeboYarpControlBoardDriver::sendPositionToGazebo` are being called at `real_time_update_rate` (if I set r`eal_time_update_rate` to 1 I can see them called 1 per second, if I set real_time_update_rate to 3000 I can see them being called more than the standard 1000 per second).

@barbalberto I tried to set the `period` variable in `gazebo_icub_left_arm_no_hand_for_no_hand_model.ini` to 1 ms: logging from` icubSim/left_arm/state:o ` I do obtain faster messages (about 1000 msg/s as expected) but it doesn't affect the issue.

emilio-cartoni, 2017-06-06:
One further hint that might help to locate the problem:
- I tried to execute a movement with the yarpmotorgui and `real_time_update_rate` set to a low value (e.g. 100)
- while this movement was slowly executed (as time was slowed down to 0.1 RTF), I rised `real_time_update_rate` to 1001
- the iCub arm immediately dropped as if suddenly there was no force applied to it

traversaro, 2017-06-06:
I am start suspecting that there is a problem on the Gazebo level. 
`GazeboYarpControlBoardDriver::sendPositionToGazebo` is communicating with the low-level Joint Position control of Gazebo by sending messages over the custom transport system used by Gazebo, in particular it is sending commands through the `std::string("~/") + this->m_robot->GetName() + "/joint_cmd"` topic (see https://github.com/robotology/gazebo-yarp-plugins/blob/93cf6d4f4b5349a8e5942ae404b54dfec3725d76/plugins/controlboard/src/ControlBoardDriver.cpp#L307).

This message should be received by Gazebo and handled in the `OnJointCmd` (see https://bitbucket.org/osrf/gazebo/src/42215f2019f5ae64fc8a0610ef804a0fc53b0740/gazebo/physics/JointController.cc?at=default&fileviewer=file-view-default#JointController.cc-49 and https://bitbucket.org/osrf/gazebo/src/42215f2019f5ae64fc8a0610ef804a0fc53b0740/gazebo/physics/JointController.cc?at=default&fileviewer=file-view-default#JointController.cc-184 ). If you have the possibility of getting the Gazebo's debug symbols  of compiling Gazebo from source code, it would be great to debug what is going on in those functions (see https://github.com/robotology/gazebo-yarp-plugins/issues/165 for some background on this mechanism).

Which version of Gazebo are you using? This could be a related issue: https://bitbucket.org/osrf/gazebo/issues/2294/pid-controller-broken-with-non-default#comment-None .

traversaro, 2017-06-06:
> @traversaro This is getting ugly :) 

Welcome to the club. : ) 

> However, before compiling maybe I could just create a listener for the `.../joint_cmd` topic and see if the transport is working  and what we are sending to the JointController?

Good idea, I have a limited experience with the Gazebo transport system, but you could find some pointer about it in http://gazebosim.org/tutorials?tut=topics_subscribed&cat=transport .

emilio-cartoni, 2017-06-06:
Good news.
I modified the listener in the tutorial to listen to `/gazebo/default/iCub/joint_cmd `topic.
When `real_time_update_rate` is <= 1000 and we move the arm in yarpmotorgui, we send multiple JointCmd messages (I guess 1 per world update) from the start of the movement up to its completion
When `real_time_update_rate `is > 1000 instead, I only see 1 JointCmd message, then nothing else.

So I would guess that it's the plugin that it's not sending JointCmd messages.

I previously checked that `GazeboYarpControlBoardDriver::sendPositionToGazebo` was invoked, but not that the publish line was actually executed... now I noticed there is a nice if...

```
bool GazeboYarpControlBoardDriver::sendPositionToGazebo(int j,double ref)
{
    yInfo()<< deviceName << "sendPositionToGazebo was called:" << j ;
    gazebo::msgs::JointCmd j_cmd;

    if(ref != m_oldReferencePositions[j])
    {
        // yDebug() << "Sending new command: new ref is " << ref << "; old ref was " << m_oldReferencePositions[j] ;
        prepareJointPositionMsg(j_cmd,j,ref);
        m_jointCommandPublisher->WaitForConnection();
        m_jointCommandPublisher->Publish(j_cmd);
        m_oldReferencePositions[j] = ref;
    }
    return true;
}
```

traversaro, 2017-06-06:
Good catch. Apparently the if was added in https://github.com/robotology/gazebo-yarp-plugins/commit/9f9a4d9e0e708fd615aae0fbbb015aa67b899d0f by @barbalberto , perhaps he can comment on that. I don't know if that is still necessary. 

If you only see one message, I guess something is not working properly in the trajectory generation, i.e. in https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/controlboard/src/ControlBoardDriver.cpp#L477 . I already saw some places where there is some hidden assumption on the physics update rate, that probably need to be fixed.

emilio-cartoni, 2017-06-06:
I think the if just avoids sending again the same position when the joint is motionless (i.e. its calculated position is the same as the previous one) - it looks like a good idea to stop overflowing the gazebo topic with messages when not necessary.

Anyway, the problem is not really there, even commenting out the if doesn't work.
As you said, the problem is in the trajectory generation as this call

>                 m_jntReferencePositions[j] = m_trajectory_generator[j]->computeTrajectory();

is always returning the same value when `real_time_update_rate` > 1000

traversaro, 2017-06-06:
Note that if you send a new desired position **after** you updated the `real_time_update_rate`, the trajectory  stuff should work fine because `initTrajectory` is correctly called in 
https://github.com/robotology/gazebo-yarp-plugins/blob/93cf6d4f4b5349a8e5942ae404b54dfec3725d76/plugins/controlboard/src/ControlBoardDriverPositionControl.cpp#L19 . I wonder if this line (for getting the "controller period") is working correctly. 
If you want to change the trajectory generator for debug, you can change it by changing the `TRAJECTORY_GENERATION::trajectory_type` parameter, for example in https://github.com/robotology/icub-gazebo/blob/master/icub/conf/gazebo_icub_left_arm_no_hand.ini#L21 .

emilio-cartoni, 2017-06-06:
Starting the movement **after** the update to `real_time_update_rate` doesn't work either :)

Yeah, there are some suspicious lines....
... I'm currently looking at `MinJerkTrajectoryGenerator` in `ControlBoardDriverTrajectory`

emilio-cartoni, 2017-06-06:
Ok I just went by instinct and changed this:

```
bool MinJerkTrajectoryGenerator::initTrajectory (double current_pos, double final_pos, double speed)
{
    m_mutex.wait();
    m_controllerPeriod = (unsigned)(this->m_robot->GetWorld()->GetPhysicsEngine()->GetUpdatePeriod() * 1000.0);

```
to this:

```
bool MinJerkTrajectoryGenerator::initTrajectory (double current_pos, double final_pos, double speed)
{
    m_mutex.wait();
    m_controllerPeriod = (unsigned)(this->m_robot->GetWorld()->GetPhysicsEngine()->GetUpdatePeriod() * this->m_robot->GetWorld()->GetPhysicsEngine()->GetRealTimeUpdateRate());

Now it works!
```

emilio-cartoni, 2017-06-06:
It seems that the problem is that `m_controllerPeriod `is an `unsigned int`, so when real_time_update_rate is > 1000 the resulting UpdatePeriod of the PhysicsEngine is below 0.001 and the whole expression that I changed above is less than 1, so it becomes a 0 unsigned int.

The "fix" that I applied (1000.0 looked suspiciously as the update_rate - so I substituted it) simply forces the expression to be always 1 but this is probably not the intended behavior, even tought it apparently "fixes" the problem as the arm can now be moved with real_time_update_rate > 1000 :)

traversaro, 2017-06-06:
Thanks a lot @emilio-cartoni  ! 
Indeed, the fact that `m_controllerPeriod` is an `unsigned int` seems to be a bug, that can be easily fixed by converting it to a `double`. 
Furthermore, there are around ~8 usage of the `1000.0` constant in the `ControlBoardDriverTrajectory.cpp`. I will investigate the meaning of such a constant with the original author of the code.

traversaro, 2017-06-06:
After taking a look at the code, I think the `1000.0` is there because `m_controllerPeriod` is expressed in milliseconds rather then seconds as in the rest of the code. Clearly this can be cleaned up, but the actual fix of the bug is indeed to convert `m_controllerPeriod` to be a `double`.

If you want to open a PR with this change @emilio-cartoni you are welcome to do that, otherwise we will do it when we have a bit of time on our side. 

Thanks again for the in-depth debug!

emilio-cartoni, 2017-06-07:
> Thanks again for the in-depth debug!

Thanks for the pointers, it was a joint effort!

I can make the PR but I noticed that while the robot now moves with `real_time_update_rate` > 1000, the issue isn't totally fixed yet.
Infact, now that the robot moves I noticed that:
1) a full movement* at RTF 1 with 1000 update rate takes about 8 real seconds
2) a full movement at RTF 2 with 2000 update rate takes about 8 real seconds 
3) a full movement at RTF 4.1 (CPU limited) with 41000 update rate takes about 80 real seconds (!)

* e.g. moving the shoulder_pitch joint from one limit to the other at speed 10

So it seems to me that `m_controllerPeriod` actually scales the speed of the movement so that it remains constant with respect to the real time.
While this may help with respect to a controller that expects the movement to be completed in a certain amount of real time, it is doesn't make sense with:
- the fact that the purpose of a simulation going faster than realtime is that you would like things to happen faster
- the fact that Yarp controllers should be aware of the simulation time and not try to keep things going according to real time (see also the clock we were discussing earlier)

Also, even if the intended behavior was to keep things constant with respect to real time, the actual code doesn't work anyway since it is tied to the `real_time_update_rate` setting and not to the actual passing of simulation time - so when the simulation is CPU limited things will start going slower than real time anyway (see case 3).

The first fix that comes to my mind is to simply get rid of m_controllerPeriod.
Indeed by putting m_controllerPeriod = 1, things go nicely (they speed up in proportion to how fast simulation time is passing, independently of the update_rate setting).
However, m_controllerPeriod was there for a reason (maybe imported from some other implementations of Minimum Jerk trajectory generation?) so it would be good to have the opinion of someone who developed that code to know if we are breaking something else by removing m_controllerPeriod :)

traversaro, 2017-06-07:
> maybe imported from some other implementations of Minimum Jerk trajectory generation?

Exactly, that is the case. 

> So it seems to me that m_controllerPeriod actually scales the speed of the movement so that it remains constant with respect to the real time.

I don't understand. Did you get rid of all other `1000.0` spread around in the code?

emilio-cartoni, 2017-06-07:
>>    So it seems to me that m_controllerPeriod actually scales the speed of the movement so that it remains constant with respect to the real time.

> I don't understand. Did you get rid of all other 1000.0 spread around in the code?

Nope, I didn't change the 1000.0 spread around in the code.

Initially, I just changed m_controllerPeriod from unsigned int to double, then I got the behavior described in the cases 1), 2), 3).

Then, to "fix" the fact that the movement remains constant respect to real time (unless when simulation is CPU limited), I tried to put m_controllerPeriod = 1 (which is equivalent to removing it since it's used as a multiplier). 

Notice that under normal conditions this->m_robot->GetWorld()->GetPhysicsEngine()->GetUpdatePeriod() equals to 0.001 so m_controllerPeriod = 0.001 * 1000.0 = 1.0;

Indeed by putting in the code m_controllerPeriod = 1 the simulation works as expected regardless of the `real_time_update_rate`: the robot moves faster when simulation time goes faster than real time and viceversa.

emilio-cartoni, 2017-06-12:
After reading the code again, it seems to me that GetUpdatePeriod()  should be replaced with GetMaxStepSize().
Under "normal" conditions (0.001 max step size), this gives m_controllerPeriod = 0.001*1000.0 = 1.0 which is well behaved.
Compared to simply removing m_controllerPeriod (or setting it to 1.0 constant) this also gives the nice feature that movements in the simulation can be speed up by changing the step size.
I think this also reflects the original intentions of the  m_controllerPeriod variable,  as changing the step size changes how frequent in sim time the plugin ("the controller") is called.

I have also done a check using trajectory_type constant_speed and everything seems to works ok (changing both real_time_update_rate and max_step_size*).

I've made a fork and I'll probably send a PR as soon as I resolve a build issue that I've encountered after git pulling the latest master commit.

Interestingly, in the simulations using one arm only, I was able to change max_step_size to 0.002 without encountering the PID issues mentioned in   robotology/QA#161

traversaro, 2017-06-12:
I think that @randaz81 can comment on this.

emilio-cartoni, 2017-06-12:
Opened PR :)