robotology/gazebo-yarp-plugins #305: gazebo-yarp | videotexture plugin loaded twice

pattacini, 2017-06-14:
@charleswilmot commented on [Thu May 11 2017](https://github.com/robotology/QA/issues/222)

Hello all,

I'm experiencing a problem with the video texture plugin.
Alone, the plugin is working fine. Issue arise when trying to use the plugin together with a model of the iCub in Gazebo. As explained in [this](http://answers.gazebosim.org/question/15939/visual-plugin-loaded-twice/?answer=15943#post-id-15943) thread, the plugin is loaded once on the gzserver and once in the gzclient, resulting in conflict with port access.

Is there a way to use the plugin, camera sensors and the GUI at the same time?

Best regards

---

@traversaro commented on [Thu May 11 2017](https://github.com/robotology/QA/issues/222#issuecomment-300715494)

Probably @aerydna is the best person to answer this. 

---

@charleswilmot commented on [Thu May 11 2017](https://github.com/robotology/QA/issues/222#issuecomment-300737899)

I kind of hacked the code to make it work. This is how it looks like:

```c++
m_texName      = "/"+m_model->GetName();
if (yarp::os::NetworkBase::exists(m_texName))
{
    m_texName = m_texName + "_2";
}

m_VideoPort.open(m_texName);
```

---

@aerydna commented on [Mon May 22 2017](https://github.com/robotology/QA/issues/222#issuecomment-303123578)

hi @charleswilmot,
i can't reproduce the problem here. can you please give me all the details about the issue? which icub model are you loading in gazebo? the video texture plugin is loaded via the examble in the tutorial or in a custom sdf? thanks in advance..

---

@pattacini commented on [Mon May 29 2017](https://github.com/robotology/QA/issues/222#issuecomment-304646193)

Closing for now; feel free to add more info if needed.

---

@charleswilmot commented on [Tue May 30 2017](https://github.com/robotology/QA/issues/222#issuecomment-304851566)

Hello @aerydna ,

And sorry for the delay. Here is a screenshot of the problem:

[image]

I tried to display white noise on a screen. As you can see, the so called 'Figure 1' recieved from the robot's camera shows the white noise, whereas the GUI displays uninitialized graphic memory area. The reason is that the plugin is loaded once by the server and once by the client. It tries to open two times the same address, resulting in a conflict.

Here is the custom sdf file I'm using for the screen:
(only difference is that I removed the 'static' part)

```xml
<?xml version='1.0'?>
<sdf version="1.4">
<model name="my_name">
  <pose>2 0 2 1.57 0 -1.57</pose>
  <!--static>true</static-->
    <link name="my_name_link">
      <pose>0 0 0 0 0 0</pose>
      <visual name="my_name_visual">
      <geometry>
        <plane>
            <size>1 1</size>
            <normal>0 0 1</normal>
        </plane>
      </geometry>
      <material>
      <diffuse>1.0 1.0 1.0 1.0</diffuse>
      </material>
      <plugin name="VideoTexture" filename="libgazebo_yarp_videotexture.so">
      <yarpConfigurationFile>file:///tmp/screen/vtcfg.ini</yarpConfigurationFile>
      </plugin>
      </visual>
      <gravity>0</gravity>
    </link>
  </model>
</sdf>
```

The icub model I'm using is the one I found on the Robotology GitHub:
[icub_fixed](https://github.com/robotology/icub-gazebo)

I'm using Gazebo 7.7.0

Best,
Charles

---

@charleswilmot commented on [Tue May 30 2017](https://github.com/robotology/QA/issues/222#issuecomment-304853728)

Here is a simple work-around that worked for me:

[VideoTexture.cc](https://gist.github.com/charleswilmot/bf6dbdd183e01244cd58721037291b52)

---

@aerydna commented on [Wed Jun 14 2017](https://github.com/robotology/QA/issues/222#issuecomment-308501503)

i'm on the issue.. when loading the default sdf no problem appears.. by opening your sdf ogre (and so gazebo) crashes. so probably, underneath, there are more serious issues than port conflicts. however i think all this to fit better in the issue section of gazebo-yarp-plugin repository (https://github.com/robotology/gazebo-yarp-plugins) than Q&A

vvasco, 2019-10-01:
Hi, I'm recovering this issue because it affects `cer-sim` with face expressions enabled. 
The test I did is the following: I ran gazebo and inserted the model [cer](https://github.com/robotology/cer-sim/blob/master/gazebo/cer/cer.sdf). 
This is the output that I get:
[image]

The [videotexture](https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/videotexture/src/VideoTexture.cc) plugin opens the port which reads the face expressions in the load function. 
As you can see, the load function seems to be called 3 times, the first time it opens the port `/default::SIM_CER_ROBOT::head::link::Visual_0`, the second and the third times it opens twice the same port `/SIM_CER_ROBOT::head::link::head_link_visual `, which produces an address conflict. 

cc @traversaro 

gazebo version: 9.0.0
yarp version: 3.2.101+20190926.15+git79d5a2246
gazebo-yarp-plugins: commit bb47c42f0c798873ef6a74c008346211e998e7ad
cer-sim: commit 54d8032fd923d19053933e8bda9be30b3aa708aa
system: Ubuntu 18.04.2 LTS

traversaro, 2019-10-01:
The answer in http://answers.gazebosim.org/question/15939/visual-plugin-loaded-twice/?answer=15943 is quite clear w.r.t. to this: basically a VisualPlugin needs to be loaded once in the gzserver (to be rendered correctly by camera sensors) and once in the gzclient (to be rendered correctly in the user GUI), so the `gazebo_yarp_videotexture` plugin needs to account for  this, and make sure that the same image is rendered in both instances of the plugin. A possible solution to handle this would be to create a YARP topic (see https://www.yarp.it/persistent_connections.html#persistent_connections_topics) for each couple of plugins, and let the two ports of the plugins subscribe to that topic. 

I do not know why the plugin is instantiated three times in @vvasco's example.

traversaro, 2019-10-01:
@aerydna I have unassigned you as I guess you are not going to handle this soon, let me know if instead you prefer to be assigned.

aerydna, 2019-10-02:
though i don't partecipate to the conversation i'm following it. I see the problem. sadly as you say i cannot handle it at the moment and i cannot say when i will able to