robotology/gazebo-yarp-plugins #315: Resetting iCub in Gazebo from C++

serena-ivaldi, 2017-07-12:
Hi all

we are trying to make the iCub in Gazebo learn by reinforcement learning some WB behaviors that frequently fail and cause the robot to fall/crash in Gazebo
I was looking for the correct procedure to send the reset() to the world in Gazebo and to the controlBoards of the iCub but I cannot find the information.

Could you help us?

We are calling that from C++.

If we manually reset Gazebo (with CTRL+R) the world is reset, the iCub ports are still there but we basically loose the connection with the iCub motor controllers and we cannot move it anymore.

Thanks!

@lothas

traversaro, 2017-07-12:
Hi @serena-ivaldi , 
  which version of Gazebo are you using? 
As documented in http://robotology.gitlab.io/docs/gazebo-yarp-plugins/master/troubleshooting.html , we had problems in the past with the reset world of Gazebo < 7.2 . And recently @diegoferigo experienced problems with the latest Gazebo 7, and he had to update to Gazebo 8 to get the reset world to work.

diegoferigo, 2017-07-12:
Adding more details, the only issue still present with Gazebo 8 is the following:
```
[ERROR]An hardware fault occurred on joint  2  torque too big!
```
But on this version, removing iCub and adding it again does work, Gazebo 7 always crashed (cc @nunoguedelha).

For what concern the C++ side, let's wait someone that has more experience (than me at least) in dealing with its code.

traversaro, 2017-07-12:
The hardware fault error is when a joint has a torque value bigger than 2000 https://github.com/robotology/gazebo-yarp-plugins/blob/93644b645efc56ce124e65b80f64b92dcc1a7cdb/plugins/controlboard/src/ControlBoardDriver.cpp#L69 .

lothas, 2017-07-13:
We are running Gazebo 7.7.
Right now we are solving the issue by creating and destroying all the ports between two resets, but this is not optimal.

traversaro, 2017-07-13:
What you mean "creating and destroying all the ports" ?

lothas, 2017-07-13:
Opening and connecting the ports, starting the threads etc.
Then doing the opposite.

traversaro, 2017-07-13:
Mhh, are you using the `YARP_CLOCK` enviromental variable and the `libgazebo_yarp_clock.so` plugin?

lothas, 2017-07-13:
No. What would I use that for?
I've attached the code below.
[icub_yarp_gazebo.txt](https://github.com/robotology/gazebo-yarp-plugins/files/1145963/icub_yarp_gazebo.txt)

So we call connectGazebo(), run an episode, then disconnectGazebo() and reset Gazebo (CTRL+r). Rinse and repeat.