robotology/gazebo-yarp-plugins #316: Torques references in gazebo simulation with differents URDF

InriaBrice, 2017-07-17:
Hi,

We are trying to do a torque control with a new QP framework using the gazebo-yarp-plugins, and we have an other URDF than icub-gazebo. So we sent the command with : 

`bool ctl = itrq->setRefTorques(command.data());`

it works properly, then when we try to read the torques references with : 

`trqRightArm_->getTorques(torquesRightArm.data());`

The values doesn't look like the same. So my questions are : 

How the values are computed from the force/torques sensors ?
Do they use a different URDF or one that i can set in a init file or parameter ?
Should i use wholebodydynamicstree or something like this to use theses functions ?

Best regards,
Brice.

traversaro, 2017-09-08:
Hi @InriaBrice , sorry for the late reply. 

The use of `setRefTorques` and `getTorques` is indeed correct, and there is not need to use any additional module when performing torque control on the iCub simulated in Gazebo. The issue you are describing seems similar to https://github.com/robotology/gazebo-yarp-plugins/issues/238 . 
Unfortunately until now we never investigated it in depth, but I suspect it is a Gazebo issue.

InriaBrice, 2017-09-08:
Thanks for your reply @traversaro . 

Ok so maybe we could try in an other simulator to see what happen.

traversaro, 2017-12-15:
Closing as duplicate of https://github.com/robotology/gazebo-yarp-plugins/issues/238 .

closed 2018-01-08 by traversaro