robotology/gazebo-yarp-plugins #332: Remove hardcoded parameters in controlboard plugin

traversaro, 2017-12-27:
There are a few hardcoded parameters in the controlboard plugin, including:
* The `maxTorques` vector, used to decide when the joint goes in hardware fault mode, hardcoded to 2000 ( https://github.com/robotology/gazebo-yarp-plugins/blob/master/plugins/controlboard/src/ControlBoardDriver.cpp#L69 ) )

* There are also the parameters:
~~~
const double RobotPositionTolerance_revolute = 0.9;      // Degrees
const double RobotPositionTolerance_linear = 0.004; // Meters
~~~
but with a preliminary check it seems to be that those values are unused. In that case they can be directly removed.

triccyx, 2018-01-08:
What is the correct section of *.ini to put  maxTorques ?

triccyx, 2018-01-08:
const double RobotPositionTolerance_revolute = 0.9;      // Degrees
const double RobotPositionTolerance_linear = 0.004; // Meters

are used in:

bool GazeboYarpControlBoardDriver::configureJointType()

traversaro, 2018-01-08:
> What is the correct section of *.ini to put maxTorques ? 

I would go in the generic section, similar to what we do for `min_damping` and `min_stiffness` : 
https://github.com/robotology/icub-gazebo/blob/master/icub/conf/gazebo_icub_left_leg.ini#L32
(even if this one should be eventually moved in `IMPEDANCE_CONTROL` . : )

> bool GazeboYarpControlBoardDriver::configureJointType() 

Great, then we should add parameters in the configuration file also for them.

triccyx, 2018-01-08:
What should be the policy if a param is wrong/bad formatted?
keep on analyze the next params?

traversaro, 2018-01-08:
In general I would fail early to make sure that the wrong parameter is properly noted by the model author. In particular in this context I would keep the code consistent with the rest of the plugin, and return false in the `gazebo_init` function.

triccyx, 2018-01-08:
Ok, I ask this because the gazebo_init has some return false, but sometimes the submethods are always true see for example setMinMaxImpedance.
For the new ones I'll follow your suggestion

PR #334 (2018-01-10): Removing hardcoded parameters in gazebo_yarp_controlboard plugin 
Fix #332
closed 2018-01-10 by traversaro