robotology/gazebo-yarp-plugins #333: Weird behavior from ControlBoardDriver

lothas, 2018-01-08:
I'm running a simple experiment with a 2D model (a box attached to one vertical and one horizontal prismatic joints) in Gazebo. I'm controlling it with the gazebo yarp plugin.

I am seeing two (probably related) problems:
* When setting a new command (setPositions on IPositionDirect) the box will head to the new set-point as expected, but then it overshoots and comes back very very slowly. I've played with the PID values in the ini file and that doesn't help (my values are kp,kd,ki = 2, 5, 0). The problem is less noticeable with a reduced kd, and the box moves much slower in this case (which is weird).
* When moving between 2 set-points (say, (x,z)=(0,5) and (x,z)=(10,5)), if the box encounters an obstacle going to the right, it will overshoot way more when going to the left, as if it's accumulating some error while trying to reach the target on the right (although ki = 0...) 

(I was also getting an error with a high torque before which could be related, see issue #331 )
https://github.com/robotology/gazebo-yarp-plugins/issues/331

traversaro, 2018-01-08:
Hi @lothas , 
   can you provide the model (and ideally also the code) that you are using for your tests?

lothas, 2018-01-08:
I'll try to prepare some minimal code and upload it here

lothas, 2018-01-09:
Hi Silvio, here's the minimal code that you asked for. Let me know if you have problem building it.

There's a box_description folder with the box model for gazebo and a robot_framework folder for the yarp-gazebo bridge and a reset plugin. You should build and build install.
Just update the path for the yaml config file in box_simple_main.cpp.

[min_code_for_yarp.tar.gz](https://github.com/robotology/gazebo-yarp-
plugins/files/1615107/min_code_for_yarp.tar.gz)