robotology/gazebo-yarp-plugins #335: Code quality

triccyx, 2018-01-08:
I suggest to write a (very) small document in order to give some indications in writing c++ code.
For example I notice that in some parts of the code there is a large use of c arrays that should be avoided in c++. While in others parts the std::vector are correctly used.
Perhaps just a small guide document can help and avoid wrong Copy&Paste from old style code.

traversaro, 2018-01-09:
I agree, I have two comments about this: 
* It should be a document written at a robotology level, not at the project level. 
* There are a lot of good resources for this kind of material. In particular, the Cpp Core Guidelines are a sort of de-facto standard for C++ : http://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines . Given that in this document I would mostly provided pointers to relevant material, for example what you mention is described in [Cpp Core Guideline E.6](http://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#Re-raii).

triccyx, 2018-01-09:
Ok for me!
Three other things:
*Just I would not use the doc as it is but write down some existential points.  In my experience people are not able to use such a document too big and complex (as they are not professional code writer). Also some points of the doc are not existential.
*Before write SW in the project is a must read our guidelines.
*In my opinion also a naming convention doc is essential.

traversaro, 2018-01-09:
> Just I would not use the doc as it is but write down some existential points. In my experience people are not able to use such a document too big and complex (as they are not professional code writer). Also some points of the doc are not existential. 

I totally agree on this. A summary of essential points + link to a more in-depth document are the best format. 

> Before write SW in the project is a must read our guidelines.
> In my opinion also a naming convention doc is essential.

For both this points, in the past we had problems in enforcing this at the iCub Facility-wide level from the organizational point of view. 
I think the compromise we found for the code style in https://github.com/robotology/human-dynamics-estimation/issues/52#issuecomment-308710675 is worth embracing also for naming and guidelines in general: we collaborate at the central on shared documents, that can be optionally be enforced by the maintainer of a given repo.