robotology/gazebo-yarp-plugins #343: Erron on cmake in superbuild

triccyx, 2018-01-31:
Using ubuntu 16.04 and gazebo 9.0.0
Using cmake .. on build folder the following errors occours:

 FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/ignition-cmake0/cmake0/FindFreeImage.cmake:46 (try_run):
  Unknown extension ".cc" for file

    /home/ltricerri/isaac-superbuild/build/CMakeTmp/test_freeimage.cc

  try_compile() works only for enabled languages.  Currently these are:

    C NONE

  See project() command to enable other languages.
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:65 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/ignition-common1/ignition-common1-config.cmake:109 (find_dependency)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:203 (find_package)
  cmake/BuildGazeboYARPPlugins.cmake:11 (find_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:231 (include)
  cmake/BuildISAAC.cmake:23 (find_or_build_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:231 (include)
  CMakeLists.txt:70 (find_or_build_package)

-- FreeImage test failed to compile - This may indicate a build system bug
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:203 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/ignition-common1/ignition-common1-config.cmake

  but it set ignition-common1_FOUND to FALSE so package "ignition-common1" is
  considered to be NOT FOUND.  Reason given by package:

  ignition-common1 could not be found because dependency FreeImage could not
  be found.

Call Stack (most recent call first):
  cmake/BuildGazeboYARPPlugins.cmake:11 (find_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:231 (include)
  cmake/BuildISAAC.cmake:23 (find_or_build_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:231 (include)
  CMakeLists.txt:70 (find_or_build_package)

-- Configuring incomplete, errors occurred!
See also "/home/ltricerri/isaac-superbuild/build/CMakeFiles/CMakeOutput.log".
See also "/home/ltricerri/isaac-superbuild/build/CMakeFiles/CMakeError.log".

triccyx, 2018-01-31:
As a workaround you can try to comment out
/usr/lib/x86_64-linux-gnu/cmake/ignition-cmake0/cmake0/FindFreeImage.cmake
from line 49 to 64
the part for check FreeImage version

traversaro, 2018-07-17:
The problem is that in that superbuild the `enable_language(CXX)` command was missing in the CMake configuration of the project. See https://github.com/robotology/how-to-export-cpp-library/issues/25 for the details of the issue, and https://bitbucket.org/ignitionrobotics/ign-cmake/pull-requests/44 for the issue in the FindFreeImage file.

closed 2018-07-17 by traversaro