robotology/gazebo-yarp-plugins #348: Control Board wrong joint limits for fingers

xEnVrE, 2018-02-28:
The `.ini` configuration files for fingers, e.g. [gazebo_icub_left_hand_index.ini](https://github.com/robotology/icub-gazebo/blob/master/icub/conf/gazebo_icub_left_hand_index.ini) contain a group `[LIMITS]` that specifies limits for each __physical__ joint. These limits are loaded within the control board plugin and used to, e.g., specify constraints on the generation of trajectories (see [here](https://github.com/robotology/gazebo-yarp-plugins/blob/5f595c10321b7afae4cc4c89f23fc6c8c6a23493/plugins/controlboard/src/ControlBoardDriver.cpp#L394)).

Since trajectories are generated for each controlled DoF of the finger and not for each physical DoF, e.g. for the index `l_index_proximal` and ` l_index_distal`, but the variable `m_jointPosLimits` is not updated to take into account coupling then joint limits used by the trajectory generator are wrong.

Could it be an option to add a method like 

````
decouplePosLimits(std::vector<Range> &posLimits)
````
to the decoupling handlers `ThumbCouplingHandler`, `IndexCouplingHandler`, `MiddleCouplingHandler`, `PinkyCouplingHandler`

and call it in the method [`GazeboYarpControlBoardDriver::setMinMaxPos`](https://github.com/robotology/gazebo-yarp-plugins/blob/5f595c10321b7afae4cc4c89f23fc6c8c6a23493/plugins/controlboard/src/ControlBoardDriver.cpp#L629)?

Or is it better to change the meaning of the section `[LIMITS]` so that it contains the limits of the controlled DoFs instead that of all the physical DoFs?

traversaro, 2018-02-28:
cc @randaz81 @triccyx

randaz81, 2018-02-28:
> decouplePosLimits(std::vector<Range> &posLimits)
to the decoupling handlers ThumbCouplingHandler, IndexCouplingHandler, MiddleCouplingHandler, PinkyCouplingHandler

seems the right solution to me. 
But please allow me some time to check the current status of the code

traversaro, 2018-05-23:
Any updated on this @randaz81 ? Do you think we can proceed with the solution proposed by @xEnVrE ?

traversaro, 2019-01-21:
@xEnVrE I think you can go forward and open a PR for  this on `icub-gazebo`, @randaz81 can comment on the PR before if he has any additional comment.

xEnVrE, 2019-01-21:
@traversaro I'm planning to do this very soon.

traversaro, 2020-03-06:
cc @ale-git given your recent work on Mk3 hand you may be interested in this issue.

xEnVrE, 2023-09-08:
Can we close this given that the above PR was merged? 

cc @traversaro

traversaro, 2023-09-08:
Ok, thanks!

closed 2023-09-08 by traversaro