robotology/gazebo-yarp-plugins #349: Resetting a YARP controlboad to a different joint configuration

traversaro, 2018-03-05:
@lothas commented on [Fri Mar 02 2018](https://github.com/robotology/yarp/issues/1571)

I wrote a gazebo plugin a while ago to reset the simulation and reset my model to a desired position and/or joint configuration. Up until now it seemed that everything worked OK because I was always resetting to the initial configuration stated in the sdf model file.
I'm now trying to reset my model to a different desired joint configuration. My gazebo plugin correctly updates the model but the gazebo-yarp-plugin always resets to the value specified in the sdf file, quickly driving the joint positions to those ones.
Is there a way of programmatically specifying the desired "OnReset" configuration for a yarp control board?

---

@jgvictores commented on [Mon Mar 05 2018](https://github.com/robotology/yarp/issues/1571#issuecomment-370446516)

Maybe [yarp::dev::IEncoders::setEncoders](http://www.yarp.it/classyarp_1_1dev_1_1IEncoders.html#a8c5adefdc8c0627a7ecfe7f6d972deaa) is what you need?

---

@traversaro commented on [Mon Mar 05 2018](https://github.com/robotology/yarp/issues/1571#issuecomment-370447750)

Hi @lothas , I am afraid the "reset to arbitrary configuration" functionality is not supported in gazebo-yarp-plugins (moving the issue there as it is a gazebo-related problem).

traversaro, 2018-03-05:
@jgvictores I am not sure about the intend semantics of the [yarp::dev::IEncoders::setEncoders](http://www.yarp.it/classyarp_1_1dev_1_1IEncoders.html#a8c5adefdc8c0627a7ecfe7f6d972deaa) function.

traversaro, 2018-03-05:
Resetting a Gazebo Model is possible at the Gazebo level using [`Joint::SetPosition`](http://osrf-distributions.s3.amazonaws.com/gazebo/api/dev/classgazebo_1_1physics_1_1Joint.html#ad02e021e1745f416f45e13024925714a) , but resetting to an user-provided position at runtime is not currently supported in the gazebo_yarp_controlboard plugin.

lothas, 2018-03-05:
Yeah, I figured that much by looking at the code. I hoped that there would be a way to change the yarp::os::Property m_parameters somehow. Oh well, I'll stick to just resetting to the default values.

traversaro, 2018-11-17:
@lothas If you are still interested in this, I guess the discussion in https://github.com/robotology/gazebo-yarp-plugins/issues/392 could be related .