robotology/gazebo-yarp-plugins #350: Error with gazebo-config.cmake

mhubii, 2018-03-06:
Hi,
when I try to compile the plugins I get an error related to the gazebo-config.cmake file located in /usr/lib/x86_64-linux-gnu/cmake/gazebo
1) it does not find  Bullet > 2.82 (line 95)
2) when setting GAZEBO_HAS_BULLET to FALSE we run into a Package not found error for OGRE.
Gazebo itself runs normally.
Thanks

mhubii, 2018-03-06:
1) The problem could be resolved by setting GAZEBO_HAS_BULLET to FALSE in /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake 

and

 2) building the gazebo yarp plugin via cmake .. -DCMAKE_MODULE_PATH=/usr/share/OGRE/cmake/modules

traversaro, 2018-03-06:
Hi @mhubii , as ` /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake` is a file installed by `.deb` packages, it is strictly recommended to not edit it manually.  : )

Can you tell us a bit about this problem? What is the OS? Did you installed Gazebo following the official instructions http://gazebosim.org/tutorials?tut=install_ubuntu or in some other way? Thanks.

mhubii, 2018-03-07:
Hi and thanks for the reply Silvio,

the OS is Ubuntu 16.04 and I installed gazebo8 and libgazebo8-dev according to the alternative installation, stated on this side. Gazebo itself runs well but the gazebo-yarp-plugins refuse to build following [these installation instructions](http://robotology.gitlab.io/docs/gazebo-yarp-plugins/master/install.html)

traversaro, 2018-03-07:
Can you provide us the exact error message? I suspect for some reason the bullet packages was uninstalled even if it is a dependency of the gazebo package.

mhubii, 2018-03-07:
The full error message is the following

`
/gazebo-yarp-plugins/build$ cmake ..
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   serialization
--   system
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "2.6.1") 
-- Checking for module 'bullet>=2.82'
--   No package 'bullet' found
-- Checking for module 'bullet2.82>=2.82'
--   No package 'bullet2.82' found
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:95 (message):
  Error: Bullet > 2.82 not found, please install libbullet2.82-dev.
Call Stack (most recent call first):
  CMakeLists.txt:23 (find_package)
`

traversaro, 2018-03-07:
Is the package `libbullet-dev` installed? I suspect for some reason it is not listed anymore as a dependency, so as long as you have it installed on your machine, you do not notice any difference, but this is breaking the installation for new users. 

If this is the problem, the old Travis-system would have catched it, but the new docker-based Travis has hided it instead, unfortunately. cc @diegoferigo

mhubii, 2018-03-07:
Yes it is installed, thanks for spending your time.

traversaro, 2018-03-07:
This is really strange. Can you paste the content of the file `/usr/lib/x86_64-linux-gnu/pkgconfig/bullet.pc` and the output of `env` on the terminal in which you launch the compilation?

traversaro, 2018-03-07:
I just tried with the following Dockerfile: https://gist.github.com/traversaro/723ee718e30315303c3a1f70cf267812 and everything is working fine. For this reason, I am afraid there is something corrupted in your machine, for example the gazebo or the bullet installation.

mhubii, 2018-03-07:
The `/usr/lib/x86_64-linux-gnu/pkgconfig/bullet.pc``holds

```
Name: bullet
Description: Bullet Continuous Collision Detection and Physics Library
Requires:
Version: 2.83
Libs: -L/usr/lib/x86_64-linux-gnu -lBulletSoftBody -lBulletDynamics -lBulletCollision -lLinearMath
Cflags:  -I/usr/include/bullet -I/usr/include
```

and the `env` prints

```
/gazebo-yarp-plugins/build$ env
XDG_VTNR=7
LC_PAPER=en_US.UTF-8
LC_ADDRESS=en_US.UTF-8
XDG_SESSION_ID=c2
XDG_GREETER_DATA_DIR=/var/lib/lightdm-data/mhuber
LC_MONETARY=en_US.UTF-8
CLUTTER_IM_MODULE=xim
GPG_AGENT_INFO=/home/mhuber/.gnupg/S.gpg-agent:0:1
TERM=xterm-256color
VTE_VERSION=4205
SHELL=/bin/bash
QT_LINUX_ACCESSIBILITY_ALWAYS_ON=1
GAZEBO_MODEL_PATH=:/home/mhuber/icubheidelberg01:/home/mhuber/icub-gazebo
WINDOWID=52439633
LC_NUMERIC=en_US.UTF-8
UPSTART_SESSION=unix:abstract=/com/ubuntu/upstart-session/1001/1575
GNOME_KEYRING_CONTROL=
GTK_MODULES=gail:atk-bridge:unity-gtk-module
USER=mhuber
LD_LIBRARY_PATH=/opt/openrobots/lib:
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
LC_TELEPHONE=en_US.UTF-8
QT_ACCESSIBILITY=1
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
SSH_AUTH_SOCK=/run/user/1001/keyring/ssh
DEFAULTS_PATH=/usr/share/gconf/ubuntu.default.path
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/usr/share/upstart/xdg:/etc/xdg
DESKTOP_SESSION=ubuntu
PATH=/opt/openrobots/bin:/home/mhuber/bin:/home/mhuber/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
QT_IM_MODULE=ibus
QT_QPA_PLATFORMTHEME=appmenu-qt5
LC_IDENTIFICATION=en_US.UTF-8
XDG_SESSION_TYPE=x11
PWD=/home/mhuber/gazebo-yarp-plugins/build
JOB=gnome-session
XMODIFIERS=@im=ibus
GNOME_KEYRING_PID=
LANG=en_US.UTF-8
GDM_LANG=en
MANDATORY_PATH=/usr/share/gconf/ubuntu.mandatory.path
LC_MEASUREMENT=en_US.UTF-8
COMPIZ_CONFIG_PROFILE=ubuntu
IM_CONFIG_PHASE=1
GDMSESSION=ubuntu
SESSIONTYPE=gnome-session
GTK2_MODULES=overlay-scrollbar
SHLVL=1
HOME=/home/mhuber
XDG_SEAT=seat0
LANGUAGE=en
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
UPSTART_INSTANCE=
GAZEBO_PLUGIN_PATH=:/usr/local/lib
PYTHONPATH=/usr/local/lib/python2.7/dist-packages:/home/mhuber/anaconda2/lib/python2.7/site-packages:
UPSTART_EVENTS=started starting
XDG_SESSION_DESKTOP=ubuntu
LOGNAME=mhuber
DBUS_SESSION_BUS_ADDRESS=unix:abstract=/tmp/dbus-velmjY6etv
XDG_DATA_DIRS=/usr/share/ubuntu:/usr/share/gnome:/usr/local/share:/usr/share:/var/lib/snapd/desktop:/var/lib/snapd/desktop
QT4_IM_MODULE=xim
PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:
LESSOPEN=| /usr/bin/lesspipe %s
INSTANCE=Unity
UPSTART_JOB=unity-settings-daemon
XDG_RUNTIME_DIR=/run/user/1001
DISPLAY=:0
XDG_CURRENT_DESKTOP=Unity
GTK_IM_MODULE=ibus
LESSCLOSE=/usr/bin/lesspipe %s %s
LC_TIME=en_US.UTF-8
LC_NAME=en_US.UTF-8
XAUTHORITY=/home/mhuber/.Xauthority
OLDPWD=/home/mhuber/gazebo-yarp-plugins
_=/usr/bin/env
```

mhubii, 2018-03-07:
Okay thanks for taking the time. Then I will try to reinstall everything.

traversaro, 2018-03-07:
Yep, sorry for not being of more help. Happy to re-open if you are able to reproduce it on a different setup.

closed 2018-03-07 by traversaro