robotology/gazebo-yarp-plugins #351: Compilation error using robotology-superbuild on macOS High Sierra 10.13.3 (17D102)

DanielePucci, 2018-03-14:
As per the issue title, I get this error message
~~~
sysadmins-MacBook-Pro-2:GazeboYARPPlugins dpucci$ make VERBOSE=1
/usr/local/Cellar/cmake/3.10.2/bin/cmake -H/Users/dpucci/src/robotology-superbuild/robotology/GazeboYARPPlugins -B/Users/dpucci/src/robotology-superbuild/build/robotology/GazeboYARPPlugins --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/Cellar/cmake/3.10.2/bin/cmake -E cmake_progress_start /Users/dpucci/src/robotology-superbuild/build/robotology/GazeboYARPPlugins/CMakeFiles /Users/dpucci/src/robotology-superbuild/build/robotology/GazeboYARPPlugins/CMakeFiles/progress.marks
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/Makefile2 all
/Library/Developer/CommandLineTools/usr/bin/make -f libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/build.make libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/depend
cd /Users/dpucci/src/robotology-superbuild/build/robotology/GazeboYARPPlugins && /usr/local/Cellar/cmake/3.10.2/bin/cmake -E cmake_depends "Unix Makefiles" /Users/dpucci/src/robotology-superbuild/robotology/GazeboYARPPlugins /Users/dpucci/src/robotology-superbuild/robotology/GazeboYARPPlugins/libraries/singleton /Users/dpucci/src/robotology-superbuild/build/robotology/GazeboYARPPlugins /Users/dpucci/src/robotology-superbuild/build/robotology/GazeboYARPPlugins/libraries/singleton /Users/dpucci/src/robotology-superbuild/build/robotology/GazeboYARPPlugins/libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/DependInfo.cmake --color=
/Library/Developer/CommandLineTools/usr/bin/make -f libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/build.make libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/build
[  1%] Building CXX object libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/Handler.cc.o
cd /Users/dpucci/src/robotology-superbuild/build/robotology/GazeboYARPPlugins/libraries/singleton && /Library/Developer/CommandLineTools/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_yarp_singleton_EXPORTS -I/Users/dpucci/src/robotology-superbuild/robotology/GazeboYARPPlugins/libraries/singleton/include/GazeboYarpPlugins -isystem /Users/dpucci/src/robotology-superbuild/build/install/include -isystem /usr/local/include -isystem /usr/local/Cellar/gazebo9/9.0.0_3/include -isystem /usr/local/Cellar/gazebo9/9.0.0_3/include/gazebo-9 -isystem /usr/local/Cellar/bullet/2.87/include/bullet -isystem /usr/local/Cellar/bullet/2.87/include -isystem /usr/local/Cellar/simbody/3.5.4_2/include/simbody -isystem /usr/local/Cellar/sdformat6/6.0.0/include/sdformat-6.0 -isystem /usr/local/include/ignition/math4 -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_5/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_5/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_5/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_5/include/OGRE/Paging -isystem /usr/local/include/ignition/transport4 -isystem /usr/local/include/ignition/msgs1 -isystem /usr/local/include/ignition/common1 -isystem /usr/local/include/ignition/fuel_tools1 -I/Users/dpucci/src/robotology-superbuild/robotology/GazeboYARPPlugins/libraries/singleton/include -isystem /usr/local/include/eigen3 -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/Cellar/ffmpeg/3.4.2/include  -std=c++11 -stdlib=libc++ -O3 -DNDEBUG -fPIC   -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -I/usr/local/Cellar/ffmpeg/3.4.2/include -std=gnu++14 -o CMakeFiles/gazebo_yarp_singleton.dir/src/Handler.cc.o -c /Users/dpucci/src/robotology-superbuild/robotology/GazeboYARPPlugins/libraries/singleton/src/Handler.cc
In file included from /Users/dpucci/src/robotology-superbuild/robotology/GazeboYARPPlugins/libraries/singleton/src/Handler.cc:9:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/dev/PolyDriver.h:10:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/dev/Drivers.h:10:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/Network.h:17:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/Time.h:13:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/NetworkClock.h:11:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/BufferedPort.h:12:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/PortReaderBuffer.h:19:
/Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/PortReaderBufferBase.h:74:10: error: 'deprecated' attribute cannot be applied to types
    void YARP_DEPRECATED setAllowReuse(bool flag = true);
         ^
/Users/dpucci/src/robotology-superbuild/build/install/include/yarp/conf/compiler.h:2472:33: note: expanded from macro 'YARP_DEPRECATED'
#      define YARP_DEPRECATED [[deprecated]]
                                ^
In file included from /Users/dpucci/src/robotology-superbuild/robotology/GazeboYARPPlugins/libraries/singleton/src/Handler.cc:9:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/dev/PolyDriver.h:10:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/dev/Drivers.h:10:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/Network.h:17:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/Time.h:13:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/NetworkClock.h:11:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/BufferedPort.h:12:
In file included from /Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/PortReaderBuffer.h:19:
/Users/dpucci/src/robotology-superbuild/build/install/include/yarp/os/PortReaderBufferBase.h:75:10: error: 'deprecated' attribute cannot be applied to types
    void YARP_DEPRECATED release(yarp::os::PortReader* completed);
         ^
/Users/dpucci/src/robotology-superbuild/build/install/include/yarp/conf/compiler.h:2472:33: note: expanded from macro 'YARP_DEPRECATED'
#      define YARP_DEPRECATED [[deprecated]]
                                ^
2 errors generated.
make[2]: *** [libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/src/Handler.cc.o] Error 1
make[1]: *** [libraries/singleton/CMakeFiles/gazebo_yarp_singleton.dir/all] Error 2
make: *** [all] Error 2
~~~

traversaro, 2018-03-14:
This is related to https://github.com/robotology/yarp/issues/1531 (apparently for Linux this is generating a warning, but on macOS this is creating an issue).

traversaro, 2018-03-14:
@DanielePucci can you check if this solves the problem: https://github.com/robotology/yarp/pull/1591 ? Thanks

traversaro, 2018-05-17:
@DanielePucci I think we can close this issue.

closed 2018-05-17 by traversaro

DanielePucci, 2018-05-17:
Definitely, thanks