robotology/gazebo-yarp-plugins #354: Compilation error against yarp/devel

pattacini, 2018-05-23:
Just got the following compilation error upon updating to `yarp/devel`:

```sh
In file included from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:8:0:
/usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/include/yarp/dev/ControlBoardDriver.h:86:27: error: expected class-name before ‘,’ token
     public IControlLimits2,
                           ^
/usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/include/yarp/dev/ControlBoardDriver.h:130:67: warning: ‘ConstString’ is deprecated [-Wdeprecated-declarations]
     virtual bool getAxisName(int axis, yarp::os::ConstString& name);
                                                                   ^
In file included from /usr/include/c++/7/string:52:0,
                 from /usr/include/gazebo-9/gazebo/common/Plugin.hh:40,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/include/gazebo/ControlBoard.hh:10,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:7:
/usr/include/c++/7/bits/basic_string.h:77:11: note: declared here
     class basic_string
           ^~~~~~~~~~~~
In file included from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:8:0:
/usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/include/yarp/dev/ControlBoardDriver.h:315:84: warning: ‘ConstString’ is deprecated [-Wdeprecated-declarations]
     virtual bool getRemoteVariable(yarp::os::ConstString key, yarp::os::Bottle& val);
                                                                                    ^
In file included from /usr/include/c++/7/string:52:0,
                 from /usr/include/gazebo-9/gazebo/common/Plugin.hh:40,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/include/gazebo/ControlBoard.hh:10,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:7:
/usr/include/c++/7/bits/basic_string.h:77:11: note: declared here
     class basic_string
           ^~~~~~~~~~~~
In file included from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:8:0:
/usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/include/yarp/dev/ControlBoardDriver.h:316:90: warning: ‘ConstString’ is deprecated [-Wdeprecated-declarations]
     virtual bool setRemoteVariable(yarp::os::ConstString key, const yarp::os::Bottle& val);
                                                                                          ^
In file included from /usr/include/c++/7/string:52:0,
                 from /usr/include/gazebo-9/gazebo/common/Plugin.hh:40,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/include/gazebo/ControlBoard.hh:10,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:7:
/usr/include/c++/7/bits/basic_string.h:77:11: note: declared here
     class basic_string
           ^~~~~~~~~~~~
/usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc: In member function ‘virtual void gazebo::GazeboYarpControlBoard::Load(gazebo::physics::ModelPtr, sdf::ElementPtr)’:
/usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:95:43: warning: ‘yarp::os::ConstString’ is deprecated: Use std::string instead [-Wdeprecated-declarations]
                     yarp::os::ConstString ROS;
                                           ^~~
In file included from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Searchable.h:18:0,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:16,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/PortReader.h:13,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Portable.h:13,
                 from /usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/Network.h:15,
                 from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/Drivers.h:13,
                 from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/PolyDriver.h:13,
                 from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/PolyDriverDescriptor.h:14,
                 from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/PolyDriverList.h:13,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/include/gazebo/ControlBoard.hh:14,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:7:
/usr/local/src/robot/yarp/src/libYARP_OS/include/yarp/os/ConstString.h:18:68: note: declared here
 YARP_DEPRECATED_TYPEDEF_MSG("Use std::string instead") std::string ConstString;
                                                                    ^~~~~~~~~~~
In file included from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/PolyDriver.h:13:0,
                 from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/PolyDriverDescriptor.h:14,
                 from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/PolyDriverList.h:13,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/include/gazebo/ControlBoard.hh:14,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:7:
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/Drivers.h: In instantiation of ‘yarp::dev::DeviceDriver* yarp::dev::DriverCreatorOf<T>::create() [with T = yarp::dev::GazeboYarpControlBoardDriver]’:
/usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:192:1:   required from here
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/Drivers.h:124:16: error: invalid new-expression of abstract class type ‘yarp::dev::GazeboYarpControlBoardDriver’
         return new T;
                ^~~~~
In file included from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:8:0:
/usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/include/yarp/dev/ControlBoardDriver.h:79:18: note:   because the following virtual functions are pure within ‘yarp::dev::GazeboYarpControlBoardDriver’:
 class yarp::dev::GazeboYarpControlBoardDriver:
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/ControlBoardInterfacesImpl.h:13:0,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/include/yarp/dev/ControlBoardDriver.h:16,
                 from /usr/local/src/robot/gazebo-yarp-plugins/plugins/controlboard/src/ControlBoard.cc:8:
/usr/local/src/robot/yarp/src/libYARP_dev/include/yarp/dev/ControlBoardInterfaces.h:474:18: note:       virtual bool yarp::dev::IControlCalibration::calibrate(int, unsigned int, double, double, double)
     virtual bool calibrate(int axis, unsigned int type, double p1, double p2, double p3)=0;
                  ^~~~~~~~~
plugins/controlboard/CMakeFiles/gazebo_yarp_controlboard.dir/build.make:62: recipe for target 'plugins/controlboard/CMakeFiles/gazebo_yarp_controlboard.dir/src/ControlBoard.cc.o' failed
make[2]: *** [plugins/controlboard/CMakeFiles/gazebo_yarp_controlboard.dir/src/ControlBoard.cc.o] Error 1
CMakeFiles/Makefile2:521: recipe for target 'plugins/controlboard/CMakeFiles/gazebo_yarp_controlboard.dir/all' failed
make[1]: *** [plugins/controlboard/CMakeFiles/gazebo_yarp_controlboard.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

traversaro, 2018-05-23:
Unfortunately, I will be unable to fix these for at least a couple of days. Regular user are recommend to use the released version of `yarp`/`gazebo-yarp-plugins`. If someone needs to use `gazebo-yarp-plugins` with YARP on the devel branch is welcome to try to fix this and open a PR.

jeljaik, 2018-05-25:
I believe to have made all the necessary changes in the [**cleanupForYARP3.0**](https://github.com/robotology/gazebo-yarp-plugins/tree/cleanupForYARP3.0) branch, using:
`YARP` in `devel` and `icub-main` in `devel` as of the moment of this comment and I managed to compile everything on my machine (macos). The only test I've done afterwards is to launch gazebo and put "a robot" in it.

traversaro, 2018-05-25:
Thanks @jeljaik I opened a PR: https://github.com/robotology/gazebo-yarp-plugins/pull/356 .

traversaro, 2018-05-25:
Fixed in https://github.com/robotology/gazebo-yarp-plugins/pull/356 for the `devel` branch.

pattacini, 2018-05-25:
Still having the same compilation error on my WSL.

pattacini, 2018-05-25:
Now, it works like a charm.
Thanks!

closed 2018-05-25 by pattacini