robotology/gazebo-yarp-plugins #372: Linking of GazeboYARPPlugins in robotology-superbuild fails.

nunoguedelha, 2018-07-13:
### Issue description

This issue was introduced by commit 8bf4e24ada037d99d1fa50c67fc9239d012675bd. When building and linking the component within the superbuild we get the following error:
```
Ld /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/GazeboYARPPlugins/plugins/controlboard/Release/libgazebo_yarp_controlboard.dylib normal x86_64
    cd /Users/nunoguedelha/dev/robotology-superbuild/robotology/GazeboYARPPlugins
    export MACOSX_DEPLOYMENT_TARGET=10.12
...
...
ld: warning: directory not found for option '-L/usr/local/Cellar/gazebo8/8.6.0/lib/Release'
ld: warning: directory not found for option '-L/usr/local/Cellar/gazebo8/8.6.0/lib/gazebo-8/plugins/Release'
ld: warning: directory not found for option '-L/usr/local/Cellar/bullet/2.87/lib/Release'
ld: warning: directory not found for option '-L/usr/local/lib/Release'
ld: warning: directory not found for option '-L/usr/local/Cellar/ignition-transport3/3.1.0_1/lib/Release'
ld: warning: directory not found for option '-L/usr/local/Cellar/zeromq/4.2.5/lib/Release'
ld: warning: directory not found for option '-L/usr/local/Cellar/ossp-uuid/1.6.2_2/lib/Release'
Undefined symbols for architecture x86_64:
  "yarp::dev::GazeboYarpControlBoardDriver::getMotorTorqueParams(int, yarp::dev::MotorTorqueParameters*)", referenced from:
      vtable for yarp::dev::GazeboYarpControlBoardDriver in ControlBoardDriver.o
  "yarp::dev::GazeboYarpControlBoardDriver::setMotorTorqueParams(int, yarp::dev::MotorTorqueParameters)", referenced from:
      vtable for yarp::dev::GazeboYarpControlBoardDriver in ControlBoardDriver.o
  "non-virtual thunk to yarp::dev::GazeboYarpControlBoardDriver::getMotorTorqueParams(int, yarp::dev::MotorTorqueParameters*)", referenced from:
      vtable for yarp::dev::GazeboYarpControlBoardDriver in ControlBoardDriver.o
  "non-virtual thunk to yarp::dev::GazeboYarpControlBoardDriver::setMotorTorqueParams(int, yarp::dev::MotorTorqueParameters)", referenced from:
      vtable for yarp::dev::GazeboYarpControlBoardDriver in ControlBoardDriver.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

** BUILD FAILED **
```

### Build environment

System: MacOS Sierra 10.12
Build within robotology-superbuild (all components up to date, latest YARP commit on devel 8c421d66b57d767f7385a56fe0b7611c8c39f975)
build commands: xcodebuild

traversaro, 2018-07-13:
> all components up to date 

All components up to date in the `devel` branch, right?

nunoguedelha, 2018-07-13:
yes.

nunoguedelha, 2018-07-13:
The commit https://github.com/robotology/gazebo-yarp-plugins/commit/8bf4e24ada037d99d1fa50c67fc9239d012675bd change adds the keyword `override` to the method declarations for forcing the compiler to verify that the function is virtual. The body of methods like `getMotorTorqueParams` still have to be defined in case the methods are pure virtual in the inherited interface class `ITorqueControl`. In this case, for instance, the body definition of `getMotorTorqueParams` has been removed from the cpp files defining `GazeboYarpControlBoardDriver` class. But there is a workaround in Yarp defining a dummy body `{return false;}` for that same method in the interface class `ITorqueControl`.

traversaro, 2018-07-17:
On Linux, the problem is  at runtime have the problem at runtime, launching `gazebo --verbose` we got : 
~~~
[Err] [Plugin.hh:180] Failed to load plugin /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: undefined symbol: _ZN4yarp3dev28GazeboYarpControlBoardDriver20getMotorTorqueParamsEiPNS0_21MotorTorqueParametersE
[Err] [Plugin.hh:180] Failed to load plugin /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: undefined symbol: _ZN4yarp3dev28GazeboYarpControlBoardDriver20getMotorTorqueParamsEiPNS0_21MotorTorqueParametersE
[Err] [Plugin.hh:180] Failed to load plugin /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: undefined symbol: _ZN4yarp3dev28GazeboYarpControlBoardDriver20getMotorTorqueParamsEiPNS0_21MotorTorqueParametersE
[Err] [Plugin.hh:180] Failed to load plugin /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: undefined symbol: _ZN4yarp3dev28GazeboYarpControlBoardDriver20getMotorTorqueParamsEiPNS0_21MotorTorqueParametersE
[Err] [Plugin.hh:180] Failed to load plugin /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: undefined symbol: _ZN4yarp3dev28GazeboYarpControlBoardDriver20getMotorTorqueParamsEiPNS0_21MotorTorqueParametersE
[Err] [Plugin.hh:180] Failed to load plugin /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: /home/straversaro/src/robotology-superbuild/build/install/lib/libgazebo_yarp_controlboard.so: undefined symbol: _ZN4yarp3dev28GazeboYarpControlBoardDriver20getMotorTorqueParamsEiPNS0_21MotorTorqueParametersE
[ERROR]ERROR Yarp Network was not found active in ApplyExternalWrench plugin
~~~

nunoguedelha, 2018-07-17:
I see. Well for now I'm using a workaround for being able to link the superbuild (branch `nunoguedelha/workaround` in this repo. I would be super happy to help you on this, but I'm hurrying to finish my tests and the paper.

I don't see why we don't keep the original `ITorqueControl` interface and restore the `{return false;}` body in the classes implementing that interface.
BTW, The keywork `override` after a given method means that that method overrides the one from the inherited interface, not that it is overridden by a derived class.

traversaro, 2018-07-17:
I think the fix in https://github.com/robotology/gazebo-yarp-plugins/pull/374  make sense, and the removal of the `return false` in https://github.com/robotology/gazebo-yarp-plugins/commit/8bf4e24ada037d99d1fa50c67fc9239d012675bd#diff-732527c53bcfabfbb15e5399654ff8b6L123 was just an oversight.

nunoguedelha, 2018-07-17:
Yes absolutely. Just approved.

nunoguedelha, 2018-07-17:
Thanks @traversaro !

closed 2018-07-17 by nunoguedelha

nunoguedelha, 2018-07-17:
@traversaro I'm having the same issue now with `GazeboYarpFakeControlBoardDriver`.  I'm pushing the fix.